esphome/feature-requests #1412: Espresense

hermanndererdmann, 2021-09-13:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

This Project might be very interessting when integrated in Esphome.

https://espresense.com/
**Additional context**

mloebl, 2021-09-15:
I'd be interested in this as well. I just started using Espresense and I am really impressed. With some of the Chinese OEMs starting to make ESP32 based plug in /outlet modules with bluetooth built in, this would make distributing this around a house seamless for many people and a good fit for the other ESPHome abilities like power monitoring and such.

travipross, 2021-10-04:
Also very interested!

I'm currently considering desiging an ESP-based module to be placed around the home for primarily temperature sensing. Simultaneously I'm interested in exploring ESPresence. Without integration between these two projects it means up to 2x ESPs per room for this to work nicely. Having this integration support would possibly cut the number of devices in half.

FM-17, 2021-10-04:
I'm hoping to add ESPresence-enabled ESP32's to my home for pet/asset tracking as well as ESPHome-enabled ESP32's for temp + humidity monitoring and PIR motion detection. Using two ESP32's per room for this would be wasteful, so I too would love to see this added.

corgan2222, 2021-10-04:
An integration into ESPHome would be really nice! 

If you don't want to wait till this happens, I implemented temp + humidity + pir + radar sensor support into ESPresence. 
https://github.com/ESPresense/ESPresense/pull/54
https://github.com/ESPresense/ESPresense/pull/48

coming this week:
- lux sensor

If you need other sensors, just let us know.

virtualducttape, 2021-10-05:
This would be great! Esp home is wonderful, adding espresence into it would be a killer feature.

Robert7997, 2021-11-03:
Building a Multisensor with color led notifikation AND espresence would be a dream 😉👍

I will follow this very closely!

corgan2222, 2021-11-05:
> 
> 
> Building a Multisensor with color led notifikation AND espresence would be a dream 😉👍
> 
> I will follow this very closely!

I'm actually playing around with LED Support for ESPresense.
How would you use the LED? Should the LED be shown in HA as Light Entity, or in which way you want to use it?

rpatel3001, 2022-01-24:
If anyone is interested, I put together some code for my own use with esphome that borrows some of the filtering scheme from ESPresense. It doesn't do fingerprinting and is only setup for iBeacons, but if you're handy with code you could easily extend it however you like.

https://github.com/rpatel3001/BleDistance

murtoz, 2022-02-07:
> How would you use the LED? Should the LED be shown in HA as Light Entity

Yes please. HA is my notification engine and has all the logic. Ideally I would be able to add simple effects to the led like you can in esphome so I can flash it to get my attention. But at least on/off/colour/brightness control from HA to make it usable.

Hedda, 2022-05-16:
+1 for native room presence detection in ESPHome for Home Assistant's [mqtt_room component](https://www.home-assistant.io/components/sensor.mqtt_room/) or similar "esphome_presence" or "esphome_room" component for localized device presence detection using "trackable beacons" and "base stations" in all rooms.

https://www.home-assistant.io/integrations/mqtt_room/

[ESPresense](https://espresense.com/) was a rewrite of [ESP32-mqtt-room](https://jptrsn.github.io/ESP32-mqtt-room) so maybe ESPHome developers would prefer rewriting it again to not use MQTT?

Ex. Home Assistant built-in integration for presence-based lights builds on the [device_tracker](https://www.home-assistant.io/integrations/device_tracker/) and the [person](https://www.home-assistant.io/integrations/person/) integrations:

https://www.home-assistant.io/integrations/device_sun_light_trigger/

https://www.home-assistant.io/integrations/device_tracker/

https://www.home-assistant.io/integrations/person/

MRobi1, 2022-09-07:
This PR looks amazing! Thank you @wjtje for working on this. Having the ability to do room-level presence while still having access to all the sensors available in ESPHome will be game-changing for many!

virtualducttape, 2022-09-07:
Agreed, so excited to see this feature added! (And lucky for me, I've been so lazy that I haven't even started setting up the ESPs around my new house yet)

danps1, 2023-01-24:
Hi,

I created a feature request outlining a way to solve this problem, if you're like to look at https://community.home-assistant.io/t/apple-devices-presence-using-esphome-ble-proxy-espresense-alternative/524019/1

Thanks

Dan