esphome/feature-requests #1413: Climate/Daikin - support also 0.5C temperature steps

grob6000, 2021-09-14:
**Describe the problem you have/What new integration you would like**

Currently, daikin ir remote only allows setting target temperatures in 1.0C steps (refer to daikin.h). My Daikin unit's remote does 0.5C steps. ESPHome is ignoring 0.5C increments (ignoring the last bit of the temperature field), so when I set to '22.5C' on the remote, ESPHome is updating state to '22C'.

Suggest adding a config line for daikin climate  "halfdegree: true", which then:
(a) inits the ir climate with 0.5 degree increments instead of 1.0 degree increments
(b) receiver reads the last bit of the temp field and sets the 0.5C correctly
(c) transmitter sends the 0.5C bit
For users with whole-degree remotes/units, they can leave the setting out (default false).

**Please describe your use case for this integration and alternatives you've tried:**

To work with my air conditioner / support more seamless operation with remote.
Alternatives: none available. Not sure whether the custom climate could be used to create this situation.

**Additional context**

jnimmo, 2022-06-14:
I'll be submitting a PR to add support for 0.5 degree increments in the coming weeks (fingers crossed) - work in progress here https://github.com/jnimmo/esphome/commit/adb7eeb1cd544c4d66801bf64dda38f61b4e63fb

jnimmo, 2022-10-05:
Still waiting on the PR to be approved - do you have any feedback on how the Daikin preset modes should be handled? (see discussion on PR)