esphome/feature-requests #1414: ESP-IDF Conversion Tracking

OttoWinter, 2021-09-14:
Not quite a feature-request, but an progress discussion.

We've been working on making esphome directly target esp-idf on ESP32s (without the arduino middle layer). This has been combined with extensive work to generalize a bunch of hardware abstractions.

Current state: (see also [esp-idf branch](https://github.com/esphome/esphome/tree/esp-idf))

- [x] Create system so "target platforms" like "esp32" are in their own integration module.
- [x] Convert ESP32 to that modular system
- [ ] Convert ESP8266 to that modular system
- [x] Create arduino-independent core abstractions for GPIOPin
- [x] Make esp-idf use those abstractions
- [ ] Rewrite esp8266/esp32 arduino to use them
- [x] Backport some commonly used functions like millis/yield/etc
- [x] Add ESP-IDF implementation of wifi integration
- [ ] Small changes to ESP32/ESP8266 Arduino wifi integrations (new methods)
- [x] Create an abstraction for Preferences backend
- [x] Implement Preferences for ESP32 Arduino/IDF
- [ ] Implement Preferences for ESP8266
- [ ] i2c Component
- [ ] spi Component
- [ ] uart Component
- [ ] ota Component (networking part is ready with sockets conversion)
- [ ] Convert all `ARDUINO_ARCH_ESPXX` to `#ifdef USE_ESPXX`
- [ ] Add new clang-tidy target for esp-idf

Not in scope:
- Porting some Arduino Libraries we use including:
- `web_server` (and `captive_portal`): `ESPAsyncWebServer` and `AsyncTCP`
- `mqtt`: `AsyncMQTTClient` and `AsyncTCP`

Contributions are welcome! Just create a PR against the `esp-idf` branch

OttoWinter, 2021-09-17:
Good news! The implementation part of the conversion is pretty much complete now :tada: - all APIs that needed refactoring are now refactored and any compile problems in dependent components fixed, including all automated test compiles/static-analysis.

Now comes the testing... I've not checked the code for if it works in most places, so things might be broken; It looks like Arduino/ESP-IDF do work with some simple configs, but we need to test if this is true even for more complex things like BLE/i2c/spi/uart.

The way I see it the best way forward is now: 
1. ensure no existing stuff is broken (meaning arduino-based ESP32/ESP8266)
2. merge the changes
3. fix any issues with ESP-IDF (for example BLE looks a bit broken)

jesserockz, 2021-09-23:
One more thing I found today is that ESP32 using IDF no longer has hardware SPI support. This will need to be added at some point. Probably good to split the SPI now similar to I2C and UART using separate classes.

grischard, 2022-02-21:
Are there already separate issues to follow for the `web_server`, `captive_portal`, `mqtt` etc. features? I tried searching but didn't find any.

OttoWinter, 2022-02-22:
There is not, indeed would be good to split it up. I will close this as the main idf conversion is done now.

closed 2022-02-22 by OttoWinter