esphome/feature-requests #1417: Add support for TM1640 chipset to fastled_spi

rolandsteinmeyer, 2021-09-16:
**Describe the problem you have/What new integration you would like**

The 8x8 matrix LED Shield for the wemos D1 mini platform should be supported by esphome. This is a cheap device  targeted at DIY projects. It uses a TM1640 controller, which is currently not supported by esphome.
 
I want to use this device as a flexible smart home status display, e.g. integrated into key hanger with one LED per open window, so you can check when leaving the house with one glance if windows have been left open.
 

TM1640 currently unsupported.

Link to device description / arduino library:  [  https://www.wemos.cc/en/latest/d1_mini_shield/matrix_led.html

pascaltippelt, 2022-09-03:
I would appreciate this too.

itanczos, 2022-09-03:
I would be very happy about it!

realthk, 2022-12-10:
It is very easy to create your own component with external_components feature of Esphome.
I just did this for a very old 16 digit TM1640 board, by copying and modifying TM1637 component 

https://github.com/realthk/esphome_tm1640

[image]

mteletin, 2023-07-07:
Would very much appreciate native support for this little LED matrix screen.

nevada750club, 2024-04-20:
I purchased the Shield Matrix Led of the WeMos D1 Mini and I would like to program it in ESPHome.
Could we have native integration for all the WeMos D1 Mini Shields?

https://www.wemos.cc/en/latest/d1_mini_shield/index.html

AdamM68, 2024-10-13:
> It is very easy to create your own component with external_components feature of Esphome. I just did this for a very old 16 digit TM1640 board, by copying and modifying TM1637 component
> 
> https://github.com/realthk/esphome_tm1640
> 
> [image]

How should I add this library to HA ? what dir ? should I only copy the lib or it should be registered somehow ? 
Can I find it on HACS ?

Thanks for help

realthk, 2024-10-16:
> TM1640 github repo : how should I add this library to HA ? what dir ? should I only copy the lib or it should be registered somehow ? Can I find it on HACS ?

I rarely build any ESPHome firmware under HA, but on my desktop computer where it is much easier to config, faster to build, and easily connected to Github repos. After it is uploaded to the ESP board, HA discovers it just as one created there.

AdamM68, 2025-06-29:
there is still an issuse how to use it on that LED shield (that also has TM1640) : 
 [(https://www.wemos.cc/en/latest/d1_mini_shield/matrix_led.html)]

has anyone an idea how to use it with esphome ?

[image]

pantgr, 2026-10-03:
For anyone still looking for TM1640 support: I wrote an ESPHome external component for the **AiP1640**, which per both datasheets has the same SOP28 pinout, commands, RAM map and bus timing as the TM1640 (differences: VDD range, VIL threshold, oscillator 400 vs 450 kHz).

It gives raw access to the 16 RAM bytes (`ram[]` + `write_all()`, `write(addr, byte)`, `brightness()`), so a 7-segment clock or an 8x8 matrix like the Wemos shield can be drawn from a lambda.

Tested on AiP1640 only (two LED clocks on ESP32-S3, in daily use) - a report from someone with a real TM1640 would be welcome.

https://github.com/pantgr/esphome-aip1640-clock - wiring with a 74HCT245 level shifter and pinouts are in the README.