esphome/feature-requests #1422: Wemos motor Shield (TB6612NFG)

nemocaledonie, 2021-09-19:
**Describe the problem you have/What new integration you would like**

I have some Wemos motor shield and I would like use for driving a fan. It's an I2C controler.
I've tried to using custom component without success.
the library using is : https://github.com/wemos/LOLIN_I2C_MOTOR_Library
Thanks a lot.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

WeekendWarrior1, 2021-09-19:
Can you link the shield?
I didn't think the TB6612NFG was i2c: [https://www.pololu.com/product/713](https://www.pololu.com/product/713)
And if it isn't, you can control it with [https://esphome.io/components/fan/hbridge.html](https://esphome.io/components/fan/hbridge.html)

nemocaledonie, 2021-09-20:
Hi,
Thanks for the response.
https://www.wemos.cc/en/latest/d1_mini_shield/motor.html.  
it's the link for the shield . It's indicated : I2C dual motor driver shield.

I'll trying with H-bridge but I'm doubt it's working ;)
The I2C chip on the board is : STM32F030 

https://github.com/wemos/LOLIN_I2C_MOTOR_Library
The library.
Thanks for your help.

WeekendWarrior1, 2021-09-20:
lol wth
They added all this extra i2c complexity to such a simple component that just needs 2 GPIO per motor...
And in doing so they take away the control of choosing FAST or SLOW decay modes

My advice is you grab a TB6612NFG breakout board (or another common H-bridge that fits your needs) without the i2c interface, which will "just werk" with the h-bridge library.
I will purchase and build a component for this d1 mini motor shield but it's not going to be a priority of mine, and I'll be buying it off Ali so it'll take awhile to arrive.

nemocaledonie, 2021-09-22:
Thanks for the response.
I'll waiting until you grab one :) . 
If I buy another H-Bridge I'll wait for a looooong time, China to New Caledonia take approx 3/5 months... if they arrive.
Have a good day.