esphome/feature-requests #1423: esphome snapshot or backup bin functionality

WeekendWarrior1, 2021-09-19:
**Describe the problem you have/What new integration you would like**
When a new version of esphome is released, their can be unforeseen breaking changes.
Users often use the esphome dashboard within their HA installation to update their esphome devices, which may break them.
These users then struggle to rollback their devices because it is challenging to run an older version of esphome from within a supervised HA installation.
esphome should keep snapshots or backups of old BINs that users can upload through  an simple command.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**
My simple solution:
esphome detects that the compiled version on the filesystem is from an older version of esphome.
prior to compiling, it copies that to another directory as a backup
Then if something messes up in the most recent version, you could run something like "esphome revert myConfig.yaml" to upload the compiled bin from the previous version.

MauriceM's more advanced idea (discussed on discord):

But sometimes you'd want to revert, because you messed up within a single ESPHome version. It feels snapshotty, so it'd be something like:
```sh
$ esphome list snapshots myConf   (myConf being a search string)
2021-09-17 11:12:01 build-2021.9.0-871b39 myConfig.yaml
2021-08-14 16:01:55 build-2020.8.2-871b39 myConfig.yaml
2021-08-14 12:26:31 build-2020.8.2-77fa6e myConfigTest.yaml
2021-08-13 16:01:55 build-2020.8.2-4f77a1 myConfig.yaml
$ esphome upload snapshot build-2020.8.2-871b39 myConfig.yaml
```

The  snapshot name consisting of the ESPHome version and an MD5 (partially shown) for the config file that was used. Such build file could even be a zip containing both the bin file and the config that was used. In which case commands like these could be usable too:
```sh
$ esphome upload snapshot build-2020.8.2-871b39
$ esphome show snapshot config build-2020.8.2-871b39
```
(upload snapshot should still be usable with a specific yaml file, to be able to upload with modified network parameters or a different device altogether).

OttoWinter, 2021-10-03:
In principle: yeah that would be nice.

But to have it be useful it needs to be in the dashboard interface (most people use ESPHome that way).

Also: it's not as simple as just backing up the .yaml file; for example the yaml can contain `package`s, so it would also need to detect those and have some logic to restore them too.

WeekendWarrior1, 2021-10-04:
> But to have it be useful it needs to be in the dashboard interface (most people use ESPHome that way).

Yes that would be the priority, and is the thinking behind this FR.

> Also: it's not as simple as just backing up the .yaml file; for example the yaml can contain packages, so it would also need to detect those and have some logic to restore them too.

I would think the yaml, .esphome and all:
[image]
should be backed up to facilitate the restore process.
In my mind this features main use would be for rolling back from new versions of esphome that may have broken a component, rather than for being an alternative to config version control, so perhaps the yaml is of the least priority to backup.