esphome/feature-requests #1424: SI1145 UV index sensor support

markusg1234, 2021-09-20:
**Describe the problem you have/What new integration you would like**
Support for the SI1145 UV index sensor that communicates over I2C

examples: 
https://www.adafruit.com/product/1777
https://wiki.seeedstudio.com/Grove-Sunlight_Sensor/

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

berfenger, 2021-09-28:
A few weeks ago I published an external component to support the SI1145 uv sensor.
Here is an example: https://github.com/berfenger/esphome_components/blob/main/example_si1145.yaml

Let me know if it works for you before issuing a merge request to the esphome mainline.

markusg1234, 2021-09-29:
Thankyou so much.
I will be able to try it in the next few days.
Regards,
Markus

markusg1234, 2021-10-01:
Hi, 
Tested and it seems to work.
The only change I made was adding a device class of illumiance for the UV index as no icon was displayed by default like all of the others.
Regards,

Markus

markusg1234, 2021-10-19:
Hi, 
Just started building a weather station.
for some reason I am now getting a delay error and the build fails using the example yaml.

Regards,
Markus 

INFO Reading configuration /config/esphome/weatherstation.yaml...
INFO Cloning https://github.com/berfenger/esphome_components.git@None
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/weatherstation
Processing weatherstation (board: esp-wrover-kit; framework: arduino; platform: espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF Modes: Finder ~ off, Compatibility ~ soft
Dependency Graph
|-- <WiFi> 1.0
|-- <Update> 1.0
|-- <Wire> 1.0.1
|-- <ESPmDNS> 1.0
Compiling /data/weatherstation/.pioenvs/weatherstation/src/esphome/components/si1145/si1145.cpp.o
Compiling /data/weatherstation/.pioenvs/weatherstation/src/esphome/components/switch/switch.cpp.o
Compiling /data/weatherstation/.pioenvs/weatherstation/src/esphome/components/template/sensor/template_sensor.cpp.o
Compiling /data/weatherstation/.pioenvs/weatherstation/src/esphome/components/template/switch/template_switch.cpp.o
Compiling /data/weatherstation/.pioenvs/weatherstation/src/esphome/components/template/text_sensor/template_text_sensor.cpp.o
Compiling /data/weatherstation/.pioenvs/weatherstation/src/esphome/components/text_sensor/filter.cpp.o
Compiling /data/weatherstation/.pioenvs/weatherstation/src/esphome/components/text_sensor/text_sensor.cpp.o
Compiling /data/weatherstation/.pioenvs/weatherstation/src/esphome/components/uptime/uptime_sensor.cpp.o
src/esphome/components/si1145/si1145.cpp: In member function 'virtual void esphome::si1145::SI1145Component::update()':
src/esphome/components/si1145/si1145.cpp:111:11: error: 'delay' was not declared in this scope
   delay(20);
           ^
src/esphome/components/si1145/si1145.cpp: In member function 'void esphome::si1145::SI1145Component::reset_()':
src/esphome/components/si1145/si1145.cpp:333:11: error: 'delay' was not declared in this scope
   delay(10);
           ^
*** [/data/weatherstation/.pioenvs/weatherstation/src/esphome/components/si1145/si1145.cpp.o] Error 1
========================== [FAILED] Took 1.62 seconds ==========================

closed 2022-07-01 by nagyrobi