esphome/feature-requests #1426: Ability to select which devices to group update in Home Assistant Addon

AneticsUK, 2021-09-20:
**Describe the problem you have/What new integration you would like**

Currently we only have the options of "Update All" or manually update one device,
Having the option to select which devices to group update would be a really nice feature to have,

**Please describe your use case for this integration and alternatives you've tried:**

Like many others i have devices that may have issues with the newest version, may be temporarily offline or are just no longer used but have a config i don't want to delete, being able to exclude devices from update would save a lot of time.

**Additional context**

Jpsy, 2022-03-16:
Me too, I am waiting for this feature for a long time.

Roving-Ronin, 2024-09-04:
Would be handy, especially if you use power monitoring plugs with relays, as updating them means the relays reset momentarily and crash any devices / PCs connected to them.