esphome/feature-requests #1429: Firebeetle ESP-32E

rmandjes, 2021-09-23:
**Describe the problem you have/What new integration you would like**
Hi,

There is a new Firebeetle ESP32-E device.
Could you support this device?

More information could be found on this site
https://wiki.dfrobot.com/FireBeetle_Board_ESP32_E_SKU_DFR0654

**Please describe your use case for this integration and alternatives you've tried:**
See above

**Additional context**
See above

kozmaz87, 2022-08-10:
I can confirm. This device works fine. I have a question though. Where do I document that the battery voltage measurement is done like this?  Is there a wiki somewhere for each device in esphome and the tricks regarding their programming?

```  - platform: adc
    name: "Battery voltage"
    pin: GPIO34
    accuracy_decimals: 3
    update_interval: 5s
    attenuation: 11dB
    filters:
      - multiply: 2.0
```

It took me a while to find out what to do... for example the attenuation and which pin it was meant to be was not all that obvious...

But it now works well....

```
![image](https://user-images.githubusercontent.com/4070837/183937850-89714a2b-7531-428a-9666-2e21439983c2.png)

```

nagyrobi, 2022-08-10:
I'd suggest to add this info to the https://esphome.io/components/sensor/adc.html page at the bottom.
Like it's done for Ethernet: https://esphome.io/components/ethernet.html

closed 2022-08-10 by nagyrobi

JeroenTuinstra, 2022-11-29:
I am sorry, I understand the topic is closed. But I tried the battery settings on FireBeetle ESP32 v4.0, but I get really low battery readings. 

```
- platform: adc
    pin: A0
    name: "Agenda Display Battery"
    id: voltage
    attenuation: 11db
    filters:
      - multiply: 20
```
This the result ... but it is charging and the level doesn't seem to change: 
```
[22:55:06][D][sensor:127]: 'Agenda Display Battery': Sending state 2.84000 V with 2 decimals of accuracy
[22:56:06][D][sensor:127]: 'Agenda Display Battery': Sending state 2.84000 V with 2 decimals of accuracy
[22:57:06][D][sensor:127]: 'Agenda Display Battery': Sending state 2.84000 V with 2 decimals of accuracy
[22:58:06][D][sensor:127]: 'Agenda Display Battery': Sending state 2.84000 V with 2 decimals of accuracy
[22:59:06][D][sensor:127]: 'Agenda Display Battery': Sending state 2.84000 V with 2 decimals of accuracy
[23:00:06][D][sensor:127]: 'Agenda Display Battery': Sending state 2.84000 V with 2 decimals of accuracy
[23:01:06][D][sensor:127]: 'Agenda Display Battery': Sending state 2.84000 V with 2 decimals of accuracy
[23:02:06][D][sensor:127]: 'Agenda Display Battery': Sending state 2.84000 V with 2 decimals of accuracy
[23:03:06][D][sensor:127]: 'Agenda Display Battery': Sending state 2.84000 V with 2 decimals of accuracy
```
Does this board not have an internal voltage divider? Its this board: 
[https://wiki.dfrobot.com/FireBeetle_ESP32_IOT_Microcontroller(V3.0)__Supports_Wi-Fi_&_Bluetooth__SKU__DFR0478](url)

kozmaz87, 2022-11-29:
Did you try my original suggestion? 
```
- platform: adc
    name: "Battery voltage"
    pin: GPIO34
    accuracy_decimals: 3
    update_interval: 5s
    attenuation: 11dB
    filters:
      - multiply: 2.0
```

JeroenTuinstra, 2022-11-29:
@kozmaz87 Yep tried that too. This was the result:
```
[23:28:57][D][sensor:127]: 'Agenda Display Battery': Sending state 0.28400 V with 3 decimals of accuracy
```

kozmaz87, 2022-11-29:
It is very possible that the new version of the firebeetle is wired
differently. Check its schematics on the website it should tip you off... I
think you may be looking at the wrong pin or the voltage divider is
configured differently and therefore you need different multiplier.

Best regards,
Zoltan Kozma

JeroenTuinstra, 2022-11-29:
@kozmaz87 Unfortunately I am not able to read the schematics (all abracadabra for me, sorry). Would you be able to take a look and see if pin A0 is not the right pin: 
[https://raw.githubusercontent.com/DFRobot/Wiki/master/DFR0478/%5BDFR0478%5DFireBeetle%20Board-ESP32(V4.0)%E7%94%9F%E4%BA%A7%E6%A3%80%E6%9F%A5%E5%9B%BE.PDF](https://raw.githubusercontent.com/DFRobot/Wiki/master/DFR0478/%5BDFR0478%5DFireBeetle%20Board-ESP32(V4.0)%E7%94%9F%E4%BA%A7%E6%A3%80%E6%9F%A5%E5%9B%BE.PDF)

kozmaz87, 2022-11-29:
So this used to have the voltage divider that was connected to the right
pin... one can of course trace the traces on the board to verify but on the
new schematic I do not see this part:
https://dfimg.dfrobot.com/nobody/wiki/fd28d987619c16281bdc4f40990e5a1c.PDF

[image: image.png]
oh wait... here it is:

[image: image.png]

They made the circuit much more elaborate and I can see a capacitor there
that I think is there to smooth out the signal and reduce ripple I think.
So yes you have been looking at the right input(A0) but you have to fiddle
with the multiplier and the attenuation to get a decent reading.

kozmaz87, 2022-11-29:
I would try lower attenuation and the multiplier should only be 2 as the 2 1MOhm resistors should halve the voltage.

kozmaz87, 2022-11-29:
OK so here is the schematic for the 3.0 
https://images-na.ssl-images-amazon.com/images/I/A1VO7RyKHzL.pdf

I am  seeing the same voltage divider on it and it is on A0

JeroenTuinstra, 2022-11-29:
@kozmaz87 I will try fiddling with that. I don't see version 3.0 - the link i provided is for version 4.0. 
I tried multiplier 2.0, but gave a very low reading 0.28400 volt. I will try with different attenuation levels.

kozmaz87, 2022-11-29:
You do not have anything else connected to A0 right?

JeroenTuinstra, 2022-11-29:
With attenuation level 6db the result (multiplier 2):
```
[00:28:26][D][sensor:127]: 'Agenda Display Battery': Sending state 0.21400 V with 2 decimals of accuracy
```
With attenuation level 2.5db the result (multiplier 2):
```
[00:31:03][D][sensor:127]: 'Agenda Display Battery': Sending state 0.15600 V with 2 decimals of accuracy
```
With attenuation level 0db the result (multiplier 2):
```
[00:32:44][D][sensor:127]: 'Agenda Display Battery': Sending state 0.15000 V with 2 decimals of accuracy
```

So I don't seem to be able to get a proper reading at all.

kozmaz87, 2022-11-29:
Well this is odd... what is also odd is that 2.5 and 0db seems to cap your reading... a reading that should be good within the 0-3.3V range I think... it is behaving very weird... I wonder if the A0 input is busted in the MCU..... Can you connect A0 to another analog input and try to measure with that?

kozmaz87, 2022-11-30:
Alternatively you can use a multimeter measure the battery voltage use `raw: true` and come up with your own calibration coefficients and make filters using those... it should be linear as long as the adc works at all... it is possible that the calibration of the adc is busted on the board.... mine was fine but I have the USB-C version of this board that is newer and better.

kozmaz87, 2022-11-30:
In case you give up on it this is my board that my original code should be working with https://www.dfrobot.com/product-2195.html

JeroenTuinstra, 2022-11-30:
I have a second one laying around. Will try that one and see if the ADC is broken. I have also contacted DFRobot to see if they can make any sense of this. 
How would I connect A0 to another anolog connector. Just with a jumper cable? I can try that.

JeroenTuinstra, 2022-11-30:
Okay when I connect A0 to A3 or GPIO35 it gives a reading: 
```
[D][sensor:127]: 'Agenda Display Battery': Sending state 0.50800 V with 2 decimals of accuracy
```

JeroenTuinstra, 2022-11-30:
After a reset it is back to its old voltage. It just remains on:
```
[01:30:47][D][sensor:127]: 'Agenda Display Battery': Sending state 0.28400 V with 2 decimals of accuracy
```
I will try with a multimeter. To see the voltage of the battery.

JeroenTuinstra, 2022-12-08:
Got an answer from DFRobot about pin A0:
```Sorry, esp32 cannot use A0 to read battery level.```

Made my own divider now and hooked up to pin A2/GPIO34. And it seems to work. Get good readings. Now waiting to see if the divider drains the battery.

Followed this guide: [https://community.home-assistant.io/t/onboard-battery-level-of-this-esp32-voltage-divider-battery-level/326056/6](https://community.home-assistant.io/t/onboard-battery-level-of-this-esp32-voltage-divider-battery-level/326056/6)

Ended up using a 220k and 470k resistors for the divider. Seems to work good.