esphome/feature-requests #1430: Matter (formerly Project CHIP / Connected Home over IP) Application Layer support in ESPHome?

Hedda, 2021-09-23:
**Describe the problem you have/What new integration you would like**

Request native support for Matter (formerly **_Project CHIP_** / **_Connected Home over IP_**) Application Layer in the ESPHome firmware?

- https://buildwithmatter.com
- https://github.com/project-chip
- https://github.com/project-chip/connectedhomeip

Espressif Systems who makes ESP8266/ESP32 is part of the Connectivity Standards Alliance (formerly known as the Zigbee Alliance who governs the Matter/CHIP standard), and Espressif ESP32 Wi-Fi platform is one of the reference platforms for Matter/CHIP.

- https://www.espressif.com/en/news/Matter_on_ESP32
- https://zigbeealliance.org/member/espressif-systems/

To get all the details on exactly how developers can use Matter on ESP32, you can read Hrishikesh Dhayagude's blogpost on the ESP Journal.

- https://blog.espressif.com/announcing-matter-previously-chip-on-esp32-84164316c0e3
  - https://medium.com/the-esp-journal/announcing-matter-previously-chip-on-esp32-84164316c0e3

While still a work-in-progress you can find the ESP32 code in the Matter/CHIP SDK here:

- https://github.com/project-chip/connectedhomeip/search?q=esp32

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

Not sure if it is still the case but Project CHIP initially mandated the use of IPv6 addresses, so not sure if IPv6 support is still a requirement or if latest Matter Application Layer specification now also support IPv4. Also see IPv6 support request -> https://github.com/esphome/feature-requests/issues/718

oxan, 2022-01-11:
I tried to ascertain how hard it would be to implement this, but there doesn't seem to be a getting-started guide, and I couldn't understand how the `temperature-measurement-app` works (where does it actually report a value?).

Hedda, 2022-01-12:
FYI, there looks to maybe be some related discussion in https://github.com/espressif/esp-idf/issues/8215 about Matter with ESP-IDF as far as I can tell?

It sounds as if Espressif is working on an upcoming Matter (Project CHIP) specific SDK which will generate an external component?

Their Matter-specific SDK is not ready yet so for for they refer to [all-clusters-app example](https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app) and [matter cmake](https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/CMakeLists.txt) in shell environment.

Hedda, 2022-04-08:
FYI, Project CHIP is also working on "OSKI" (Operating System Kernel Interface) as a OSAL (Operating System Abstraction Layer):

https://github.com/project-chip/osal

As I understand it is meant to allow a different apps or device layer codebase to run on different hardware platforms?

https://github.com/project-chip/osal/blob/main/README.md

jhansche, 2022-06-15:
Not ESPHome relatted, but also cool to see this progress in Home Assistant: https://www.home-assistant.io/blog/2022/05/29/matter-in-home-assistant-workshop-announcement/

Will be great to be able to add very simple yaml config to add matter support right to the ESP32 device config.
```
matter:
  ... node info, endpoints, services, etc
```

Hedda, 2022-09-27:
FYI, ESP-IDF (Espressif IoT Development Framework) look to now have posted an OpenThread RCP (Radio Co-Processor) example:

https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp

I think that is needed for ESPHome to support OpenThread and Matter over Thread on ESP32-H2 with its IEEE 802.15.4 radio?

https://www.espressif.com/en/news/ESP32_H2

At least if that could be built so firmware provides OpenThread RCP (Radio Co-Processor) mode with SPI interface over the Spinel protocol (spinel+hdlc+uart serial protocol) and tunnel that serial stream to agners OTBR addon (OpenThread Border Router add-on via OpenThread adapter in RCP mode) from Home Assistant addons-development repository for testing Thread/Matter support inside Home Assistant:

https://community.home-assistant.io/t/chip-matter-support-including-thread/305633/88

https://github.com/home-assistant/addons-development/tree/master/openthread_border_router

https://github.com/openthread/ot-br-posix/

https://openthread.io/platforms/co-processor

https://github.com/agners

https://community.home-assistant.io/u/agners/summary

The main reason for OpenThread RCP (Radio Co-Processor) support in an "RPC" firmware with OTBR (OpenThread Border Router) support will make ESP32-H2 with its IEEE 802.15.4 radio compatible with the upcoming Thread stack / IEEE 802.15.4 radio based "Matter" (Project CHIP / Connected Home over IP) devices if used in Home Assistant with their other add-ons for the that is also in development. That addon by agners requires that the IEEE 802.15.4 radio have a OpenThread firmware in "RPC" mode instead of the no traditional "NCP" mode of OpenThread stack running on the IEEE 802.15.4 radio:

https://github.com/home-assistant/addons-development/tree/master/chip_controller_repl

https://github.com/home-assistant/addons-development/tree/master/chip_tool

https://github.com/project-chip/connectedhomeip

https://buildwithmatter.com

https://csa-iot.org/all-solutions/matter/

Also having optional OpenThread "NCP" border router firmware would allow users to alternatively use other existing OpenThread applications that use "NCP" mode instead of the newer "RPC" mode which require additional component running on the host. Ex:

https://github.com/openthread/wpantund

Note that as far as I read so far agners has only worked with Silicon Labs based adapter with OpenThread "RPC" firmware for Thread / EEE 802.15.4 radio based Matter (as well as ESP32-C3 based devkit for Matter over WiFi) and that is only because it is a Silabs EFR32MG21 chip based adapter that will ship inside the official Home Assistant Yellow (formerly Home Assistant Amber) hardware as well as the upcoming official Home Assistant SkyConnect USB Stick dongle:

https://github.com/zigpy/zigpy/discussions/894

https://github.com/home-assistant/addons-development/tree/master/silabs-multiprotocol

https://www.home-assistant.io/blog/2021/09/13/home-assistant-yellow/

https://www.crowdsupply.com/nabu-casa/home-assistant-yellow

https://community.home-assistant.io/t/home-assistant-skyconnect-usb-stick-announced-will-be-compatible-with-both-zigbee-and-thread-including-matter-chip-over-thread/433594

closed 2022-09-27 by nagyrobi

Hedda, 2022-09-27:
@nagyrobi Not a duplicate at all because https://github.com/esphome/feature-requests/issues/1430 is only a request for Matter support which can be done over just Wi-Fi, while https://github.com/esphome/feature-requests/issues/1397 is a request for Zigbee and Thread/OpenThread via the 802.15.4 radio on the ESP32-H2 which does not even support WiFi.

southalladam, 2022-10-05:
Matter has gone final, and espressif has released an SDK and programming guide here:
https://docs.espressif.com/projects/esp-matter/en/main/esp32/index.html

Hedda, 2022-10-05:
Yeah, Espressif as a company is now going full steam ahead with the release of an all-in-one Thread/Zigbee to Matter bridge solution (development platform) + support in official ESP-IDF and including many examples for their new ESP32-H2 SoC (ESP32-H2-DevKitC-1 V2.1 board) with 802.15.4 Thread/Zigbee (+ Bluetooth Low Energy).

More interestingly as indirectly related to a concept similar to [Tasmota's Zigbee2Tasmota (Z2T) project](https://tasmota.github.io/docs/Zigbee/) and the ideas behind [ESPHome Bluetooth Proxy](https://esphome.io/components/bluetooth_proxy.html), Espressif has recently posted a lot of news on Espressif's ESP Matter Solution (e.i. "**_esp-matter_**") which uses ESP32 as a Matter-bridge for non-Matter devices such as Zigbee, Thread, and Bluetooth devices, as well as the related news about ESP32-H2 SoC (and ESP32-H2 DevKitC Board) with more news from Espressif about newly established support for Thread/OpenThread (e.i. "**_esp-thread_**") + ZBOSS Open Initiative (ZOI) and Zigbee certification announcements (e.i. "**_esp-zigbee_**").

In essence it seems like Espressif are currently fully focusing on embedded Zigbee gateway and Zigbee/Thread to Matter bridge implementation (with an all-in-one two-SoC gateway/bridge solution combining a ESP32-H2 for Zigbee/Thread with an ESP32-S3 for Wi-Fi). For that Zigbee/Thread/BLE to Matter-bridge solution they now even have two reference hardware designs that I linked below. 

"**_We offer both Matter-Zigbee and Matter-BLE Mesh bridge solutions with full functional software SDK support. A Matter-Zigbee Bridge uses two SoCs (Wi-Fi + 802.15.4), they are connected via a serial interface like UART or SPI, while a Matter-BLE Mesh Bridge can be done on single SoC with both Wi-Fi and BLE interfaces._**"

[image]

[image]

### Espressif ESP Matter news:

https://www.espressif.com/en/news/ESP_Matter_Solution

https://www.espressif.com/en/solutions/device-connectivity/esp-matter-solution

https://www.espressif.com/en/news/ESP32_H2

https://www.espressif.com/en/news/ESP32-H2_Thread_and_Zigbee_certified

https://en.dsr-corporation.com/news/espressif-joins-zoi

### esp-matter (Matter / Project CHIP) solutions:

https://medium.com/the-esp-journal/matter-bridge-for-non-matter-devices-d3b7f003a004

**https://blog.espressif.com/matter-thread-border-router-in-matter-240838dc4779**

https://docs.espressif.com/projects/esp-matter/en/main/esp32/

https://github.com/espressif/esp-matter

https://github.com/espressif/esp-matter/tree/main/examples/zigbee_bridge

https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/esp32

https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool

### esp-thread:

https://openthread.io/guides/border-router/espressif-esp32

https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp

https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli

https://github.com/espressif/esp-thread-lib

https://github.com/espressif/esp-thread-br

https://openthread.io/vendors/espressif

### esp-zigbee:

https://github.com/espressif/esp-zigbee-sdk

https://github.com/espressif/esp-zigbee-sdk/tree/main/components/esp-zigbee-lib

https://github.com/espressif/esp-zboss-lib/

https://components.espressif.com/components/espressif/esp-zboss-lib

https://github.com/espressif/esp-idf/tree/master/examples/zigbee/

https://github.com/espressif/esp-idf/blob/master/examples/zigbee/esp_zigbee_rcp

https://github.com/espressif/esp-idf/tree/master/examples/zigbee/esp_zigbee_gateway

[... 77 lines cut ...]

* **Bi-directional IPv6 Connectivity**: The devices on the backbone link (typically Wi-Fi) and the Thread network can reach each other.
* **Service Discovery Delegate**: The nodes on the Thread network can find the mDNS services on the backbone link.
* **Service Registration Server**: The nodes on the Thread network can register services to the border router for devices on the backbone link to discover.
* **Multicast Forwarding**: The devices joining the same multicast group on the backbone link and the Thread network can be reached with one single multicast.
* **NAT64**: The devices can access the IPv4 Internet via the border router.
* **RCP Update**: The built border router image will contain an updatable RCP image and can automatically update the RCP on version mismatch or RCP failure.
* **Web GUI**: The border router will enable a web server and provide some practical functions including Thread network discovery, network formation and status query. 

In the future releases, the following features will be added:

* SPI Communication
* RF Coexistence

## Post-note

PS: I think esp-zigbee support is of interest for ESP32-based Zigbee remote adapter bridges, like ex. Tube's and ZigStar Gateways:

https://github.com/tube0013/tube_gateways

https://github.com/mercenaruss/zigstar_gateways

https://github.com/codm/cc2652p2-tcp-zigbee

https://github.com/cyijun/OpenZ3Gateway

Hedda, 2022-12-07:
FYI, it looks like Espressif ESP32-C Series SoCs (ESP32-C3 and ESP32-C2) and their official reference implementation based on Espressif’s "ESP-Matter" SDK for Matter have now passed certification for the Matter over Wi-Fi standard according to the relevant certification issued by CSA (Connectivity Standards Alliance).:

https://csa-iot.org/csa_product/esp32-c-series/

https://api.knack.com/v1/applications/54e658034b4f44e42fb18201/download/asset/635b9bc49c2eb30024590776/picsesp32.tar.gz

https://www.espressif.com/en/news/Matter-compatible_Device_Certificate_Generation_and_Pre-provisioning_Services

"Further more, ESP-ZeroCode Modules will provide a ready-made, plug-and-play solution for building common lighting and electrical devices, such as LED lights, outlets, switches, dimmers, relays, and fans."

https://www.espressif.com/en/solutions/device-connectivity/esp-matter-solution

Hedda, 2023-02-18:
**ESP32-C6** SoC, module and development board has now been launched with WiFi 6, BLE 5.0, and 802.15.4, ink. OpenThread and Zigbee!    ...so think these could be perfect for ESPHome as either a Thread Board Router (for Matter) or as a Zigbee Router device.

https://www.espressif.com.cn/en/news/ESP32-C6_Available

https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/index.html

https://github.com/espressif/esp-idf/issues/10423

Espressif launched the ESP32-C6 WiFi 6, Bluetooth 5.0 LE, and 802.15.4 IoT microcontroller by making the ESP32-C6-WROOM-1 module:

https://www.cnx-software.com/2021/04/11/esp32-c6-wifi-6-and-bluetooth-le-risc-v-soc-for-iot-devices/

[image]

[image]

_Espressif Systems has finally launched the ESP32-C6 WiFi 6, Bluetooth 5.0 LE, and 802.15.4 IoT microcontroller by making the ESP32-C6-WROOM-1 module available on [Aliexpress for $14.25](https://fave.co/3VYE3RW) for a pack of 5 modules as well as the ESP32-C6-DevKitC-1 development board that’s [out of stock on the official store](https://fave.co/3vU14ea) at the time of writing. but somehow listed in another store [for $16.22 plus shipping](https://s.click.aliexpress.com/e/_DlvBgFz)._

_Besides the availability announcement, another surprise is the 802.15.4 radio for Zigbee and Thread that was not part of the original [ESP32-C6 announcement](https://www.cnx-software.com/2021/04/11/esp32-c6-wifi-6-and-bluetooth-le-risc-v-soc-for-iot-devices/) in April 2021. The ESP32-C6-DevKitC-1 development board comes with a module with 8MB flash, exposes most of the I/O’s from the ESP32-C3, and comes with two USB-C ports, two buttons, and an RGB LED._

ESP32-C6-DevKitC-1 development board specifications:

- Wireless module  – ESP32-C6-WROOM-1 with
  - ESP32-C6 single-core 32-bit RISC-V clocked up to 160 MHz 320KB ROM, 512KB SRAM, low-power RISC-V core @ up to 20 MHz
  - Wireless – 2.4 GHz WiFi 6, Bluetooth 5.0 LE, 802.15.4 radio for Zigbee 3.0 and Thread. Matter compatible. PCB antenna
8MB SPI flash
- USB – 1x USB Type-C host port, 1x USB Type-C port via UART console
- I/Os – 2x 16-pin headers with GPIO, UART, SPI, ADC, SDIO, USB, I2S, etc…
- Misc – Reset and BOOT buttons, RGB LED, J5 header/jumper for current measurement
- Power Supply (mutually exclusive options)
  - 5V via USB-UART port (recommended)
  - 5V and GND pin headers
  - 3V3 and GND pin headers
- Dimensions – 51.8 x ~20 mm

[image]

[image]

_The new ESP32-C6 module and development board work with the ESP-IDF v5.1, currently under development (see [progress report](https://github.com/espressif/esp-idf/issues/10423)), and some [basic documentation](https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/index.html) can be found on read the docs._

_If you can’t purchase an ESP32-C6 module or board just yet due to limited supplies, don’t despair as more are coming very soon as [Espressif tells](https://www.espressif.com.cn/en/news/ESP32-C6_Available?position=0&list=WQQQ-4A66v8oesBoFKp-rNz9bSQ9dg5w7wc_nNvDt_I) us to expect more products on Adafruit, Akizuki, Digikey, and Mouser._

Hedda, 2023-03-27:
Tip if early adopters don't want to wait for ESPHome Matter support is to check out Tasmota's initial experimental development:

https://github.com/arendst/Tasmota/discussions/17872

Another alternative could be to run the node-matter project on ESP32 (FYI, the node-matter project has recently been merged and moved upstream to become a part of the official project-chip/matter.js project and the code for node-matter will very soon fully be moved to matter.js/packages/matter-node.js)

https://github.com/mfucci/node-matter/discussions/272

https://github.com/project-chip/matter.js/tree/main/packages/matter-node.js

https://github.com/project-chip/matter.js/blob/main/README.md#matterjs-usage

https://github.com/mfucci/node-matter/pull/273

Hedda, 2023-06-21:
Any news about Matter support over WiFi for ESPHome on ESP32 (for at least feature-parity of [the previous Matter workshop](https://www.home-assistant.io/blog/2022/05/29/matter-in-home-assistant-workshop-announcement/)?)?

Matter 1.1 specification/SDK been released bringing enhancements for developers (but no features from end-users point-of-view):

https://csa-iot.org/newsroom/matter-1-1-release-enhancements-for-developers-and-devices/

Off-topic but great to read about the development of the [Matter (BETA) integration for Home Assistant](https://www.home-assistant.io/integrations/matter) keep moving along nicely:

https://building.open-home.io/success-of-the-open-home/

https://staceyoniot.com/hands-on-with-the-matter-beta-for-home-assistant/

https://www.theverge.com/23641940/home-assistant-skyconnect-thread-matter-zigbee-smart-home

Also interesting from an early-adoptors point-of-view is that the CSA has certified 1000+ Matter devices/products and seen 60+ new members joining the Connectivity Standards Alliance since the initial release of the Matter 1.0 specification in October 2022.

https://www.theverge.com/23568091/matter-compatible-devices-accessories-apple-amazon-google-samsung

https://csa-iot.org/csa-iot_products/?p_keywords=&p_type%5B%5D=14&p_program_type%5B%5D=1049&p_certificate=&p_family=

Hedda, 2023-07-01:
FYI, Espressif ESP-IDF v5.1 with ESP32-C6 and ESP32-H2 has now been released upstream:

https://github.com/espressif/esp-idf/releases/tag/v5.1

DoomHammer, 2023-08-17:
There's this project that might be helpful: https://github.com/Yacubane/esp32-arduino-matter

mvgijssel, 2023-11-30:
Sonoff managed to pull it off in their new minir4m! https://sonoff.tech/product/diy-smart-switches/minir4m/

reinisb, 2023-11-30:
> Sonoff managed to pull it off in their new minir4m! https://sonoff.tech/product/diy-smart-switches/minir4m/

I have it and it works great. Though I'd prefer Thread over WiFi.

Hedda, 2023-12-06:
@jesserockz can you tell if anyone at Nabu Casa has begun working on or planning to add Matter support over WiFi in ESPHome?

Hedda, 2023-12-08:
> https://nabucasa.github.io/matter-example-apps/
> 
> https://github.com/NabuCasa/matter-example-apps

FYI, other manufacturers including Espressif have updated their Matter examples

https://github.com/espressif/esp-matter/tree/main/examples

Silicon Labs even has a few EFR32 examples specific to the Thread protocol so wonder if may be portable to ESPHome on ESP32?

https://github.com/SiliconLabs/matter_applications

darkxst, 2023-12-08:
> Espressif have kept updating their Matter examples, check out for example:
> 
> https://github.com/espressif/esp-matter/tree/main/examples

Espressif maintain two different sets of example apps. The ones linked above are using the esp-matter SDK. The ones in the CHIP repo are pure esp-idf SDK only.

For example
https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32

omaramin-2000, 2024-05-31:
Finally, I have a plan working on this based on the matter SDK https://www.espressif.com/en/solutions/device-connectivity/esp-matter-solution#sdk-for-matter

```yaml
esphome:
  name: matter_smart_device
  platform: ESP32
  board: esp32dev

# Enable Wifi
wifi:
  networks:
  - ssid: '!secret wifi_ssid'
    password: '!secret wifi_password'

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: matter-device
    ap_timeout: 1min
captive_portal:

# Example configuration entry for enabling Matter
matter:
  project_id: "example-project-id"
  node_id: "example-node-id"
  fabric_id: "example-fabric-id"
  device_type: "example-device-type"
  # Additional Matter configuration options...

# Enable Home Assistant API
api:

logger:

ota:

# Enable the web server to display the Matter pairing QR code
web_server:
  port: 80
  version: 2
  local: true

# Custom sensor to expose the Matter pairing code to Home Assistant
sensor:
  - platform: custom
    lambda: |-
      auto matter_pairing_code = new MatterPairingCodeSensor();
      App.register_component(matter_pairing_code);
      return {matter_pairing_code};
    sensors:
      name: "Matter Pairing Code"

# Custom text sensor to expose the Matter QR code to Home Assistant
text_sensor:
  - platform: custom
    lambda: |-
      auto matter_qr_code = new MatterQRCodeSensor();
      App.register_component(matter_qr_code);
      return {matter_qr_code};
    text_sensors:
      name: "Matter QR Code"
```

brotherdust, 2024-06-16:
> Finally, I have a plan working on this based on the matter SDK https://www.espressif.com/en/solutions/device-connectivity/esp-matter-solution#sdk-for-matter
>
>
>
> ```yaml
>
> esphome:
>
>   name: [...]

This is exciting! Do you have a repo up for it?

Hedda, 2024-06-17:
> Finally, I have a plan working on this based on the matter SDK https://www.espressif.com/en/solutions/device-connectivity/esp-matter-solution#sdk-for-matter

@omaramin-2000 if your solution depend on Espressif’s Matter SDK then does it then also delend on ESP-IDF?

 If so should it be asssumed that it will only work on the ESP32 platform or should it still also work on other platforms for ESPHome so that it can be used on a SoC not made by Espressif as well?

That is, I believe that all ESPHome platforms supported uses the Arduino framework by default, (not ESP-IDF framework by default though that is supported as an alternative base framework for ESP32 chips, however ESP-IDF is for example not used or supported used for ESP8266 platform, RP2040 platform, or LibreTiny platform). 

https://esphome.io/components/esp32.html

https://esphome.io/components/esp8266.html

https://esphome.io/components/rp2040.html

https://esphome.io/components/libretiny.html

FYI, there is also an open pull request for a nRF52 / nRF53 platform (Nordic Semiconductor nRF5 familiy of SoCs):

https://github.com/esphome/esphome/pull/6075

omaramin-2000, 2024-06-17:
This is just a conceptual example of how we might enable Matter on ESPHome using YAML configuration. actually, matter works only on ESP32 directly except when connecting it with other devices remotely using HTTP requests so that these non-matter devices can be converted to matter. However, for using the Arduino framework, this is the Matter IoT protocol library for ESP32 working on the Arduino framework, that can help with implementing matter on esphome as in TASMOTA as well: https://github.com/Yacubane/esp32-arduino-matter

omaramin-2000, 2024-06-17:
> @omaramin-2000 if your solution depend on Espressif’s Matter SDK then does it then also delend on ESP-IDF?
>
>  If so should it be asssumed that it will only work on the ESP32 platform or should it [...]

This is just a conceptual example of how we might enable Matter on ESPHome using YAML configuration. Matter works only on ESP32 directly except when connecting it with other devices remotely using HTTP requests so that these non-matter devices can be converted to matter. However, for using the Arduino framework, this is the Matter IoT protocol library for ESP32 working on the Arduino framework, which can help with implementing matter on esphome as in TASMOTA as well: https://github.com/Yacubane/esp32-arduino-matter

andyearnshaw, 2024-06-18:
> ```yaml
> # Custom sensor to expose the Matter pairing code to Home Assistant
> sensor:
>   - platform: custom
>     lambda: |-
>       auto matter_pairing_code = new MatterPairingCodeSensor();
>       App.register_component(matter_pairing_code);
>       return {matter_pairing_code};
>     sensors:
>       name: "Matter Pairing Code"
> ```

I don't think a dynamically created pairing code is ideal. With this approach, you'd need to pair with Home Assistant over WiFi in order to find the Matter pairing code and it's not forwards-compatible with a potential future implementation of Thread-only connections. It also seems like the code would change between device firmware updates, which is less than ideal. I think the pairing code would make more sense as part of the `matter` configuration options:

```yaml
# Example configuration entry for enabling Matter
matter:
  project_id: "example-project-id"
  node_id: "example-node-id"
  fabric_id: "example-fabric-id"
  device_type: "example-device-type"
  pairing_code: ABCD-1234-XYZ
  # Additional Matter configuration options...
```

This way, you could print a label and stick it to the device if you wanted, or view the QR code in the ESPHome web interface (or maybe even the terminal), as well as (potentially) having Home Assistant display it in the device information.

omaramin-2000, 2024-06-18:
> This is exciting! Do you have a repo up for it?

still exploring...

stefa168, 2024-06-18:
> I don't think a dynamically created pairing code is ideal

I second this.

dominik4545, 2024-07-18:
> This way, you could print a label and stick it to the device if you wanted, or view the QR code in the ESPHome web interface (or maybe even the terminal), as well as (potentially) having Home Assistant display it in the device information.

Would it be an option to use the MAC-address of the device (as default)? It's unique anyway.

graphichasan, 2024-09-07:
> Would it be an option to use the MAC-address of the device (as default)? It's unique anyway.

I don't think MAC-address would be a good idea ... `# Example configuration entry for enabling Matter
matter:
  project_id: "example-project-id"
  node_id: "example-node-id"
  fabric_id: "example-fabric-id"
  device_type: "example-device-type"
  pairing_code: ABCD-1234-XYZ` looks to be mutch simpler & discriptive ...

kquinsland, 2024-09-07:
> Would it be an option to use the MAC-address of the device (as default)? It's unique anyway.

If an attacker only needs knowledge of the MAC to determine the pairing code of a device, you could have quite a few issues depending on how easy it is to trigger a (re-)pair of a device.

Ideally the pairing code would be something that is random and unique per device and very hard to change without user consent.

RebbePod, 2025-02-11:
Any update on this?

nicx, 2025-05-05:
waiting for matter support too... is this somehow in development?

RebbePod, 2025-05-05:
I'm with you :)

It will be a big game changer for esphome when zigbee and especially matter gets supported. It will allow for mass adoption of esphome allowing for people to make custom devices for their not so tech knowledge friends. Bringing the device straight into their Google or Alexa!

stefa168, 2025-05-05:
> It will be a big game changer for esphome when zigbee and especially matter gets supported.

There are plans for ESPHome on Zigbee?

RebbePod, 2025-05-05:
> > It will be a big game changer for esphome when zigbee and especially matter gets supported.
> 
> There are plans for ESPHome on Zigbee?

I don't think anything official. We have this:
https://github.com/luar123/zigbee_esphome

nicx, 2025-05-05:
> Theoretically, Matter can be run on ESP32 using esphome over WiFi/Ethernet. Based on the default Arduino framework, it should use this library: https://github.com/espressif/arduino-esp32/tree/master/libraries/Matter So, we need to include another option when using the esp-idf framework it runs matter using this way: https://www.espressif.com/en/solutions/device-connectivity/esp-matter-solution#sdk-for-matter https://github.com/espressif/esp-matter This will be configured from the YAML as I provided in my pre-example. But, till now, I still haven't tried working on it yet.
> 
> https://esphome.io/components/esp32.html#esp-idf-framework

hopefully you will find the time to try it out, as I am unfortunately no developer :)

Hedda, 2025-05-05:
> > There are plans for ESPHome on Zigbee?
> 
> I don't think anything official. We have this: https://github.com/luar123/zigbee_esphome
> 
> With the esp32c6 it would be really nice to have.

Zigbee is off-topic in this specific feature request so please instead see this other request about that to not derail this:

- https://github.com/esphome/feature-requests/issues/1397

omaramin-2000, 2025-05-12:
I tried including the matter library in Arduino framework as an external component which requires the __init__.py file to work as an ESPHome component:

```yaml
external_components:
  - source:
      type: local
      path: arduino-esp32/libraries/ # Root of the cloned repo
    components: ["Matter"]

esphome:
  name: matter-device
  friendly_name: Matter Device

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  - platform: esphome

wifi:
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Matter-Device"

captive_portal:

# Matter component configuration
src:
  id: matter_device

switch:
  - platform: gpio
    name: "Relay"
    pin: GPIO2
    id: relay
    on_turn_on:
      then:
        - matter.generic_switch.click: matter_device
    on_turn_off:
      then:
        - matter.generic_switch.click: matter_device

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode:
        input: true
        pullup: true
      inverted: true
    name: "Onboard Button"
    on_multi_click:
      # Single press
      - timing:
          - ON for at most 0.5s
          - OFF for at least 0.5s
        then:
          - switch.toggle: relay
      # Double press
      - timing:
          - ON for at most 0.5s
          - OFF for at most 0.3s
          - ON for at most 0.5s
          - OFF for at least 0.2s
        then:
          # Add your action for double press here
          - logger.log: "Double press detected"
          # Long press
      - timing:
          - ON for at least 10s
        then:
          - button.press: factory_reset_button

button:
  - platform: factory_reset
    name: "Factory Reset Button"
    id: factory_reset_button

# Enable Web Server
web_server:
  port: 80
  version: 2
  local: true

# Add Matter text sensors
text_sensor:
  - platform: template
    name: "Matter Manual Pairing Code"
    id: matter_manual_code
    icon: "mdi:qrcode"
    update_interval: 60s
    lambda: |-
      return matter_device.getManualPairingCode();

  - platform: template
    name: "Matter QR Code URL"
    id: matter_qr_url
    icon: "mdi:qrcode-scan"
    update_interval: 60s
    lambda: |-
      return matter_device.getOnboardingQRCodeUrl();
```

The __init__.py file:

```python
import esphome.codegen as cg
import esphome.config_validation as cv
from esphome.const import CONF_ID
from esphome import automation

DEPENDENCIES = ['esp32']
AUTO_LOAD = ['switch']

CONF_MATTER_ID = "matter_id"

# Define namespaces and components
matter_ns = cg.esphome_ns.namespace('matter')
MatterComponent = matter_ns.class_('ArduinoMatter', cg.Component)
MatterGenericSwitch = matter_ns.class_('MatterGenericSwitch', cg.Component)

# Component schema
CONFIG_SCHEMA = cv.Schema({
    cv.GenerateID(): cv.declare_id(MatterComponent),
}).extend(cv.COMPONENT_SCHEMA)

# Define action schema
GENERIC_SWITCH_CLICK_SCHEMA = cv.Schema({
    cv.Required(CONF_MATTER_ID): cv.use_id(MatterComponent)
})

# Register the action
@automation.register_action(
    'matter.generic_switch.click',
    MatterGenericSwitch,
    GENERIC_SWITCH_CLICK_SCHEMA
)
async def matter_generic_switch_click_to_code(config, action_id, template_arg, args):
    var = cg.new_Pvariable(action_id, template_arg)
    await cg.register_parented(var, config[CONF_MATTER_ID])
    return var

async def to_code(config):
    var = cg.new_Pvariable(config[CONF_ID])
    await cg.register_component(var, config)
    
    # Add Arduino ESP32 Matter library paths
    cg.add_build_flag('-I"/config/arduino-esp32/libraries/Matter/src"')
    cg.add_build_flag('-I"/config/arduino-esp32/libraries/Matter/src/MatterEndpoints"')
    
    # Add Arduino framework flags
    cg.add_build_flag('-DARDUINO_ARCH_ESP32=1')
    cg.add_build_flag('-DESP32')
    
    # Create Matter header file
    cg.add_global(cg.RawStatement('''
#include <esphome/core/component.h>
#include <esphome/core/automation.h>
#include <esphome/components/switch/switch.h>

namespace esphome {
namespace matter {

// Forward declare Matter base class
class Matter {
public:
    void begin() {}
    void loop() {}
    class Endpoint {
    public:
        void setState(bool state) {}
    };
    Endpoint* createEndpoint() { return new Endpoint(); }
};

// Matter component class
class ArduinoMatter : public Component {
private:
    Matter* matter_{nullptr};
public:
    ArduinoMatter() = default;
    void setup() override {
        matter_ = new Matter();
        matter_->begin();
    }
    void loop() override {
        if (matter_) matter_->loop();
    }
    Matter* get_matter() { return matter_; }
};

// Matter switch class template
template<typename... Ts>
class MatterGenericSwitch : public Component, public switch_::Switch {
private:
    ArduinoMatter* parent_{nullptr};
    Matter::Endpoint* endpoint_{nullptr};
public:
    void set_parent(ArduinoMatter* parent) { parent_ = parent; }
    void setup() override {
        if (parent_ && parent_->get_matter())
            endpoint_ = parent_->get_matter()->createEndpoint();
    }
    void write_state(bool state) override {
        if (endpoint_) endpoint_->setState(state);
        this->publish_state(state);
    }
    void click() { this->write_state(!this->state); }
};

} // namespace matter

// Global type definitions
inline namespace matter {
    using ArduinoMatter = matter::ArduinoMatter;
    using MatterGenericSwitch = matter::MatterGenericSwitch<>;
}

} // namespace esphome
'''))
```

But, it gives errors and fail to compile... so, maybe anyone can try to investigate and find out: https://github.com/espressif/arduino-esp32/tree/master/libraries/Matter/src

Gunni, 2025-11-03:
No, thread is the communications protocol which carries an application protocol called matter. They are distinct and separate things and we would like matter support directly over either WI-Fi or thread.

Hedda, 2025-11-03:
> [OpenThread](https://esphome.io/components/openthread/) support has been added. If I'm not mistaken this solves this ticket, right?

No, please read and look at the pictures of architecture layer diagrams/schematics in the first post. This is is a request for Matter application layer implementation, (that is the core IoT application functionality of the Matter standard, not which network layer it uses for transport).

Thread (OpenThread) is just one of the transport layers that Matter can make use use. Because it mainly requires IPv6 support from the transport layer it can also make use of WiFi and Ethernet (wired LAN network) as a transport medium too.

oerix, 2026-02-01:
Are there any developments regarding this?