esphome/feature-requests #1433: xiaomi_lywsd03mmc doesn't keep the hassio's entities updated

amaciuc, 2021-09-27:
**Describe the problem you have/What new integration you would like**

From Esphome's logs I notice that the values of sensors are updated, but the entities in hassio are out of date and they are updated only if the value is changing.

Is there a chance to keep the entities updated every time the Esphome got the values of sensors ?

Here you can find some captures with an example:

[image]
[image]

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

dhutuleac, 2021-09-27:
I think that is useful to know when the current state of the entity was checked.
I use [this device](https://esphome.io/components/sensor/xiaomi_ble.html?highlight=lywsd03mmc&fbclid=IwAR3fImC13m_w-kS3FUW6lU5xBAGpDVD7TLbISBWSJlYeGFrt_8PT4NZRFCE#lywsd03mmc) and I want to see when the sensor was read.
Probably, there are users that don't want to change the current comportment, so can you implement an attribute `keep_sync: true` for the users who want to see the timestamps "Last updated" with the last value read?

OttoWinter, 2021-10-03:
That is due to HA (if I understand correctly)

HA's state machine doesn't update the state when it's equal to the one before.

See `force_update` to change this behavior.

nagyrobi, 2022-07-06:
`force_update` is the way to fix this indeed.

closed 2022-07-06 by nagyrobi