esphome/feature-requests #1437: Log level line tag fixes - INFO=slow & default, DEBUG=repetitive

defl, 2021-09-28:
**Describe the problem you have/What new integration you would like**

_tl;dr: Update log line tagging so that we get more non-high-speed-repetitive lines on INFO and make this the default_

By default ESPHome uses DEBUG log levels. If you want less logging and decrease this to the next available level of granularity (INFO) logging drops to being useless. The reason why I would like more finegrained control is that I've found that reducing UART output has beneficial effects to responsiveness. On 2021.09 with the following settings I have perfect continuous uptime. 

```
logger:
  level: INFO   # Less logs to reduce load
  baud_rate: 0  # Disable serial logging via UART, reduces load
```

Increasing of the baud_rate leads to detectable packetloss. In similar vein, I don't need every sensor displayed every second (DSMR w 5.0 hardware for example) leading to continuous 20 lines/sec log output which serves no purpose. All I want is simply to know the device has started, has wifi, has sensors and that the sensors output something sane (once). All the other info I'll get through looking at the sensor values in the consumer of the data (home assistant for example).

**Proposal:**
- INFO shows startup lines, wifi (ex anything that's >1x second), detected sensors and if possible first sensor value
- DEBUG shows individual measurements by sensors

**Please describe your use case for this integration and alternatives you've tried:**
No alternatives exist

**Additional context**
None

OttoWinter, 2021-10-03:
Why not just reduce the offending components log level with the `logger->logs` option?

defl, 2021-10-30:
@OttoWinter thanks for the tip, I've tried this to see how far I could get with it. The approach I took was a blacklist one, eg.:

```
logger:
  baud_rate: 0  # Disable serial logging via UART, reduces load
  logs:  # Reduce logging load
    dallas.sensor: INFO
    sensor: INFO
    adc: INFO
```

This is obviously better than to eliminate nearly all logging by setting the global level to INFO, so in that sense it's a win. 

However it does not fix the points raised above. Specifically there is no way to set logging to see sensor startup (for example Dallas addresses or ADC) + config and not the data.