esphome/feature-requests #1438: Support for SCD40 CO2 sensor

comatose-tortoise, 2021-09-28:
**Describe the problem you have/What new integration you would like**

I would like to add support for the Sensirion SCD40/41 CO2 sensor.
**Please describe your use case for this integration and alternatives you've tried:**

This is the next generation of CO2 sensors from Sensirion, much smaller and better specifications than the SCD30 that is currently supported in esphome.
**Additional context**

All relevant data for integrating this sensor in esphome should be available at Sensirion's website: https://www.sensirion.com/en/environmental-sensors/evaluation-kit-sek-environmental-sensing/evaluation-kit-sek-scd41/

martgras, 2021-09-30:
https://github.com/esphome/esphome/pull/2217 was merged recently to the dev branch

OttoWinter, 2021-10-03:
closing per https://github.com/esphome/esphome/pull/2217

closed 2021-10-03 by OttoWinter