esphome/feature-requests #1440: Color wheel offset to mach real light color with set color.

jaddel, 2021-09-28:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

OttoWinter, 2021-10-03:
Well I think the first thing that you'd need to do is figure out _how_ your light differs from RGB. Like what sort of shift are we talking about here? It needs a model to explain what sort of shift you mean

> I tried reducing the output off some channels but it only seems to limit the overall brightness when switching fully to the limited color.

No it changes the max power for each channel individually, and in all cases, not just when it's just that one color displayed; effectively the channel output gets multiplied by this percentage.

jaddel, 2021-10-08:
I hope this will explain well enough what I think would be useful. 

How ever the coordinates on the color wheel are processed and then handed over to the esphome api. I would like to be able to set filters for each variable. 
Those filters could rearrange / remap the scale the variable applies to or set a factor to change it. 

This would allow us to fine tune the colors and e.g to remap the brightness scale.

closed 2025-04-04 by 10bn