esphome/feature-requests #1441: Prana Recuperator support

VladNord, 2021-09-29:
**Describe the problem you have/What new integration you would like**

Add Prana recuperator control over Bluetooth. 
Devices supports control via a regular IR remote control and via bluetooth using a smartphone application

[image]

**Please describe your use case for this integration and alternatives you've tried:**

On github found two integrations for these devices, but the main problem is that the controller (raspberry) must be located near the recuperator itself, which is impractically.
this https://github.com/alextud/ha-prana
and this https://github.com/corvis/prana_rc

In second case it can be installed to another linux machine with bluetooth and connect it over the network to HA, but this is also not practical

**Additional context**

Thanks in advance, I think this integration will be useful to the community.

CagunA, 2022-05-28:
Would be very nice to have this feature!

zagi988, 2024-03-06:
My friend and I are working on "integration" for ESPHome which involves changing the ESP32 to another one, flashed with ESPHome and full configuration for Prana to work locally.

mario-tux, 2024-03-06:
> My friend and I are working on "integration" for ESPHome which involves changing the ESP32 to another one, flashed with ESPHome and full configuration for Prana to work locally.

This is very interesting: if you need a tester, I'm available to use your component. I got two recent units: Prana 210C ErP Pro and Prana 160 ErP Pro (2022 models with Bluetooth and WiFi).

VladNord, 2024-03-07:
> e which involves changing the ESP32 to another one, flashed with ESPHome and full configuration for Prana to work locally.

Interesting too.
Do you mean replacing the built-in chip with another one with ESPHOME firmware?

mario-tux, 2024-03-07:
> Yes, desolder the ESP32 that's on it and solder another ESP32 which then gets flashed with ESPHome.

Oh... I didn't catch your idea to locally manage the VMC unit... it is quiet invasive... :-)

I was supposing you were using an esphome component to simulate a bluetooth controller. Such approach, for example, is use the Daikin Madoka controller that was recently submitted for inclusion (https://github.com/esphome/esphome/pull/5951). I'm using it with profits.

VladNord, 2024-03-07:
> I don't know, open it up and show us. 😁
Chip is CY8C4127LQIBL483 :(

[image]
[image]
[image]

nagyrobi, 2024-03-07:
OK, we need a list of devices having ESP32 inside

VladNord, 2024-03-07:
> OK, we need a list of devices having ESP32 inside

Definitely Prana 150 Apr2020  revision doesnt (I suspect that all devices which do not support Wi-Fi too)

Add to the incompatible list :)

voed, 2024-10-23:
I have made a component for it to contol via Bluetooth. Feedback in issues is appreciated
https://github.com/voed/esphome_prana_ble