esphome/feature-requests #1444: WireGuard Support for ESP32

davey, 2021-10-01:
**Describe the problem you have/What new integration you would like**

Please help me adding support for [WireGuard](https://www.wireguard.com/) to ESPHome on ESP32 boards. 
WireGuard is an extremely simple yet fast and modern VP.

There already seems to be an [Arduino library](https://github.com/ciniml/WireGuard-ESP32-Arduino) in [Platformio](https://platformio.org/lib/show/12571/WireGuard-ESP32) for the ESP32.

**Please describe your use case for this integration and alternatives you've tried:**

I need to have a bidirectional, secure channel between multiple ESPHome/ESP32s at multiple locations and some remote backend server. 
From these backend servers I want to be able to push e.g. OTA updates centrally to many different NATed locations (where a site2site VPN is not an option). 
Also I'd like to reach all those different locations from one central HomeAssistant instance. 
On the ESP side, only traffic to a specific remote subnet (the backend) needs be routed through Wireguard, all remaining network traffic should use the default gateway. 

AFAIK there are no VPN alternatives available for ESPhome yet.

**Additional context**

Basically this is all about being able to directly reach (on IP level) multiple ESPHome boards at different locations/networks behind their NAT routers from some central place / VPN server / backend.

davey, 2021-10-01:
I've already tried to add the library by putting this in my ESPHome yaml config:
```platformio_options: { lib_deps: "ciniml/WireGuard-ESP32" }```
but I got this error:
```
Processing my_test (board: esp32dev; framework: arduino; platform: platformio/espressif32@3.2.0)
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Dependency Graph
|-- <WireGuard-ESP32> 0.1.2
[......]
Compiling .pioenvs/my_test/libc96/WireGuard-ESP32/crypto/cortex/cortex_m0_mpy121666.s.o
xtensa-esp32-elf-as: unrecognized option '-x'
Compiling .pioenvs/my_test/libc96/WireGuard-ESP32/crypto/cortex/cortex_m0_reduce25519.s.o
xtensa-esp32-elf-as: unrecognized option '-x'
Compiling .pioenvs/my_test/libc96/WireGuard-ESP32/crypto/cortex/mul.s.o
xtensa-esp32-elf-as: unrecognized option '-x'
*** [.pioenvs/my_test/libc96/WireGuard-ESP32/crypto/cortex/cortex_m0_mpy121666.s.o] Error 1
*** [.pioenvs/my_test/libc96/WireGuard-ESP32/crypto/cortex/cortex_m0_reduce25519.s.o] Error 1
*** [.pioenvs/my_test/libc96/WireGuard-ESP32/crypto/cortex/mul.s.o] Error 1
Compiling .pioenvs/my_test/libc96/WireGuard-ESP32/crypto/cortex/scalarmult.c.o
================================================ [FAILED] Took 7.92 seconds ================================================
```

Any help / hint is much appreciated!

mrkeuz, 2021-10-01:
@davey Very cool solution. It is potentially will add ability to communicate with `Esphome` from anywhere. Really impressive. I like idea, maybe it is even good alternative over MQTT. 

I know one of embedded developer. Will try to consult with him. Also, will try to google problem to help you.

mrkeuz, 2021-10-02:
Faced with same issue. I've even created fresh project to test. Same result. `xtensa-esp32-elf-as` assembler tool chain (in contrast GNU Assembler) has not an option "-x" (if I understand things correctly) 

What I found and tried, options: 
- Rename `*.s` ⇾ `*.S` sources as recommended somewhere on `PlatformIO` forum. But also not works, errors same as logs below.
  https://community.platformio.org/t/unknown-assembler-option-when-compiling-main-cpp-and-asm/968/6

- Remove invalid parameters from assembler options from `ASFLAGS` (solution and logs see below). 
  It helps to step forward, but does not help. I got error sheet from ASM compiler. Seems it need somehow configure assembler-compiler. I'm not fluency embedded developer, so don't know how to prepare it properly. 

  - Links:
    - Similar problem with `-x` flag
      https://github.com/platformio/platformio-core/issues/3513
    - `-Wa,option` notation, with this option we can remove flags
      https://gcc.gnu.org/onlinedocs/gcc/Assembler-Options.html
    - Xtensa `-mlongcalls` option 
      https://gcc.gnu.org/onlinedocs/gcc/Xtensa-Options.html

- Also, I asked question to author, maybe he helps:
  https://github.com/ciniml/WireGuard-ESP32-Arduino/issues/2

**Summary:** It is necessary to understand, how configure `PlatformIO` assembler tool chain for compile right in way.

Hope it helps somehow.

`platformio.ini`:
```ini
...
; Removes paramaters from ASFLAGS
build_unflags =
   -Wa,-x
   -Wa,assembler-with-cpp
   -Wa,-mlongcalls
...
```

Error from assembler after remove unsupported flags:

```console
...
.pio/libdeps/esp-wrover-kit/WireGuard-ESP32/src/crypto/cortex/cortex_m0_mpy121666.S:14: Error: unknown pseudo-op: `.cpu'
.pio/libdeps/esp-wrover-kit/WireGuard-ESP32/src/crypto/cortex/cortex_m0_mpy121666.S:15: Error: unknown pseudo-op: `.fpu'
.pio/libdeps/esp-wrover-kit/WireGuard-ESP32/src/crypto/cortex/cortex_m0_mpy121666.S:16: Error: unknown pseudo-op: `.eabi_attribute'
.pio/libdeps/esp-wrover-kit/WireGuard-ESP32/src/crypto/cortex/cortex_m0_mpy121666.S:17: Error: unknown pseudo-op: `.eabi_attribute'
.pio/libdeps/esp-wrover-kit/WireGuard-ESP32/src/crypto/cortex/cortex_m0_mpy121666.S:18: Error: unknown pseudo-op: `.eabi_attribute'
.pio/libdeps/esp-wrover-kit/WireGuard-ESP32/src/crypto/cortex/cortex_m0_mpy121666.S:19: Error: unknown pseudo-op: `.eabi_attribute'
.pio/libdeps/esp-wrover-kit/WireGuard-ESP32/src/crypto/cortex/cortex_m0_mpy121666.S:20: Error: unknown pseudo-op: `.eabi_attribute'
.pio/libdeps/esp-wrover-kit/WireGuard-ESP32/src/crypto/cortex/cortex_m0_mpy121666.S:21: Error: unknown pseudo-op: `.eabi_attribute'
.pio/libdeps/esp-wrover-kit/WireGuard-ESP32/src/crypto/cortex/cortex_m0_mpy121666.S:22: Error: unknown pseudo-op: `.eabi_attribute'
.pio/libdeps/esp-wrover-kit/WireGuard-ESP32/src/crypto/cortex/cortex_m0_mpy121666.S:23: Error: unknown pseudo-op: `.eabi_attribute'
.pio/libdeps/esp-wrover-kit/WireGuard-ESP32/src/crypto/cortex/cortex_m0_mpy121666.S:24: Error: unknown pseudo-op: `.eabi_attribute'
.pio/libdeps/esp-wrover-kit/WireGuard-ESP32/src/crypto/cortex/cortex_m0_mpy121666.S:25: Error: unknown pseudo-op: `.code'
cortex_m0_reduce25519.S:33: Error: unknown pseudo-op: `.code'
cortex_m0_reduce25519.S:34: Error: unknown pseudo-op: `.thumb_func'
cortex_m0_reduce25519.S:38: Error: unknown opcode or format name 'push'
cortex_m0_reduce25519.S:39: Error: unknown opcode or format name 'ldr'
cortex_m0_reduce25519.S:40: Error: extra comma
cortex_m0_reduce25519.S:40: Error: syntax error
cortex_m0_reduce25519.S:41: Error: extra comma
cortex_m0_reduce25519.S:41: Error: syntax error
cortex_m0_reduce25519.S:42: Error: extra comma
cortex_m0_reduce25519.S:42: Error: syntax error
cortex_m0_reduce25519.S:43: Error: extra comma
cortex_m0_reduce25519.S:43: Error: syntax error
cortex_m0_reduce25519.S:44: Error: unknown opcode or format name 'uxth'
cortex_m0_reduce25519.S:45: Error: unknown opcode or format name 'mul'
cortex_m0_reduce25519.S:46: Error: unknown opcode or format name 'mul'
cortex_m0_reduce25519.S:47: Error: bad register name: r5
cortex_m0_reduce25519.S:47: Error: junk at end of line, first unrecognized character is `r'
cortex_m0_reduce25519.S:47: Internal error!
Assertion failure in ignore_rest_of_line at /builds/idf/crosstool-NG/.build/src/binutils-2.25.1/gas/read.c line 3694.
[... 17 lines cut ...]
home/petr/.platformio/cortex_m0_reduce25519.S:35: Error: unknown opcode or format name 'push'
cortex_m0_reduce25519.S:36: Error: extra comma
cortex_m0_reduce25519.S:36: Error: syntax error
cortex_m0_reduce25519.S:37: Error: extra comma
cortex_m0_reduce25519.S:37: Error: syntax error
cortex_m0_reduce25519.S:38: Error: unknown opcode or format name 'uxth'
cortex_m0_reduce25519.S:39: Error: extra comma
cortex_m0_reduce25519.S:39: Error: syntax error
cortex_m0_reduce25519.S:40: Error: unknown opcode or format name 'mul'
cortex_m0_reduce25519.S:41: Error: unknown opcode or format name 'mul'
cortex_m0_reduce25519.S:42: Error: extra comma
cortex_m0_reduce25519.S:42: Error: syntax error
cortex_m0_reduce25519.S:43: Error: extra comma
cortex_m0_reduce25519.S:43: Error: syntax error
cortex_m0_reduce25519.S:44: Error: extra comma
cortex_m0_reduce25519.S:44: Error: syntax error
cortex_m0_reduce25519.S:45: Error: extra comma
cortex_m0_reduce25519.S:45: Error: syntax error
cortex_m0_reduce25519.S:46: Error: bad register name: r4
cortex_m0_reduce25519.S:46: Error: junk at end of line, first unrecognized character is `r'
cortex_m0_reduce25519.S:46: Internal error!
Assertion failure in ignore_rest_of_line at /builds/idf/crosstool-NG/.build/src/binutils-2.25.1/gas/read.c line 3694.
Please report this bug.
...
```

mrkeuz, 2021-10-03:
Solved. By exclude files in `src/crypto/cortex/` from build. 

In the end reveals that `src/crypto/cortex` is some platform-specific implementation of cryptography and even not used (call of asm implementation `crypto_scalarmult_curve25519` is commented):

 -  https://github.com/ciniml/WireGuard-ESP32-Arduino/blob/ce65891c207141f9c75dd318a6faa53c6c20444e/src/crypto.h#L20-L21

So it can safety exclude from build. One thing, I don't find easy way to exclude some lib files from build via `platformio.ini` flags, it can achieve little tricky via python script (`extra_scipts = ...` flag). But I decided just fork project and fix library.

So you can use `lib_deps` like 
```ini
lib_deps = https://github.com/mrkeuz/WireGuard-ESP32-Arduino#a788ab8
```
Or fork you own repo (just delete `src/crypto/cortex` folder). It compiles successfully. But I don't tested exactly `wg` in on real hardware.

davey, 2021-11-10:
**I finally got a basic example setup working** and would like to share that. 🚀 
 
Thanks a lot for all those who contributed to this - this is great!

This is the content of my ESPhome config yaml (`wireguard-test.yaml`):
```
esphome:
  name: "wireguard-test"
  platform: esp32
  board: esp32dev
  arduino_version: latest
  platformio_options:
    lib_deps: "ciniml/WireGuard-ESP32"
  libraries: "WireGuard-ESP32"
  includes:
    - wireguard-test.wg.h

wifi:
  networks:
    - { ssid: "wifi-ssid-1",      password: "xxxxxxxxxxxx" }
    - { ssid: "wifi-ssid-2", password: "yyyyyyyyyyyyy" }
  ap: { ssid: "wgtest-SETUP",  password: "secretsetup" }

logger: { level: DEBUG, baud_rate: 115200}
web_server: { port: 80 }
time:
 - { platform: sntp, id: "sntp_time", timezone: "Europe/Berlin" }

custom_component:
- lambda: |-
    auto wireguard_component = new WireguardComponent();
    return {wireguard_component};
```

This is the content of my Wireguard config (`wireguard-test.wg.h`), which is mentioned in the file above:
```
#include "esphome.h"
#include "WireGuard-ESP32.h"

static WireGuard wg;

class WireguardComponent : public Component {
 public:

  // make sure we are starting LATE, after WIFI is initialized
  float get_setup_priority() const override { return esphome::setup_priority::LATE; }

  void setup() override {
    ESP_LOGD("custom", "wireguard setup: start ...");

    char private_key[] = "XXXXXXXXXXXXXXX="; // private key of the client (ESP)
    char public_key[]  = "YYYYYYYYYYYYYYYY="; // public key of the server (remote linux VPN server)
    IPAddress local_ip(192,168,99,123);           // VPN IP for this VPN client
    char endpoint_address[] = "vpn.example.com"; // VPN/Wireguard server hostname
    int endpoint_port = 51820;                   // VPN/Wireguard server port

    ESP_LOGD("custom", "wireguard setup: delay(5000) ...");
    delay( 5000 );
    ESP_LOGD("custom", "wireguard setup: begin ...");
    wg.begin(
        local_ip,
        private_key,
        endpoint_address,
        public_key,
        endpoint_port
    );
    ESP_LOGD("custom", "wireguard setup: end ...");
  }

  void loop() override {
    delay( 1000 );
    ESP_LOGD("custom", "wireguard loop ...");
  }
};
```

This is all that is needed on the **ESP / EPShome** side.

For completeness of this example, this is the essential part on the VPN/Wireguard **server** side (`/etc/wireguard/wg0.conf` on an Ubuntu server):

```
[Interface]
Address = 192.168.99.1/32
PrivateKey = PPPPPPPPPPPPPPPPPPP
ListenPort = 51820

## wireguard-test
[Peer]
PublicKey = VVVVVVVVVVVVVVVVVV
AllowedIPs = 192.168.99.123/32
```

I am sure there a a lot things that still need to be improved, e.g. properly waiting for the Wifi being established, making sure we reconnect in case the connection is lost and so on - I just wanted to share the earliest success to everyone waiting for it :-)

Next steps in my opinion would be to make the config more dynamic, e.g. being able to change the settings (server host, port, public and private keys and so on) or even generate a private key on the ESP itself, only showing/providing it's publickey part somewhere...

CarlosGS, 2021-11-11:
Thanks for the great work! You can also make it run upon time sync, which simplifies things:

wireguard-test.yaml
```yaml
esphome:
  name: wireguard-test
  platform: ESP32
  board: esp32dev
  libraries: ciniml/WireGuard-ESP32
  includes: wg-settings.h

wifi:
  networks:
    - { ssid: "wifi-ssid-1", password: "xxxxxxxxxxxx" }
  ap: { ssid: "wgtest-SETUP", password: "secretsetup" }

ota:
logger: { level: DEBUG, baud_rate: 115200}
web_server: { port: 80 }
time:
  - platform: sntp
    on_time_sync:
        then:
        - logger.log: "Starting wireguard..."
        - lambda: |
            wg.begin(local_ip,private_key,endpoint_address,public_key,endpoint_port);
```
wg-settings.h
```c++
#include <WireGuard-ESP32.h>
static WireGuard wg;

char private_key[] = "XXXXXXXXXXXXXXX="; // private key of the client (ESP)
char public_key[]  = "YYYYYYYYYYYYYYYY="; // public key of the server (remote linux VPN server)
IPAddress local_ip(192,168,99,123);           // VPN IP for this VPN client
char endpoint_address[] = "vpn.example.com"; // VPN/Wireguard server hostname
int endpoint_port = 51820;                   // VPN/Wireguard server port
```

The web server works, it's pretty amazing to have Wireguard running finally thanks a lot :D

However, I haven't been able to connect via OTA nor ESPHome API (adding `api:`). We must be missing some internal configuration.
Maybe we need to find a way to set up wireguard before api & ota. Almost there!

CarlosGS, 2021-11-21:
Actually, api & ota are reachable from Wireguard connected devices, so all OK on the ESPHome side :D

But it seems we are missing an IP route from the Home Assistant network to the Wireguard network (i.e. packets sent from HASS to the ESPHome node IP are forwarded to the router, where they are dropped). So neither HASS nor OTA can reach the node.
I think this thread https://community.home-assistant.io/t/declare-a-static-route-to-reach-other-lan/311442 can help towards a solution, but so far I haven't been able to define the routes correctly.

Does anyone know how to set up an IP route from HASS to the Wireguard addon?

CarlosGS, 2021-11-21:
OK, got it to work with Home Assistant & OTA! :smile: 

We were missing an IP route from the Home Assistant network to the Wireguard network (i.e. packets sent from HASS to the ESPHome node IP were being forwarded to the router, which dropped them). So neither HASS nor OTA could reach the nodes.

Here are early instructions to make it work with the official [Wireguard addon](https://github.com/hassio-addons/addon-wireguard), assuming it is already set up with the standard configuration. Then:
- You'll need command line access, using the [SSH & Web Terminal](https://github.com/hassio-addons/addon-ssh) addon, or directly through the HASS host display.
- In the terminal, get the IP of the wireguard addon: `ping a0d7b954_wireguard` and take note of it for the next command.
- Then create the route with `ip route add 172.27.66.0/24 via 172.30.33.2` First is the desired IP range to be forwarded, second is the internal IP for the wireguard addon (replace with the one you got in the previous command). Note: This is erased upon every reboot, a better solution will be needed.
- To add the nodes to Home Assistant you need to do it manually (Config->Integrations->Add->ESPHome) & typing the node's wireguard IP. And for the ESPHome dashboard, you need to enable `status_use_ping` so the nodes show as connected.

A bit tacky but at least we know it can work! Thanks @davey, @mrkeuz and @ciniml for making this possible! :tada:
Finally we can properly have remote ESPHome nodes :D

mrkeuz, 2021-11-23:
@CarlosGS fantastic news! 

I'm curious, did you try test connections exactly from `ESPHome`. I.e. try to connect to MQTT inside `wg` LAN. Seems `wg` in this case should start before MQTT component, right? And in general, will it be working at all. Just curious, how it works in core. 

Honestly, exposed API via `wg` is already BIG DEAL and this really cool!
Hopefully sooner or later this will land to mainstream.

@CarlosGS, @davey 🚀🚀🚀

CarlosGS, 2021-11-23:
Haven't tried MQTT yet, but it should work. Setup order shouldn't really matter as the Wireguard implementation overrides the default routing interface.

> Hopefully sooner or later this will land to mainstream.

Indeed! And ESP8266 is also based on lwip - do you think it could be possible to port it, or does it depend on internal cipher functions of ESP32?

mrkeuz, 2021-11-23:
>  ESP8266 is also based on lwip - do you think it could be possible to port it

Actually, I'm not core `c/cpp` developer. Just touching home automation a bit. And didn't dive into libraries/compatibility so deeply yet, especially with `lwip`. 

I think it needs just try. The missing functions can be added and replaced via some `#DEFINE` magic. All the more `wg` uses modern ciphers and I don't think there will be any hardware restrictions or limitations in this area. Also, as described in [original implementation](https://github.com/smartalock/wireguard-lwip#crypto-code), ciphers implemented in pure c, so don't think that here will be some problems. Just will be working little slowly.

Finally, found [next discussion](https://github.com/ciniml/WireGuard-ESP32-Arduino/issues/7#issuecomment-961262525) (see last comment). Seem it already worked on `ESP8266`. So just here a question how easily can be integrated into `esphome`.

CarlosGS, 2021-12-07:
The other item that wasn't working "out of the box" was the API logger. Connecting with the remote logger would make the ESP32 fall into some sort of loop and stop responding.
Found a solution after some debugging: we simply need to remove/comment the `ESP_LOG` statements from the wireguard code, it seems they clash with ESPHome's internal logger. Doing that, everything works just fine :tada: 

Also, for a complete official component we should consider enabling Wireguard in "OTA safe mode", this way remote nodes would be accessible even in the event of a reboot loop :)

:thinking: It would be great if someone could get ESPHome+Wireguard running on ESP8266, that would really shape an official method to have remote ESPHome nodes. Any news or testing, make sure to post here!

mrkeuz, 2021-12-16:
Just FYI. 

Found easy instruction how link `home-assistant` container to WireGuard (without `route` magic):

https://www.pedrolamas.com/2020/11/20/how-to-connect-to-a-wireguard-vpn-server-from-a-docker-container/

Maybe it might be useful for someone.

vsurkov, 2021-12-22:
Hello everyone, first of all, thanks a lot for your work, it's awesome!
I have an addition to the solution from @CarlosGS . Details are in this issue (https://github.com/ciniml/WireGuard-ESP32-Arduino/issues/13). The correct configuration should be like this to avoid reboots every ~10 min:

```
time:
  - platform: sntp
    on_time_sync:
        then:
        - lambda: |
            static const char* TAG = "wireguard";
            ESP_LOGD(TAG, "Starting...");
            if( !wg.is_initialized() ) {
                        ESP_LOGD(TAG, "Initializing WG interface...");
                        if( !wg.begin(
                                local_ip,
                                private_key,
                                endpoint_address,
                                public_key,
                                endpoint_port) ) {
                            ESP_LOGD(TAG, "Failed to initialize WG interface.");
                        }
                    }
```

I hope this information will be useful and save someone a few nights of debugging :)

trombik, 2022-01-06:
> Seem it already worked on `ESP8266`.

i did confirm that esp8266 works fine with the original implementation.

https://github.com/trombik/esp_wireguard

shirou93, 2022-01-29:
This feature will be added to esphome?

jarno83, 2022-03-01:
> > Seem it already worked on `ESP8266`.
> 
> i did confirm that esp8266 works fine with the original implementation.
> 
> https://github.com/trombik/esp_wireguard

Hi, I tried to build that with platformIO but I get error, How did you do it?

Thanks.

`.pio/libdeps/woodenbrick/esp_wireguard/src/crypto.c: In function 'crypto_equal':
.pio/libdeps/woodenbrick/esp_wireguard/src/crypto.c:18:5: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   a += 1;
     ^
.pio/libdeps/woodenbrick/esp_wireguard/src/crypto.c:19:5: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   b += 1;
     ^
Compiling .pio/build/woodenbrick/libaf1/esp_wireguard/crypto/cortex/cortex_m0_mpy121666.s.o
xtensa-lx106-elf-as: unrecognized option `-x'
*** [.pio/build/woodenbrick/libaf1/esp_wireguard/crypto/cortex/cortex_m0_mpy121666.s.o] Error 1`

trombik, 2022-03-01:
as I don't use platformio, I don't know.

fkoteam, 2022-03-14:
@CarlosGS thank you (gracias!). I'm very close to make work oracle cloud+home assistant+wireguard+esphome. Only have and issue (the last?). I've made your magic with the command sensor. With those line wireguard works, but home assistant loses internet connection. any idea?

CarlosGS, 2022-03-14:
Hmm, do you mean your Home assistant is running in an external server?

Mine runs in LAN, sorry, I'm not sure what would be needed to make it work
like that 🤔

fkoteam, 2022-03-15:
> Hmm, do you mean your Home assistant is running in an external server? Mine runs in LAN, sorry, I'm not sure what would be needed to make it work like that 🤔

yes, it's working on an oracle cloud+wireguard (i think it's great). I'd to enable ip forward to make it work (see my previous work).
I'd issues also to install all together with motioneye (+ esp32cam). I think I will have to change cidr address on command sensor:
`    command: host_result=$(host a0d7b954-wireguard); addon_ip=${host_result##* }; ip route replace 172.27.66.0/12 via $addon_ip; echo $addon_ip`

thank you!

jarno83, 2022-03-15:
Hi, can anyone make a little guide in one document, how novice users like me can get wiregurd working with esp32? Thanks :) I think it can help more people that me ;)

redstorm1, 2022-06-13:
> > Hi, can anyone make a little guide in one document, how novice users like me can get wiregurd working with esp32? Thanks :) I think it can help more people that me ;)
> 
> A quick and ugly guide. enjoy [hass.docx](https://github.com/esphome/feature-requests/files/8266767/hass.docx)

PSA this attachment is a virus.

ssieb, 2022-06-13:
Why do you think it's a virus?  It's a document describing how to setup wireguard on an esphome device.

redstorm1, 2022-06-13:
As my av picked up the doc is infected. Open it at your peril.Get Outlook for Android

ssieb, 2022-06-13:
I opened it in LibreOffice and there were no warnings.  I don't see any macros.  However, an online virus scanner does say it has a problem.  https://www.fortiguard.com/encyclopedia/virus/10092628  Use Windows at your peril. :-)

fkoteam, 2022-06-13:
the virus is on your mind. here's an online version to install Home Assistant on the cloud with Wireguard and esphome.

https://docs.google.com/document/d/1qZ69Z8mk8uFgMNyqTZbkIQTE1_aEXn3BoerFWkDh70c/edit?usp=sharing

ssieb, 2022-06-13:
@fkoteam, it's such a vague scan warning and I'm sure it's a false alarm due to the funny urls you used in the document.  You should probably make those just text and not real links.

trombik, 2022-06-26:
tried to port my `esp_wireguard` to `esphome`. however, `platformio` has been troublesome with `esp-idf`, and still is. some call it ["unpredictable wonky behavior"](https://github.com/espressif/esp32-camera#platformio-kconfig). until almost two-years-old issue https://github.com/platformio/platform-espressif32/issues/453 is fixed, it's not possible. the problem is directory of `lib_deps` is not part of  `EXTRA_COMPONENT_DIRS`, and `platformio` does not read `Kconfig` of `lib_deps`. they chose to implement their own, and failed to do it correctly. like it or not, that's the way it is.

than means you need to add `list(APPEND EXTRA_COMPONENT_DIRS path)` to `CMakeLists.txt`.

```cmake
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
list(APPEND EXTRA_COMPONENT_DIRS $ENV{PWD}/.pio/libdeps/wireguard)
project(wireguard)
```

a workaround is to add a feature to `esphome` that modifies `CMakeLists.txt` in the top directory. I don't think anyone would bother to implement it.

droscy, 2022-06-27:
Hi all, hi @trombik, I succeded in compiling with esphome (esp-idf framework) your implementation with some additional commits (see my fork and branch https://github.com/droscy/esp_wireguard/tree/esp32/espidf).

Now I'm facing the following issue: the connection to wireguard works only sometimes, the other times it fails with the following error on server syslog "Invalid handshake initiation from ip:port", and the error is repeated every 2-5 seconds. Do you have any hint?

droscy, 2022-06-28:
Many thanks trombik.

Actually both `init` and `connect` are executed only once if they return `ESP_OK` due to the two `if`s. I put them in a "crontab" just because I cannot user serial logging and the `on_time_sync` usually happens before the OTA logging is up, so this way I have 2 minutes to startup network logging.

But I was not aware that `esp_wireguard_connect()` returns immediately and not after a successful connection and that `esp_wireguardif_peer_is_up()` can be used to check the connection.

I'll change the yaml and I'll do more tests. Thanks.

trombik, 2022-06-28:
@droscy is this change clear enough?

https://github.com/trombik/esp_wireguard/commit/e92381aebcb66729b50d045228e7a9a9f8868183

droscy, 2022-06-30:
Yes, that's good, thanks.
Just a tip, both `connect()` and `peer_is_up()` accept `wireguard_ctx_t*` so the argument should be `&ctx`.

trombik, 2022-06-30:
@droscy good catch. will fix it. thanks

psxde, 2022-09-30:
The problem seems that if you use e.g. mqtt: esphome tries to connect mqtt first before executing sntp - so if mqtt is inside the wireguard network, the wg connection will never be called because esphome is in an endless loop trying to connect to an ip address which not exists.

redstorm1, 2022-10-11:
Malware:MSOffice/Follina.536C!exploit found

ssieb, 2022-10-24:
@redstorm1 as I mentioned earlier, that's a false positive for a really vague "virus" threat.  There is no virus, just invalid urls in the document.

ssieb, 2022-10-24:
@rovingclimber the ESP device doesn't connect to HA, so it doesn't need to know the address.  When you add the device to HA, you give HA the VPN address of the device.

rovingclimber, 2022-10-25:
@ssieb Thanks for the reply! I guess I'm struggling with 2 things:
1) I can't access the web interface of the ESP via wireguard. From the device debug I see the web interface is up on port 80, if I try to connect I can see traffic leave the wg interface on my router for port 80 on the device but nothing comes back. I can, however, see successful wg handshakes occurring so I know the keys etc are correct.
2) I don't know how to add a device in this context - normally the ESPhome gui does it for me ... for instance if I spin up a "fresh" ESPhome device and flash it the device shows online and I can add it to my dashboard ... I don't see anything in the config that tells HA the address of the device, so I don't know what to change or how to add my VPN device (which just shows offline after flashing)

From OPNSense wireguard status:
```
peer: XXXXXXXXXXXXXXXXXXXXXXXX=
  endpoint: 123.123.123.123:36215
  allowed ips: 192.168.252.15/32
  latest handshake: 1 minute, 18 seconds ago
  transfer: 41.62 KiB received, 23.29 KiB sent

```
From OPNSense packet capture on interface when trying to contact web server (see only outbound packets, nothing received)
```
wg1	10:30:22.316872 IP 192.168.252.1.54080 > 192.168.252.15.80: tcp 0
wg1	10:30:22.569605 IP 192.168.252.1.51424 > 192.168.252.15.80: tcp 0
wg1	10:30:23.325823 IP 192.168.252.1.54080 > 192.168.252.15.80: tcp 0
wg1	10:30:23.573555 IP 192.168.252.1.51424 > 192.168.252.15.80: tcp 0
wg1	10:30:25.336115 IP 192.168.252.1.54080 > 192.168.252.15.80: tcp 0
```
From ESP device log
```
[I][app:062]: setup() finished successfully!
[I][app:102]: ESPHome version 2022.10.0 compiled on Oct 24 2022, 14:35:47
[C][wifi:502]: WiFi:
[C][wifi:360]:   Local MAC: 58:BF:25:9D:6B:E0
[C][wifi:361]:   SSID: [redacted]
[C][wifi:362]:   IP Address: 192.168.98.36
[C][wifi:364]:   BSSID: [redacted]
[C][wifi:365]:   Hostname: 'wireguard-test'
[C][wifi:367]:   Signal strength: -31 dB ▂▄▆█
[C][wifi:371]:   Channel: 6
[C][wifi:372]:   Subnet: 255.255.255.0
[C][wifi:373]:   Gateway: 192.168.98.202
[C][wifi:374]:   DNS1: 192.168.98.202
[C][wifi:375]:   DNS2: 0.0.0.0
[D][sntp:075]: Synchronized time: 2022-10-25 11:46:15
[D][wireguard:029]: Starting...
[D][wireguard:031]: Initializing WG interface...
[D][wireguard:041]: Done...
[C][logger:275]: Logger:
[C][logger:276]:   Level: DEBUG
[C][logger:277]:   Log Baud Rate: 115200
[C][logger:278]:   Hardware UART: UART0
[C][web_server:125]: Web Server:
[C][web_server:126]:   Address: wireguard-test.local:80
[C][mdns:100]: mDNS:
[C][mdns:101]:   Hostname: wireguard-test
[C][ota:089]: Over-The-Air Updates:
[C][ota:090]:   Address: wireguard-test.local:3232
[C][ota:093]:   Using Password.
[C][api:138]: API Server:
[C][api:139]:   Address: wireguard-test.local:6053
[C][api:143]:   Using noise encryption: NO
[C][sntp:050]: SNTP Time:
[C][sntp:051]:   Server 1: '0.pool.ntp.org'
[C][sntp:052]:   Server 2: '1.pool.ntp.org'
[C][sntp:053]:   Server 3: '2.pool.ntp.org'
[C][sntp:054]:   Timezone: 'GMT0BST,M3.5.0/1,M10.5.0'

```

rovingclimber, 2022-10-26:
Hi all!
Just a note to say I did finally get this to work, this is what worked for me:

ESPHome yaml:
```
esphome:
  name: "wireguard-test"
  platform: esp32
  board: esp32dev
  arduino_version: latest
  libraries:
    - https://github.com/ciniml/WireGuard-ESP32-Arduino.git
  platformio_options:
    lib_deps: "ciniml/WireGuard-ESP32"
  includes:
    - wireguard-test.wg.h
    
wifi:
  ssid: "xxxxxxxxx"
  password: "xxxxxxxxx"
  use_address: 192.168.252.8
  
api:
  encryption:
    key: "F3lszhY1dYcfC/d1F50RKQ4A0tzDk/xW2SRCV+SWnOM="

ota:
  password: "1b48a9e64ef8810887fe9119fef57af7"

logger: { level: DEBUG, baud_rate: 115200}
web_server: { port: 80 }
time:
  - platform: sntp
    on_time_sync:
        then:
        - lambda: |
            static const char* TAG = "wireguard";
            ESP_LOGD(TAG, "Starting...");
            if( !wg.is_initialized() ) {
                        ESP_LOGD(TAG, "Initializing WG interface...");
                        if( !wg.begin(
                                local_ip,
                                private_key,
                                endpoint_address,
                                public_key,
                                endpoint_port) ) {
                            ESP_LOGD(TAG, "Failed to initialize WG interface.");
                        }
                    }
```

/config/esphome/wireguard-test.wg.h:
```
#include "esphome.h"
#include "WireGuard-ESP32.h"

static WireGuard wg;

char private_key[] = "XXXXX="; // private key of the client (ESP)
char public_key[]  = "XXXXX="; // public key of the server (remote linux VPN server)
IPAddress local_ip(192,168,252,8);           // VPN IP for this VPN client
char endpoint_address[] = "myhost.external.com"; // VPN/Wireguard server hostname
int endpoint_port = yyyyy;                   // VPN/Wireguard server port
```
In this example my wireguard endpoint is my OPNsense router, the **use_address** statement tells HA to connect to the ESP via 192.168.252.8 which is routable via OPNsense from LAN side. I confirm web interdace, API and OTA update all working, I've successfully tested with WEMOS LOLIN32 clone and ESP32-CAM, connected to mobile phone hotspot, all accessible from LAN via wireguard.

However ...... [see next post]

rovingclimber, 2022-10-26:
...
Having got ESPHome / wireguard working nicely via @davey @CarlosGS  method above ....

My end-game actually is to wireguard enable this ESPHome BLE BMS (battery management system): https://github.com/syssi/esphome-jbd-bms

Unfortunately (or fortunately) this project is using ESP-IDF platform, not Arduino ... so the above method won't work.

@trombik @droscy I've just tried using your library as per your comments above but I can't get it to compile.

Here's my .yaml:
```
esphome:
  name: wg-test-idf
  libraries: "https://github.com/droscy/esp_wireguard.git#esphome/1"
  includes:
    - secrets.h
    - wg_settings.h

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf
    version: latest

logger:

api:
  encryption:
    key: "xxxxxxx"

ota:
  password: "xxxxxxx"

wifi:
  ssid: "xxxxxxx"
  password: "xxxxxxxxxxxx"

time:
  - platform: sntp
    on_time_sync:
        then:
          - lambda: |-
              ESP_LOGD(TAG, "time synchronized");
    on_time:
      - seconds: 0
        minutes: /2
        then:
          - lambda: |-
              ESP_LOGD(TAG, "check");
              if(wg_initialized != ESP_OK)
              {
                  ESP_LOGD(TAG, "initializing...");

                  wg_config.private_key = wg_peer_privkey;
                  wg_config.public_key = wg_server_pubkey;
                  wg_config.allowed_ip = wg_peer_ip;
                  wg_config.allowed_ip_mask = wg_peer_ip_mask;
                  wg_config.endpoint = wg_server_endpoint;
                  wg_config.port = wg_server_port;
                  wg_config.persistent_keepalive = wg_keepalive;

                  wg_initialized = esp_wireguard_init(&wg_config, &ctx);

                  if(wg_initialized == ESP_OK)
                      ESP_LOGI(TAG, "initialized");
                  else
                      ESP_LOGW(TAG, "cannot initialize, error code %d", wg_initialized);
              }

              if(wg_initialized == ESP_OK && wg_connected != ESP_OK)
[... 95 lines cut ...]
/data/cache/platformio/packages/framework-espidf/components/log/include/esp_log.h:302:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGD( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_DEBUG,   tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
/data/wg-test-idf/.piolibdeps/wg-test-idf/esp_wireguard/src/wireguardif.c:922:2: note: in expansion of macro 'ESP_LOGD'
  ESP_LOGD(TAG, "underlying_netif = %p", underlying_netif);
  ^~~~~~~~
/data/wg-test-idf/.piolibdeps/wg-test-idf/esp_wireguard/src/wireguardif.c:922:41: note: each undeclared identifier is reported only once for each function it appears in
  ESP_LOGD(TAG, "underlying_netif = %p", underlying_netif);
                                         ^~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/log/include/esp_log.h:334:137: note: in definition of macro 'ESP_LOG_LEVEL'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/log/include/esp_log.h:302:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGD( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_DEBUG,   tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
/data/wg-test-idf/.piolibdeps/wg-test-idf/esp_wireguard/src/wireguardif.c:922:2: note: in expansion of macro 'ESP_LOGD'
  ESP_LOGD(TAG, "underlying_netif = %p", underlying_netif);
  ^~~~~~~~
/data/wg-test-idf/.piolibdeps/wg-test-idf/esp_wireguard/src/wireguardif.c:1002:1: error: label 'fail' defined but not used [-Werror=unused-label]
 fail:
 ^~~~
cc1: some warnings being treated as errors
*** [/data/wg-test-idf/.pioenvs/wg-test-idf/lib830/esp_wireguard/wireguardif.o] Error 1
========================= [FAILED] Took 23.65 seconds =========================
```

lhoracek, 2023-01-03:
Started wrapping this into component to allow easy usage:
https://github.com/esphome/esphome/pull/4256

Help wanted.

```yaml
# equivalent shorthand for GitHub pull request
  - source: github://pr#4256
    components: [ wireguard ]
```

```yaml
wireguard:
  address: 10.0.0.1
  private_key: private_key=
  peer_key: public_key=
  peer_endpoint: wg.server.example
```

my-bam, 2023-01-06:
> ```yaml
> wireguard:
>   address: 10.0.0.1
>   private_key: private_key=
>   peer_key: public_key=
>   peer_endpoint: wg.server.example
> ```

Hi. I get an error:
`src/esphome/components/wireguard/wireguard.cpp:4:10: fatal error: arpa/inet.h: No such file or directory
    4 | #include <arpa/inet.h>
      |          ^~~~~~~~~~~~~
compilation terminated.`

How do I fix it?

lhoracek, 2023-01-10:
> Hi. I get an error: `src/esphome/components/wireguard/wireguard.cpp:4:10: fatal error: arpa/inet.h: No such file or directory 4 | #include <arpa/inet.h> | ^~~~~~~~~~~~~ compilation terminated.`
>
> How [...]

What is you config? ESP32 or ESP8266 or .. ?

my-bam, 2023-01-11:
> What is you config? ESP32 or ESP8266 or .. ?

ESP8266 (board: d1_mini)

lhoracek, 2023-01-12:
@my-bam Unfortunately, this is for ESP32 only for now.

timsmithlx, 2023-01-16:
So ihave managed to get a wireguard link, i can see the ESPHOME Device in integrations and turn on and off a switch remotely

What i cant get is the ESPHOME dashboard to show the device as online or be able to update "wirelessly"

Followed Carlos method... any tips?

originalbase, 2023-02-13:
> OK, got it to work with Home Assistant & OTA! 😄
> 
> We were missing an IP route from the Home Assistant network to the Wireguard network (i.e. packets sent from HASS to the ESPHome node IP were being forwarded to the router, which dropped them). So neither HASS nor OTA could reach the nodes. I've arranged these instructions to make it work:
> 
>     1. Set up the official [Wireguard addon](https://github.com/hassio-addons/addon-wireguard) with the standard configuration, verify you can connect to HASS from your phone on data.
> 
>     2. Add the following to your Home Assistant's configuration.yaml. It creates a "command" sensor to periodically fetch the Wireguard addon's internal IP, and also configure the correct IP route.
> 
> 
> ```yaml
> sensor:
>   - platform: command_line
>     name: Wireguard addon internal IP # Creates the IP route to support ESPHome devices on Wireguard
>     command: host_result=$(host a0d7b954-wireguard); addon_ip=${host_result##* }; ip route replace 172.27.66.0/24 via $addon_ip; echo $addon_ip
> ```
> 
>     3. Restart HASS, the new text sensor will take a couple minutes to settle into an IP like 172.30.33.2.
> 
>     4. Then note: To add the ESPHome nodes to Home Assistant you need to do it manually by going to `Config->Integrations->Add->ESPHome` and typing the node's specific wireguard IP. Make sure that you have the `api:` line in the yaml for each node. And for the ESPHome dashboard, remember to enable `status_use_ping` so the nodes are properly detected.
> 
> 
> A bit tacky but at least we know it can work! Thanks @davey, @mrkeuz and @ciniml for making this possible! 🎉 Finally we can properly have remote ESPHome nodes :D

I created an automation that is triggered by HA start and WG addon start.
It fires a 'shell_command', which does exactly what your command line sensor does.

And it worked brilliantly for a while (HassOS on RPi 4). A remote microcontroller connected though a wireguard tunnel and completely servicable via OTA... amazing.

Sadly, setting a route - as mentioned above - now throws an error:
**ip: RTNETLINK answers: Operation not permitted**

Presumably, a recent update rendered this method useless.

Does somebody else face the same problem?

Is the issue related to docker capabilities (NET_ADMIN)?
Is there another simple and persistent solution to make wireguard clients accessible for home assistant?

I don't see the woods for all the trees.
Help me find the elephant in the room.

vuzdorf, 2023-02-17:
Wireguard works on esp32 thanks to [lhoracek](https://github.com/lhoracek)!, Home Assistant can show sensor data after it was manually added, BUT in my case OTA isn't working, and I don't know what do. As I'm running HA in VirtualBox from default image - it's very restricted on what can be done. After connecting device via WG, ESPhome and the device are in different subnets, as it's impossible to send mDNS through the subnet barrier - ESPHome can not resolve device's IP by name.
Modifying /etc/hosts - doesn't work, it is possible to ping and traceroute WG device by IP, but even nslookup shows nothing.
Installing DNSmasq plugin and configuring hosts there - doesn't work - after adding DNSmasq to resolv.conf - nslookup shows the device, but ESPHome still can't resolve the IP from name.
Changing address to IP in /config/esphome/.esphome/[devicename].yaml.json - only makes device online in ESPHome dashboard for a while - an attempt to flash the device seems overwriting this file.
What else can be done? Is there any possibility to enforce ESPHome use wireguard IP instead of devicename.local hostname?

lhoracek, 2023-02-19:
One workaround should be to use `use_address` in you `wifi` config to tell esphome where to look for you device, otherwise it tries mDns and that is most likely no-go with VPN configs.

originalbase, 2023-02-19:
@lhoracek Providing the ip of the wg client (and using ping for discovery) is part of the solution. The other part is routing the traffic from Home Assistant to the wireguard container.

A few days ago the shell command shown above by @CarlosGS was executable (manually via termimal (addon) or automatically via `sensor` or `shell_command` triggered by an `automation`) did the trick.

Initially everything worked fine.
Trying to set a route now the same way throws an error (ip: RTNETLINK answers: Operation not permitted).

I am not able to identify the recent change, that causes this new restriction.

originalbase, 2023-02-19:
I use Home Assistant OS and therefore I'm not able (or intended) to modify the docker network settings without corrupting the concept of a self organising system (definitely hacky, potentially tacky).

I my opinion it's worth to spend a bit of time to investigate for the **tiny missing link**, that causes the perfectly integrated solution to fail recently.

Using a dedicated vpn server and the routing capabilities on the local gateway is a proper solution but lacks the (energy) efficiency and might be oversized for most of the szenarios (here: a handful of microcontrollers).

Another idea is, to implement a respective configuration setting / a routing table on this addon (or addons and physical interfaces in general).

Before considering all of that
and intended not to blur the view on my specific issue with more warm air:

Can anyone identify the recent change that is responsible for the lack of permission to set a route in the Home Assistant container?

@CarlosGS Does the sensor still work for you?

CarlosGS, 2023-02-19:
Yes, the sensor in https://github.com/esphome/feature-requests/issues/1444#issuecomment-974915276 still works for me. Using Home Assistant Operating System on both a Raspberry 4 and also on a VirtualBox instance. Sorry not sure of what could be happening in your installation :S

originalbase, 2023-02-19:
Fine, back to the drawing board. Thanks for the immediate reply.

vuzdorf, 2023-02-19:
Set up a supervised version of HA on a virtualboxed debian, configured Wireguard on OS level (not plugin), now getting an unpredictable number of invalid handshakes before connection is finally established:
[image]
What it could be? Connecting to the same wg by a phone works perfect from the first shot.

vuzdorf, 2023-02-20:
> One workaround should be to use `use_address` in you `wifi` config to tell esphome where to look for you device, otherwise it tries mDns and that is most likely no-go with VPN configs.

YEAH!!! It works, finally! @lhoracek Thank You!

Tested setup:
 - HomeAssistant VirtualBox
 - Wireguard Plugin

Wg Plugin config:
```
 peers:
  - addresses:
      - 172.27.66.2/32
    allowed_ips: []
    client_allowed_ips: []
    name: esp
    public_key: w7/KKktxHAW6BEQvNWmeKvslaVNblkUyImLhVdMZTmI=

```

```
esphome:
  name: wgtest100500
external_components:
  - source: github://pr#4256
    components: [ wireguard ]
    
esp32:
  board: esp32dev
  framework:
    type: arduino

wireguard:
  address: 172.27.66.2
  private_key: device_private_key=
  peer_key: server_public_key=
  peer_endpoint: your_domain.com

time:
  - platform: sntp
    id: mytime

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 172.27.66.2

```

mrivero33, 2023-02-23:
What about ESP8266, it´s posible??

ZenonBombalina, 2023-03-08:
Great job Gents.
Maybe you will be able to help: 
After configuration esp  was started correctly, handshake was successful, everything was working as charm, however (without any further flashing, or changing anything) after some time I start to get in logs message like that:
```
D][sntp:078]: Synchronized time: 2023-03-08 21:01:23
[D][wireguard:038]: Starting...
[D][wireguard:040]: Initializing WG interface...
E (24824) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (24824) task_wdt:  - loopTask (CPU 1)
E (24824) task_wdt: Tasks currently running:
E (24824) task_wdt: CPU 0: IDLE
E (24824) task_wdt: CPU 1: IDLE
E (24824) task_wdt: Aborting.

abort() was called at PC 0x400f68dc on core 0

Backtrace:0x40083771:0x3ffbea3c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
```

Any ideas what I can adjust?
Many thanks in advance.

distancerunner, 2023-03-16:
> Hello everyone, first of all, thanks a lot for your work, it's awesome! I have an addition to the solution from @CarlosGS . Details are in this issue ([ciniml/WireGuard-ESP32-Arduino#13](https://github.com/ciniml/WireGuard-ESP32-Arduino/issues/13)). The correct configuration should be like this to avoid reboots every ~10 min:
> 
> ```
> time:
>   - platform: sntp
>     on_time_sync:
>         then:
>         - lambda: |
>             static const char* TAG = "wireguard";
>             ESP_LOGD(TAG, "Starting...");
>             if( !wg.is_initialized() ) {
>                         ESP_LOGD(TAG, "Initializing WG interface...");
>                         if( !wg.begin(
>                                 local_ip,
>                                 private_key,
>                                 endpoint_address,
>                                 public_key,
>                                 endpoint_port) ) {
>                             ESP_LOGD(TAG, "Failed to initialize WG interface.");
>                         }
>                     }
> ```
> 
> I hope this information will be useful and save someone a few nights of debugging :)

My ESP32 client with ESPHome yaml also connects to my VPN. In this VPN is my mqtt-broker. But the problem is mqtt on my ESP32 client. Mqtt starts before the VPN initialisation and keeps trying to connect. My VPN needs to be at first, and mqtt needs to wait till finished vpn-connection. Did someone know, how to time this?

RoboMagus, 2023-03-29:
I'd love to see this component supported not only on the Arduino framework, but ESP-idf also.
A quick look at the initial library mentioned does not actually show me any real Arduino dependencies.

Keep my hopes up? 😅

[60 more replies]

closed 2023-09-12 by jesserockz