esphome/feature-requests #1447: esphome as a c++ library

zhulik, 2021-10-04:
**Describe the problem you have/What new integration you would like**
Hello, I beleave it would be amazing to use esphome as a c++ library.

**Please describe your use case for this integration and alternatives you've tried:**
The other day I was trying to implement some near real time processing on a esp32 using esphome and found out that neither `on_loop' nor `Component::loop` give me enough presision to do my stuff and esphome does not provide any other lower level APIs like accessing cores and threads. I thought what if it was possible to use esphome as a c++ library. It would give users much more control over their hardware and flexibility when they need it. In my case I'd dedicate one core to that processing, and run esphome on the second one. Unfortunately I found little to nothing info on how to use esphome as a library which can be manually called from c++ code. The other option be providing lower level APIs accessible right from yamls

**Additional context**

WeekendWarrior1, 2021-10-05:
Check out: https://github.com/esphome/esphome/blob/dev/esphome/core/helpers.h#L75
And for an example of a component using it: https://github.com/esphome/esphome/blob/dev/esphome/components/a4988/a4988.cpp#L38

This is how the stepper can do eg. 3600 steps/s

zhulik, 2021-10-05:
@WeekendWarrior1 I did try it, but delays between calls to 'Component::loop` were still too high: I need 40000, not 3600

WeekendWarrior1, 2021-10-05:
As far as I know there's nothing stopping you from using freertos commands in your custom component, eg xTaskCreate (in fact the BLE beacon and camera components both pin tasks to a specific core):
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html
May have to tweak some things that happen in here as well: https://github.com/esphome/esphome/blob/dev/esphome/components/esp32/core.cpp

zhulik, 2021-10-05:
That's actually a good idea. I'll check it out. Thanks!

OttoWinter, 2021-10-17:
That was the original esphome model (https://github.com/esphome/esphome-core), but we moved away from it.

Fundamentally, ESPHome is a framework not a library. You can add custom code like mentioned above

closed 2021-10-17 by OttoWinter