esphome/feature-requests #1462: support for ethernet RTL8201FI

crashbash2020, 2021-10-15:
---
name: Feature request
about: Suggest an idea for ESPHome

---

**Describe the problem you have/What new integration you would like**
support for ethernet RTL8201FI

**Please describe your use case for this integration and alternatives you've tried:**

WESP32 boards from revision 7 now use the RTL8201FI PHY.

This board was great as it is as far as I know the only board with 12v+ out with POE ethernet support. Naturally WIFI still works and the Power itself can work without ethernet, but I prefer use hardwared connections. Unfortunately the manufacturer has changed from LAN8720 to RTL8201 so this no longer works. Apparently it only requires a change in the PHY definition in the code, but I have no idea how this works with respect to ESPhome so hopefully this is an easy feature to add :)

**Additional context**

xorbit, 2021-10-29:
Manufacturer of the wESP32 here.  I took a look at the code and it's a little more involved here than just changing the PHY definition from LAN8720 to RTL8201 as can be done for instance in MicroPython or ESP32 Arduino.

This code seems to be some odd mix of Arduino (listed as a requirement for the Ethernet driver) and the older IDF 3, which didn't support the RTL8201.  IDF 4 does as this support, as does ESP32 Arduino core 2.0+, but the Ethernet support code in ESPhome seems to be written for IDF 3 and will have to be changed.

As I said, I took a quick look to see if I could do it, but I'm not familiar enough with the internal workings of ESPhome to do this, I'm afraid.

crashbash2020, 2021-10-31:
Ah that's too bad, thanks for taking a look anyway! will have to see what the ESPhome team thinks when they have a chance to look at it

brad07x, 2021-12-26:
+1 for this FR

REH-dev, 2022-05-16:
+1 for this FR

hachi, 2022-05-23:
Anyone I can pay to do this? :)

adrummond-github, 2022-05-24:
Any update? Its been 7-8 months.

mrand, 2022-05-25:
Asking for updates, or posting +1 to the bug tracker for that matter, is bad form.  

If you want to help bring attention to this issue, encourage others that want it to thumbs-up the first post.

adrummond-github, 2022-05-25:
Thanks for the advice! Good job helping.

jaxer, 2022-05-28:
i wish i knew that before buying wesp32. 

might make sence to update [documentation](https://esphome.io/components/ethernet.html). it hit me by surprise that even though wesp32 is mentioned in ethernet component doc, it is in fact not supported :(

adrummond-github, 2022-05-28:
> i wish i knew that before buying wesp32.
> 
> might make sense to update [documentation](https://esphome.io/components/ethernet.html). it hit me by surprise that even though wesp32 is mentioned in ethernet component doc, it is in fact not supported :(

Same - wish the wesp manufacture would make an integration. Don’t bother asking him though.

hachi, 2022-05-29:
Could we get some comments on how the implementation might need to be attacked? If this involves porting the driver or changing IDF version or Arduino libs?

xorbit, 2022-05-30:
I do not know much about the internal structure of ESPHome so this may be of limited use, but here's the problem as I see it.

RTL8201FI support was added in ESP-IDF 4 and ESP32 Arduino Core version 2.  From looking at the current code, it looks to me like it uses a mixture of IDF 3 and Arduino Core before version 2.  I could be off on this.  When this issue was first raised I tried to rewrite the code for IDF 4 but I could not get it to compile.

What are the current versions of ESP-IDF and/or Arduino Core that ESPHome is based on?  If it's still IDF 3 and Arduino Core pre-v2, are there plans to port it to the newer versions since IDF v3 has been EOL since February?  Is there a development branch that uses IDF 4 where this driver could be added?

Despite @adrummond-github's best efforts to vilify me and make it seem like I don't care, this is not the case.  I would love to see this support added and am willing to contribute, but I don't know enough about ESPHome to take this on without help.

adrummond-github, 2022-05-30:
I wasn’t going to get all into it but hey, thanks for the mention. I just think it’s kinda unethical you knew your product was used, and advertised compatibility over at ESPHome, and you changed the hardware without notifying them, knowing it would break support. You even discussed the issue here, acknowledging the issue. Nearly a year later and they were still advertising support for your product. Your website specifically excludes returns after flashing from the default load, so those expecting ESPhome support end up with paperweights after they load esp home and it doesn’t work. But hey to be fair you never advertised it had support right? I should have read the reviews of other having the exact same issue. I notified ESPhome and they updated their documentation the same day… seems like the right thing to do, so others aren’t dealing with this issue. I’d hardly say my comment an attempt to vilify, I reached out to you directly to find out why you changed the hardware (and why wasn’t ESPHome notified when you clearly knew) and if you’d work with ESPHome to with an integration since the product no longer worked and you were all around unhelpful. Seriously pages and pages of emails of how I’m a terrible customer for leaving a poor review and a useless manager type. Seriously. childish. behavior. Either way, ESPHome is using outdated software and it needs to be updated, if it brings back support for WESP32 that’s great as that ecosystem is sorely lacking poe Ethernet options. We’ll be looking at the olimex boards in the mean time as well as non esphome options.

crashbash2020, 2022-05-31:
@adrummond-github As far as I can remember the Wesp was never advertised as compatible with ESPhome from Xorbit, it was a possible configuration on the ESPhome documentation. that documentation should have been updated more promptly, but until you have contacted them likely no-one had and they likely were not aware of this breaking change.

I don't think its on the manufacturer to trawl the internet and ask permission to change their design if it breaks compatibility with 3rd party system. remember the new WESP still works perfectly fine in other environments with up to date IDFs, and its not like the change in eth adapter was hidden, it was clearly mentioned on the store page (At least when I purchased mine) 

I posted this FR in the hope it was an easy fix, clearly its going to take a bit more work from the the ESPhome team, but given this likely only affects a tiny subset of ESPhome users its likely very far down their priority list. Until the ESPhome team is ready to work on the issue I don't know what you expect Xorbit to do given he has already made an attempt to fix himself.

adrummond-github, 2022-05-31:
Totally a fair opinion, like I said I reached out and it turned into a bunch of drama with the guy. I’ve opened a couple of feature requests to address the issues with ESPHome and I took care of the comparability notice on the ESPHome site, not sure that was my job either. But hey - trying to help the community.

crashbash2020, 2022-05-31:
yeah i never considering contacting them (well i kind of thought this feature request would be a quick fix so the documentation would just added to, not removed lol)

I'm still using the wesp, I preferred hardwired to WIFI but luckily its not essential for me as i have good wifi coverage in my house. I still have it wired via POE for power and just connect wirelessly. Not ideal but at least if there is a software fix I can likely easily fix the issue. for me the 12v out is was an absolute must to control some 10v analogue circuits, i think the olimex esp32 POE was the only other one i found that had voltage output but i think it was limited to 5v which means i need a boost circuit or something else which i couldn't be bothered with.

adrummond-github, 2022-05-31:
Not a problem - ya it seems like a good design and the fit/finish is nicer than the olimex boards. The 5v output isn’t a deal breaker for me. Unfortunately our use case is in environments where we don’t have Wi-Fi coverage so functional Ethernet is a must. Poe isn’t a hard requirement for us, but it’s certainly a bonus and makes powering them easier and cheaper (we already have poe available for ip cameras).

dklemm, 2022-05-31:
@xorbit from what I understand the main conversion has been done and closed here #1414, you can see in the comments there are still some compenents left to migrate includinging ethernet here #1683.

adamdunkley, 2022-06-19:
@xorbit I assume that branch has now been merged the branch was cleaned up. You can see the code on the [pull request](https://github.com/esphome/esphome/pull/2303) that brought the change in.

0x90ChuckTessela, 2022-06-22:
It doesn't appear that the PR linked above addresses the RTL8201 issue that is the root cause of the WESP32 rev7 not being supported for Ethernet in esphome.  Am I missing something?

genestealer, 2022-07-04:
Well this is very frustrating, I just got my WESP32 from Mouser, only to find out it's not supported!

ma-lalonde, 2022-09-14:
@xorbit The platform version refers to the PlatformIO `platform-espressif32` version. As the comment just above says, you can look at the [GitHub Release Notes](https://github.com/platformio/platform-espressif32/releases/tag/v3.5.0) to see that `Platform 3.5.0` = `Framework 4.3.2`

On another note, I need to implement the KSZ8081, which requires Arduino version 2.0.3+ and Platform version 5.0.0+   (Framework 4.4+). Still looking at the best way to do that to get a PR accepted quickly and that will work in the future. Maybe a version check in `ethernet_component` ? I know MicroPython does it and it works, I still have a [PR](https://github.com/micropython/micropython/pull/8710) open for it.

ma-lalonde, 2022-09-16:
I gave this a day, and came up with something that compiles without warnings or errors. But it doesn't run (yet), the log is just stuck at `waiting for download` after a `POWERON_RESET`. [Here is the fork](https://github.com/Connaxio/esphome/tree/feature/ethernet-arduino-2)

Implementing new PHYs requires changing from Arduino 1.0.6 to 2.0.0+, so it is not trivial (to me anyway).

Might also relate to https://github.com/esphome/esphome/pull/3564. @jesserockz, you might be interested in this.

I am myself looking for a bit of help on this, as I don't have a grasp on everything that's going on, and have absolutely no idea where to go from here.

xorbit, 2022-09-19:
@ma-lalonde Great work, I'm glad to see you got it to the point of compiling, something I never succeeded in.  You obviously have a much better understanding of how ESPHome works than I ever will.  I am a hardware and firmware guy, I don't think I'll ever understand complicated build systems with sprawling dependencies.

Since I saw in your code that you implemented all the IDF4 PHYs including the RTL8201, I'd like to test your fork to see if I get the same results as you see on the wESP32 rev 7.  But as an immediate illustration of my lack of understanding, how do I make sure I run your fork and not the official ESPHome?  Can you give me a pointer?

ma-lalonde, 2022-09-19:
@xorbit sure! I think the following should work:
```bash
pip3 uninstall esphome
git clone -b feature/ethernet-arduino-2 https://github.com/Connaxio/esphome.git
cd esphome
script/setup

esphome run esptest.yaml
```
If it runs, and you want to make sure you're using the right version, just add a typo to `ethernet_component.h`, like `#include ETH2.h` instead of `#include ETH.h`, and re-run esphome, it should throw an error.

Unfortunately I don't have more time to spend on this at the moment, unless someone comes up with more pointers. I actually had a similar issue previously (https://github.com/esphome/issues/issues/3542), which I did not manage to solve. Maybe a JTAG debugger would help, I don't have a setup ready.

xorbit, 2022-09-19:
Tried this config:

```yaml
esphome:
  name: wesptest

esp32:
  board: esp32dev
  framework: 
    type: arduino
    version: 2.0.4
    platform_version: 5.1.1

# Enable logging
logger:
  level: VERBOSE

ethernet:
  type: RTL8201
  mdc_pin: GPIO16
  mdio_pin: GPIO17
  clk_mode: GPIO0_IN
  phy_addr: 0
```

Sigh.  Seems to build, but fails to even upload.

```
xorbit@galena:~/Projects/esphome$ esphome run wesptest.yaml --device /dev/ttyUSB0 
INFO Reading configuration wesptest.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
******************************************************************************************************
Obsolete PIO Core v6.0.2 is used (previous was 6.1.4)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
******************************************************************************************************
Processing wesptest (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.1.1)
------------------------------------------------------------------------------------------------------
Removing unused dependencies...
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 2.0.0
|-- Ethernet @ 2.0.0
|-- ESPmDNS @ 2.0.0
Compiling .pioenvs/wesptest/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/wesptest/src/esphome/components/esp32/gpio_arduino.cpp.o
Compiling .pioenvs/wesptest/src/esphome/components/esp32/gpio_idf.cpp.o
Compiling .pioenvs/wesptest/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/wesptest/src/esphome/components/ethernet/ethernet_component.cpp.o
Compiling .pioenvs/wesptest/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/wesptest/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/wesptest/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
Compiling .pioenvs/wesptest/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/wesptest/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling .pioenvs/wesptest/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/wesptest/src/esphome/core/application.cpp.o
Compiling .pioenvs/wesptest/src/esphome/core/color.cpp.o
Compiling .pioenvs/wesptest/src/esphome/core/component.cpp.o
Compiling .pioenvs/wesptest/src/esphome/core/component_iterator.cpp.o
Compiling .pioenvs/wesptest/src/esphome/core/controller.cpp.o
Compiling .pioenvs/wesptest/src/esphome/core/entity_base.cpp.o
Compiling .pioenvs/wesptest/src/esphome/core/helpers.cpp.o
Compiling .pioenvs/wesptest/src/esphome/core/log.cpp.o
Compiling .pioenvs/wesptest/src/esphome/core/scheduler.cpp.o
Compiling .pioenvs/wesptest/src/esphome/core/util.cpp.o
[... 73 lines cut ...]
esp32_create_combined_bin([".pioenvs/wesptest/firmware.bin"], [".pioenvs/wesptest/firmware.elf"])
Flash params set to 0x0020
Wrote 0xcf480 bytes to file /home/xorbit/Projects/esphome/.esphome/build/wesptest/.pioenvs/wesptest/firmware-factory.bin, ready to flash to offset 0x0
==================================== [SUCCESS] Took 7.98 seconds ====================================
INFO Successfully compiled program.
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}] [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress] [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '/home/xorbit/Projects/esphome/.esphome/build/wesptest/.pioenvs/wesptest/patched_bootloader.bin'
INFO Upload with baud rate 460800 failed. Trying again with baud rate 115200.
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}] [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress] [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '/home/xorbit/Projects/esphome/.esphome/build/wesptest/.pioenvs/wesptest/patched_bootloader.bin'
```

What's this `patched_bootloader.bin` that it seems to want and doesn't have, and where is it supposed to come from?

xorbit, 2022-09-19:
Flashed the image manually with:

```
xorbit@galena:~/Projects/esphome/.esphome/build/wesptest/.pioenvs/wesptest$ esptool.py --port /dev/ttyUSB0 write_flash 0x0 firmware-factory.bin
```
Same result as reported by @ma-lalonde:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff0030,len:1184
load:0xffffffff,len:-1
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

Anyone have a clue what is wrong?

ma-lalonde, 2022-09-19:
Now that you mention it, I did have exactly the same errors (oops, my bad), and overcame them the exact same way. Not a pro on these things either, but the fact that there is a `patched_bootloader.bin` missing would match with what we're seeing: that the bootloader is faulty and that the program won't start (maybe?) 

I see there is a branch called `esp32-arduino-framework-2.0.0`, maybe the bootloader issue comes from the framework version change and has been fixed there. Had I seen it before, I would have based my modifications on it, maybe they solved this bootloader issue there.

xorbit, 2022-09-19:
The [release notes](https://github.com/platformio/platform-espressif32/releases/tag/v5.1.0) for `platform-espressif32` version 5.1.0 indicate that they switched to merged images that contain the bootloader for compatibility with Arduino 2.0.4, so they are flashed at 0x0.

ESPHome may not have caught up with that change and tries to flash the bootloader separately?  Anyhow, manually flashing the image _should_ have fixed that, but didn't.

Out of curiosity, I tried this with Arduino version 2.0.3.  Flashing did automatically work, indicating the flashing issue is indeed related to this change in 2.0.4.  Also, the run behavior is a bit different, although not good either.  The board sits in a loop doing this:

```
[15:59:26]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[15:59:26]configsip: 0, SPIWP:0xee
[15:59:26]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:59:26]mode:DIO, clock div:2
[15:59:26]load:0x3fff0030,len:1184
[15:59:26]load:0x40078000,len:12784
[15:59:26]load:0x40080400,len:3032
[15:59:26]entry 0x400805e4
[15:59:26][\xa2\xeamum\x95\xcd\xc1\xcd2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[15:59:26][I][logger:243]: Log initialized
[15:59:26][I][app:029]: Running through setup()...
[15:59:26][V][app:030]: Sorting components by setup priority...
[15:59:26][C][ethernet:069]: Setting up Ethernet...
[15:59:26]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[15:59:26]
[15:59:26]Core  1 register dump:
[15:59:26]PC      : 0x400d38ab  PS      : 0x00060030  A0      : 0x8014adc0  A1      : 0x3ffb2730  
WARNING Decoded 0x400d38ab: esphome::ethernet::EthernetComponent::setup() at /home/xorbit/Projects/esphome/.esphome/build/wesptest/src/esphome/components/ethernet/ethernet_component.cpp:77
[15:59:26]A2      : 0x3ffb943c  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000004  
[15:59:26]A6      : 0x3ffb94ac  A7      : 0x3ffb94ac  A8      : 0x00000000  A9      : 0x3ffb26e0  
[15:59:26]A10     : 0x00000000  A11     : 0xf364b13c  A12     : 0x00000028  A13     : 0x3ffc314c  
[15:59:26]A14     : 0x00060a20  A15     : 0x00000002  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
[15:59:26]EXCVADDR: 0x00000000  LBEG    : 0x40089761  LEND    : 0x40089771  LCOUNT  : 0xffffffff  
[15:59:26]
[15:59:26]
[15:59:26]Backtrace:0x400d38a8:0x3ffb27300x4014adbd:0x3ffb2760 0x4014ae1d:0x3ffb2780 0x400d54e8:0x3ffb27a0 0x400d630a:0x3ffb27e0 0x400da1aa:0x3ffb2820 
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400d38a8: esphome::ethernet::EthernetComponent::setup() at /home/xorbit/Projects/esphome/.esphome/build/wesptest/src/esphome/components/ethernet/ethernet_component.cpp:77
WARNING Decoded 0x4014adbd: esphome::Component::call_setup() at /home/xorbit/Projects/esphome/.esphome/build/wesptest/src/esphome/core/component.cpp:76
WARNING Decoded 0x4014ae1d: esphome::Component::call() at /home/xorbit/Projects/esphome/.esphome/build/wesptest/src/esphome/core/component.cpp:97
WARNING Decoded 0x400d54e8: esphome::Application::setup() at /home/xorbit/Projects/esphome/.esphome/build/wesptest/src/esphome/core/application.cpp:38
WARNING Decoded 0x400d630a: setup() at /home/xorbit/Projects/esphome/.esphome/build/wesptest/src/main.cpp:83
WARNING Decoded 0x400da1aa: loopTask(void*) at /home/xorbit/.platformio/packages/framework-arduinoespressif32@3.20003.220626/cores/esp32/main.cpp:42
[15:59:26]
[15:59:26]
[15:59:26]
[15:59:26]
[15:59:26]ELF file SHA256: 0000000000000000
[15:59:26]
[15:59:26]Rebooting...
```

I guess this is progress, since it looks like it's booting up and then crashing in your Ethernet code... 😆

ma-lalonde, 2022-09-19:
Can confirm, it pings! I added your changes and a mention to my fork.

I am absolutely not sure about the typecast on `power_pin_`, however. Would need someone with a board with a power pin.
Also now gotta test with a functional application :)

nagyrobi, 2022-09-20:
Fyi: https://github.com/esphome/esphome/pull/3751
Maybe worth a look on @kbx81's approach.

xorbit, 2022-09-20:
@nagyrobi I don't think it is relevant to this because the IP101 was already supported in IDF 3.x / Arduino core 1.x for ESP32.  The RTL8201FI and KSZ8081 PHYs require IDF 4.x / Arduino core 2.x.

xorbit, 2022-12-07:
Hurray!  Can confirm that 2022.12 (beta) works on wESP32 rev 7 using this config:

```yaml
esphome:
  name: wesptest

esp32:
  board: wesp32

logger:
  level: VERBOSE

ethernet:
  type: RTL8201
  mdc_pin: GPIO16
  mdio_pin: GPIO17
  clk_mode: GPIO0_IN
  phy_addr: 0
```

Here's the console output for `esphome run wesptest.yaml`:

```
[12:45:57]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:45:57]configsip: 0, SPIWP:0xee
[12:45:57]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:45:57]mode:DIO, clock div:2
[12:45:57]load:0x3fff0030,len:1184
[12:45:57]load:0x40078000,len:13132
[12:45:57]load:0x40080400,len:3036
[12:45:57]entry 0x400805e4
[12:45:58][     6][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[12:45:58][I][logger:258]: Log initialized
[12:45:58][I][app:029]: Running through setup()...
[12:45:58][V][app:030]: Sorting components by setup priority...
[12:45:58][C][ethernet:028]: Setting up Ethernet...
[12:46:02][V][ethernet:207]: [Ethernet event] ETH started (num=0)
[12:46:02][V][ethernet:242]: DHCP Client Status: 0
[12:46:17][W][ethernet:122]: Connecting via ethernet failed! Re-connecting...
[12:46:17][V][ethernet:242]: DHCP Client Status: 0
[12:46:26][V][ethernet:207]: [Ethernet event] ETH connected (num=2)
[12:46:28][V][ethernet:213]: [Ethernet event] ETH Got IP (num=4)
[12:46:28][I][app:062]: setup() finished successfully!
[12:46:28][I][app:102]: ESPHome version 2022.12.0b1 compiled on Dec  7 2022, 12:43:09
[12:46:28][C][logger:293]: Logger:
[12:46:28][C][logger:294]:   Level: VERBOSE
[12:46:28][C][logger:295]:   Log Baud Rate: 115200
[12:46:28][C][logger:296]:   Hardware UART: UART0
[12:46:28][C][ethernet:163]: Ethernet:
[12:46:28][C][ethernet:281]:   IP Address: 10.10.0.99
[12:46:28][C][ethernet:282]:   Hostname: 'wesptest'
[12:46:28][C][ethernet:283]:   Subnet: 255.255.255.0
[12:46:28][C][ethernet:284]:   Gateway: 10.10.0.1
[12:46:28][C][ethernet:289]:   DNS1: 10.10.0.1
[12:46:28][C][ethernet:290]:   DNS2: 0.0.0.0
[12:46:28][C][ethernet:297]:   MAC Address: A8:03:2A:F3:6C:EB
[12:46:28][C][ethernet:302]:   Is Full Duplex: YES
[12:46:28][C][ethernet:307]:   Link Speed: 100
[12:46:28][C][ethernet:168]:   MDC Pin: 16
[12:46:28][C][ethernet:169]:   MDIO Pin: 17
[12:46:28][C][ethernet:170]:   Type: RTL8201
[12:46:28][C][mdns:103]: mDNS:
[12:46:28][C][mdns:104]:   Hostname: wesptest
[12:46:28][V][mdns:105]:   Services:
[12:46:28][V][mdns:107]:   - _http, _tcp, 80
[12:46:28][V][mdns:109]:     TXT: version = 2022.12.0b1
```

nagyrobi, 2022-12-07:
Is franework `esp-idf`?

xorbit, 2022-12-07:
I tried it with both `framework: type: esp-idf` and framework: type: arduino` added under `esp32:` and it worked in both cases.

I'm not very familiar with the organization of the ESPHome repos, can you give me a link to the correct repo to update the docs?  Thanks.

xorbit, 2022-12-08:
Ok, I found it and added this PR:
https://github.com/esphome/esphome-docs/pull/2512

krbaker, 2022-12-11:
Many thanks to the work here, just dropping in to comment this is working great for me