esphome/feature-requests #1463: Allow esphome to load image from URL

henriklied, 2021-10-15:
**Describe the problem you have/What new integration you would like**
Make it possible for the image module to fetch images from a remote server using the HTTP module.

**Please describe your use case for this integration and alternatives you've tried:**
Being able to load externally generated images for use with eink and other displays

**Additional context**
This would make it very easy to e.g. generate screenshot from a HA Lovelace view and display it directly in esphome.

nagyrobi, 2021-11-01:
Let HA do the hard work and present the decoded/processed image back to the node via API.

kaystrobach, 2022-01-08:
Maybe this helps - and the I’m curious about this too 🤔

https://github.com/bitbank2/PNGdec

kaystrobach, 2022-01-08:
Here is the original source:

https://hackaday.com/2021/07/17/png-image-decoding-library-does-it-with-minimal-ram/

dgomes, 2022-01-19:
Also interested in this feature

guillempages, 2022-01-23:
The Inkplate e-ink displays (also ESP32 based) offer this functionality in their standard library (without ESPHome), so the code should be able to fit in the ESP32. It might be an issue on the ESP8266 though.
In any case, it would be nice to have this feature; so that complex images can be created off-board and just displayed.

kaystrobach, 2022-01-23:
Any idea on how to get started on this? Is the code of the ink plat MIT? 😉

nagyrobi, 2022-01-23:
Let LVGL to the hard work: https://github.com/fvanroie/esphome-lvgl It supports some image processing already.

guillempages, 2022-01-23:
The code is in https://github.com/e-radionicacom/Inkplate-Arduino-library, but it seems to be LGPL :-(
There is a special license for the TinyJPG libray though; a BSD license.

ssieb, 2022-01-30:
Why do you think LGPL is a problem?  The esphome source is GPL.

guillempages, 2022-01-30:
Sorry, it's a work-related reflex: "don't use any GPL code". But you are right, if esphome is GPL anyway, LGPL is of course good to go.
Regardless, the PNG part of that repository is used from another library, and that one is MIT.

kaystrobach, 2022-01-31:
And MIT is even less restrictive. So this should be fine.

kaystrobach, 2022-03-18:
Ok, this is insane: https://libwebsockets.org/git/libwebsockets/tree/READMEs/README.html-parser.md

guillempages, 2022-06-07:
@everyone interested in this feature; I've created a PR that implements support for downloading and displaying PNG images; if you could test it as well (I only have an inkplate device to test with) I might be able to iron out some bugs before it is finally integrated.
You can just use the pr by adding
```
external_components:
  - source: 'github://pr#3255'
    components:
      - display
      - online_image

 online_image:
  - url: "https://upload.wikimedia.org/wikipedia/commons/4/47/PNG_transparency_demonstration_1.png"
    id: png_image
```
to your configuration, and
```
  display:
    ...
   lambda: |-
      it.print(0, 0, id(png_image));
```
to your display configuration.

Thanks!

(P.S. this will only work for displays available in core; external_components displays might cause a compilation error)

guillempages, 2022-06-19:
@lux4rd0 I've done a quick test with the transparency image, and it worked flawlessly on my inkplate; I even copy& pasted the link from your comment, to make sure I didn't have a typo in it. 
Unfortunately there is no higher verbose mode; AFAIR it returns the error code of the http class. I'd have to look at the code to be sure, but I'm on the phone ATM. 

Regarding the inverted image/colors, I'll also have to do some extra tests, but IIRC it was showing right on my inkplate in binary mode. In any case, for that particular case there is already a solution available: in the "it.image" call you can add two extra parameter "COLOR_OFF" and "COLOR_ON" to explicitly select which colors to use for black and white. Maybe my coffee is wrong, or maybe there is an issue in the lilygo implementation; I'll take a look at soon as I can.

I also haven't seen the problem when the image is too big, but I'll take a look at that as well and try to reproduce it, or add some checks for it of bounds. Maybe it is an issue on the lilygo as well, that the driver doesn't check for it of bounds, while the inkplate does.

guillempages, 2022-06-19:
The inverted image issue might be related to https://github.com/esphome/issues/issues/3132; We might need to wait for a decision to be made there to decide how to proceed here.

lux4rd0, 2022-06-19:
@guillempages Thanks for the inspection on this - and tracking down that probable image inverse issue! :) I also think the driver for my Lilygo is responsible for some of the out-of-bounds issues. If I try to place text outside of the display - it also crashes (depending on which boundary - sometimes it'll wrap - other times, crashes...) I think most of it I can work around as I'm still building the dithering image collection someplace else - then using your method to pull exactly the image that my display needs (sized, inverted, and dithered to binary...) Loving it so far!!

guillempages, 2022-06-19:
@lux4rd0 I've checked the code on both the inkplate and the lilygo driver you are using; the inkplate does have a check for out-of-bounds; the lilygo doesn't. I've created a PR there (https://github.com/ashald/esphome/pull/2) to address that; but it might have a slight performance impact (I'm not sure how big though; id don't expect it to really be measurable)

guillempages, 2022-06-19:
@lux4rd0 as for the inverted; in your case, I would not invert the image on your proxy (so that if you look at it on your browser, it does look normal); I would just call the lambda as:
```
it.image(0, 0, id(image), COLOR_OFF, COLOR_ON);
```

lux4rd0, 2022-06-19:
> @lux4rd0 as for the inverted; in your case, I would not invert the image on your proxy (so that if you look at it on your browser, it does look normal); I would just call the lambda as:
> 
> ```
> it.image(0, 0, id(image), COLOR_OFF, COLOR_ON);
> ```

I like this!! Let me try it... _(although - I still need to dither externally... grrrr)_

sdwilsh, 2022-06-24:
I was trying to test this out with my inkplate 10 based on the instructions in [this comment](https://github.com/esphome/feature-requests/issues/1463#issuecomment-1148364047), but unfortunately, I'm getting a compiler error:
```
/config/esphome/living-dash.yaml: In lambda function:
/config/esphome/living-dash.yaml:93:31: error: no matching function for call to 'esphome::display::DisplayBuffer::print(int, int, esphome::online_image::OnlineImage*&)'
     it.print(0, 0, id(png_image));
                               ^
In file included from src/esphome.h:20:0,
                 from src/main.cpp:3:
src/esphome/components/display/display_buffer.h:167:8: note: candidate: void esphome::display::DisplayBuffer::print(int, int, esphome::display::Font*, esphome::Color, esphome::display::TextAlign, const char*)
   void print(int x, int y, Font *font, Color color, TextAlign align, const char *text);
        ^
src/esphome/components/display/display_buffer.h:167:8: note:   candidate expects 6 arguments, 3 provided
src/esphome/components/display/display_buffer.h:177:8: note: candidate: void esphome::display::DisplayBuffer::print(int, int, esphome::display::Font*, esphome::Color, const char*)
   void print(int x, int y, Font *font, Color color, const char *text);
        ^
src/esphome/components/display/display_buffer.h:177:8: note:   candidate expects 5 arguments, 3 provided
src/esphome/components/display/display_buffer.h:187:8: note: candidate: void esphome::display::DisplayBuffer::print(int, int, esphome::display::Font*, esphome::display::TextAlign, const char*)
   void print(int x, int y, Font *font, TextAlign align, const char *text);
        ^
src/esphome/components/display/display_buffer.h:187:8: note:   candidate expects 5 arguments, 3 provided
src/esphome/components/display/display_buffer.h:196:8: note: candidate: void esphome::display::DisplayBuffer::print(int, int, esphome::display::Font*, const char*)
   void print(int x, int y, Font *font, const char *text);
        ^
src/esphome/components/display/display_buffer.h:196:8: note:   candidate expects 4 arguments, 3 provided
*** [/data/living-dash/.pioenvs/living-dash/src/main.cpp.o] Error 1
```

I'm running on 2022.6.2.

sdwilsh, 2022-06-24:
Oh fish!  I think I copied it from the email I got from this issue, and didn't spot the difference when I checked before adding my comment.  Unfortunately, it crashes at boot with a non-symbolicated back trace in a loop, so it doesn't appear to be working still.

sdwilsh, 2022-06-25:
Oh fish!  I think I copied it from the email I got from this issue, and didn't spot the difference when I checked before adding my comment.  Unfortunately, it crashes at boot with a non-symbolicated backtrace in a loop, so it doesn't appear to be working still.

guillempages, 2022-06-26:
@sdwilsh Without the log, I cannot tell you for sure what might be broken. One thing that comes to mind is, if you try to download the image before the network has been fully setup. OTOH, I've just tried, and I get a "Connection Error: -1" at the very beginning, but it is not crashing.
Could you share the relevant parts of your configuration? Are you trying to download an image via http or https? And is it a local image or an internet one? Using DNS or IP directly?

In any case, IMHO it would be better to followup bugs/implementation related issues in the implementation PR comments ;-) (https://github.com/esphome/esphome/pull/3255)

lux4rd0, 2022-07-11:
@guillempages Getting close to finishing my screens with loading in external images. Here's an example of showing a lightning heatmap of distance vs the number of strikes. Still dithering externally to binary due to the display driver - but it's close! Thanks again for your efforts on this!!

[image]

guillempages, 2023-04-19:
@sdwilsh it's been a *long* while, but I've started another attempt at the online images. Could you try https://github.com/esphome/esphome/pull/4710, and see if you are still getting those reboot errors?

guillempages, 2023-06-21:
@mu88 The current implementation only supports esp32 devices (and might work with esp8266), RP2040 is not supported.
There are two main reasons for that: I don't have an RP2040 to test, and the http_client API is slightly different on the RPi.

Once this is finalized and integrated into the mainline, it should be possible to enhance it to support RPi.

mu88, 2023-06-22:
Thx @guillempages ! Could you recommend any other approach that would fit my needs of transferring an image onto the Raspi for displaying?

guillempages, 2023-06-24:
@mu88 I don't know the whole landscape of projects supporting displays ;-)
If your display is supported on ESPHome, you might want to try using the HTTPRequest component, and once you have the data, display it, but I'm not sure how feasible it is. When I looked at that two years ago, the HTTP_Request component could not give the data; I think this has changed now, but I'm not sure.

closed 2024-08-06 by jesserockz

DavidAntonin, 2024-08-29:
Can you add support for JPG format? Thank you

kaystrobach, 2024-11-23:
How about a conversion proxy? You max offline the heavy lifting

mortenmoulder, 2024-11-23:
@kaystrobach From my end that's already what I do. I use ffmpeg to create a PNG snapshot every second, then use pngquant to decrease the file size, by limiting the color range etc. However, the same ~35 KB PNG is about 7 KB as JPEG. That's a huge difference.

guillempages, 2025-01-22:
> Can you add support for JPG format? Thank you

Was always on my TODO list, but I had a lot of other higher prio projects to finish first.

> How hard would it be to implement JPEG next to PNG? Reason I ask is PNG is lossless, which means that no matter how much we try to compress the image, by removing colors etc., it's never going to be anywhere near as good as JPEG.

Harder than it would seem. But I finally managed to get a first version working :-)

You might want to take a look at https://github.com/esphome/esphome/pull/8127

DISCLAIMER: It is still a draft PR, not fully tested, and not much error-checking yet. But for early alpha adopters, it might be a start.