esphome/feature-requests #1469: Add support for ITead's SONOFF NSPanel touchscreen display (ESP32 with custom Nextion HMI screen)?

Hedda, 2021-10-19:
**Describe the problem you have/What new integration you would like**

Request support in ESPHome for SONOFF NSPanel touchscreen display by ITead (based on ESP32 with custom Nextion HMI screen):

https://itead.cc/product/sonoff-nspanel-smart-scene-wall-switch/

[image]

Kickstarter campaign now live

https://www.kickstarter.com/projects/sonoffnspanel/sonoff-nspanel-smart-scene-wall-switch/

A lot of information had already leaked via its FCC certification (FCC ID 2APN5NSPANEL):

https://fccid.io/2APN5NSPANEL

Looks like a model each for Europe and North America; “NSPanel-EU” and “NSPanel-US”. It also contains a temperature + humidity sensor and two integrated low-amp relays for 220V/300W or 110V/150W capacity respectively, but the point is really for it to act as remote control of scenes and running widgets (local applications or scripts) such as for example an HVAC thermostat.

As I understand all Nextion displays as their own integrated MCU so off-load most of the heavy lifting work and all text and graphics, animations, etc. are created using Nextion Editor software, and then then the project files/assets are uploaded directly into the Nextion screen. The screen then has serial commands mapped to UI elements and can be sent and received from ESPHome firmware to control it. 

There is some interesting discussions on the Home Assistant community forum about hacking it with ESPHome and/or Tasmota:

https://community.home-assistant.io/t/sonoff-nspanel-smart-scene-wall-switch-by-itead-coming-soon-on-kickstarter/332962/

HA-TB303 and blakadder are two that have especially connected a lot of information about the internal workings of these:

https://espthings.io/nspanel

https://blakadder.com/nspanel-teardown/

First thing they discovered it that has practically been designed it to be hacked and flashed with third-party firmware. Especially when taking into consideration that Efuses are not burned and pin headers on the board are easy to use for a solderless flash.

Early information is also that it does not use the existing standard protocol for Nextion HMI but instead a custom protocol:

**Nextion HMI Project Upload Protocol specifications:**

https://nextion.tech/2017/12/08/nextion-hmi-upload-protocol-v1-1/

ITead has also posted specific IO (Input-Output) definition documention for Sonoff NSPanel in a new Kickstarter campaign update:

[image]

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

jesserockz, 2021-10-20:
I will do just that @Hedda. Thanks =)

Hedda, 2021-11-11:
FYI, blakadder now said that he is trying to replicate the original default firmware but using MQTT to be able to integrate it in HA. 

Everything neatly laid out here for hacking it and configuring Tasmota and blakadder will be updated as support progresses:

https://templates.blakadder.com/sonoff_NSPanel

Again, also be sure to check out this teardown as an introduction to the hardware:

https://blakadder.com/nspanel-teardown/

jesserockz, 2021-11-11:
@Hedda https://twitter.com/Jesserockz/status/1457415401743126530 =)

Hedda, 2021-11-12:
> @Hedda https://twitter.com/Jesserockz/status/1457415401743126530 =)

Awesome stuff! Will keep and eye on https://github.com/esphome/esphome/pull/2702 and your Twitter while waiting for ITead to ship the Sonoff NSPanel!

mbo18, 2021-11-29:
@jesserockz Thank you so much for this component!
It seems that weather data are not currently supported (YAML part missing form my understanding). Do you plan to finish the NSPanel support for next release?

mbo18, 2021-12-12:
I've flashed my NSPanel with jesserockz's code and it's working pretty fine. I've found how to declare widgets in the YAML configuration. It's clear that "devices" are not currently supported but "groups" are. Regarding "devices", "CURTAIN: 11" is missing in the code.
Regarding groups, I think that I found bugs when using multi-switch (2 ou 3 or 4):
- first switch is always on, all others are off
- names are not displayed so it's OUTLET1, OUTLET2, etc.
- only the last ON_CLICK action is triggered

For example, with this code:

```yaml
    - type: group
      name: Salon
      uiid: SWITCH_VERTICAL_TRIPLE
      items: 
        - name: "Open"
          on_click:
            - homeassistant.service:
                service: cover.open_cover
                data:
                  entity_id: cover.salon_volet_roulant
        - name: "Close"
          on_click:
            - homeassistant.service:
                service: cover.close_cover
                data:
                  entity_id: cover.salon_volet_roulant
        - name: "STOP"
          on_click:
            - homeassistant.service:
                service: cover.stop_cover
                data:
                  entity_id: cover.salon_volet_roulant
``` 
If I click on OUTLET1 or 2 or 3, only the last on_click action is triggered, here it is `cover.stop_cover`

TyzzyT, 2021-12-17:
When trying to compile with Jesse's code, I'm getting the error below. Any idea what can be wrong?

```
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPmDNS> 1.0
Compiling /data/sonoff-nspanel1/.pioenvs/sonoff-nspanel1/src/esphome/components/nspanel/nspanel.cpp.o
Compiling /data/sonoff-nspanel1/.pioenvs/sonoff-nspanel1/src/esphome/components/text_sensor/text_sensor.cpp.o
Compiling /data/sonoff-nspanel1/.pioenvs/sonoff-nspanel1/src/esphome/components/time/automation.cpp.o
Compiling /data/sonoff-nspanel1/.pioenvs/sonoff-nspanel1/src/esphome/components/time/real_time_clock.cpp.o
In file included from src/esphome/components/sensor/sensor.h:3:0,
                 from src/esphome/components/nspanel/nspanel.h:6,
                 from src/esphome/components/nspanel/nspanel.cpp:1:
src/esphome/components/nspanel/nspanel.cpp: In member function 'bool esphome::nspanel::NSPanel::process_data_()':
src/esphome/components/nspanel/nspanel.cpp:99:50: error: 'format_hex_pretty' was not declared in this scope
            format_hex_pretty(message_data, length).c_str());
                                                  ^
src/esphome/core/log.h:95:89: note: in definition of macro 'esph_log_d'
   esp_log_printf_(ESPHOME_LOG_LEVEL_DEBUG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                         ^
src/esphome/components/nspanel/nspanel.cpp:98:3: note: in expansion of macro 'ESP_LOGD'
   ESP_LOGD(TAG, "Received NSPanel: Type=0x%02X PAYLOAD=%s RAW=[%s]", type, message.c_str(),
   ^
Compiling /data/sonoff-nspanel1/.pioenvs/sonoff-nspanel1/src/esphome/components/uart/uart.cpp.o
Compiling /data/sonoff-nspanel1/.pioenvs/sonoff-nspanel1/src/esphome/components/uart/uart_component.cpp.o
Compiling /data/sonoff-nspanel1/.pioenvs/sonoff-nspanel1/src/esphome/components/uart/uart_component_esp32_arduino.cpp.o
*** [/data/sonoff-nspanel1/.pioenvs/sonoff-nspanel1/src/esphome/components/nspanel/nspanel.cpp.o] Error 1
========================= [FAILED] Took 12.08 seconds =========================
```

samuelmcconnell, 2021-12-18:
On ESPHome 2021.12.1, I'm seeing a similar error on compilation as what @TyzzyT is getting:

```
Processing nspanel (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <WiFi> 1.0
|-- <Update> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPmDNS> 1.0
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/nspanel/nspanel.cpp.o
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/output/automation.cpp.o
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/output/float_output.cpp.o
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/resistance/resistance_sensor.cpp.o
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/rtttl/rtttl.cpp.o
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/sensor/automation.cpp.o
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/sensor/filter.cpp.o
In file included from src/esphome/components/sensor/sensor.h:3:0,
                 from src/esphome/components/nspanel/nspanel.h:6,
                 from src/esphome/components/nspanel/nspanel.cpp:1:
src/esphome/components/nspanel/nspanel.cpp: In member function 'bool esphome::nspanel::NSPanel::process_data_()':
src/esphome/components/nspanel/nspanel.cpp:99:50: error: 'format_hex_pretty' was not declared in this scope
            format_hex_pretty(message_data, length).c_str());
                                                  ^
src/esphome/core/log.h:95:89: note: in definition of macro 'esph_log_d'
   esp_log_printf_(ESPHOME_LOG_LEVEL_DEBUG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                         ^
src/esphome/components/nspanel/nspanel.cpp:98:3: note: in expansion of macro 'ESP_LOGD'
   ESP_LOGD(TAG, "Received NSPanel: Type=0x%02X PAYLOAD=%s RAW=[%s]", type, message.c_str(),
   ^
*** [/data/nspanel/.pioenvs/nspanel/src/esphome/components/nspanel/nspanel.cpp.o] Error 1
========================== [FAILED] Took 3.15 seconds ==========================
```

mbo18, 2021-12-18:
Yeah I'm not alone using ESPHome with a NSPanel 👍 
You need to comment those 2 lines. Using (for example) VSCode or File Editor add-on, go to config/esphome/.esphome/external_components/8d2a6156/esphome/components/nspanel/nspanel.cpp and comment line 98 and 99 by adding //:
```
//  ESP_LOGD(TAG, "Received NSPanel: Type=0x%02X PAYLOAD=%s RAW=[%s]", type, message.c_str(),
//           format_hex_pretty(message_data, length).c_str());
```
Also, you need to change 1 thing in your YAML:
```
  - platform: gpio
    name: "$devicename Screen Power"
    id: screen_power
    entity_category: config
    pin:
      number: 4
      inverted: true
    restore_mode: ALWAYS_OFF
```
Restore Mode for the screen power switch must now be set to ALWAYS_OFF

TyzzyT, 2021-12-18:
> 

Hey @mbo18,
I've managed to flash the panel with ESPHome, thank you very much for your help. Where do you define the widgets as you described?
I tried something like the code below, but that doesn't work.
```
nspanel:
  id: nspanel_id
  time_id: homeassistant_time
  temperature: temperature
  eco_mode_switch: eco_mode
  screen_power_switch: screen_power
  relays:
    - relay_1
    - relay_2

  - type: group
    name: Group1
    uiid: SWITCH_VERTICAL_SINGLE
    items: 
      - name: "Toggle"
        on_click:
          - homeassistant.service:
              service: light.toggle
              data:
                entity_id: light.christmas_tree
```

mbo18, 2021-12-18:
Currently we can only use scene or empty.
Here is my code: 
```
nspanel:
  id: nspanel_id
  time_id: current_time
  temperature: average_1st_temp
  screen_power_switch: screen_power
  eco_mode_switch: eco_mode
  relays:
    - relay_1
    - relay_2
  widgets:
    - type: scene
      name: LEntrée
      on_click:
        - homeassistant.service:
            service: light.toggle
            data:
              entity_id: light.entree_bulb
    - type: empty
    - type: empty
    - type: empty
    - type: empty
    - type: empty
    - type: empty
    - type: empty
```

TyzzyT, 2021-12-18:
Thanks @mbo18, you're the best! This is working for me.

samuelmcconnell, 2021-12-18:
Is there any support for changing the orientation of the screen yet? I have the portrait-orientation NSPanel but it displays landscape.

mbo18, 2021-12-18:
That’s in NSPanel settings and not in ESPHome I think. Slide your finger from the top off the screen to the bottom

mkaatman, 2021-12-19:
I can't get mine to register when I try to change orientation. I see the option to do it.

mbo18, 2021-12-19:
I can’t help you, this option does not exist in the EU version

samuelmcconnell, 2021-12-19:
@mkaatman When I tapped the option for orientation, nothing changed on-screen. However, after power cycling, it appeared with the orientation I had selected.

mkaatman, 2021-12-19:
That was it. I must have done the old triple tap prior. Now how do you get Fahrenheit to stick?

mbo18, 2021-12-19:
Under `nspanel:` set `temperature_unit_celcius` to false.

jesserockz, 2021-12-22:
Sorry, still catching up on Github notifications.

Great to see people are receiving their NSPanels and using ESPHome on them :heart: 

A few things:

- I decided to remove support for groups for now, but its easy enough for you to re-enable by cloning and removing one section in the `__init__.py` file
	- Further to groups, I could not get the names to change for each item. It seems they might be hard-coded to `Outlet 1/2/etc`
- Yeah orientation is a weird one. I will check to see if there is a message received when the user changes it from UI to force a power reboot of the screen itself.
- Regarding the compile issues, `format_hex_pretty` is only in `dev` and therefore either you need to use dev, or comment out the lines as @mbo18 has done.

Please do keep sending in your issues/feedback and I will try to make this work the best I can.

Jesse

mbo18, 2021-12-22:
Hi @jesserockz! thanks for working on this.
I'm just wondering how you will implement the weather part, switches (group) and devices. Do you already have worked on something?
Regarding weather:
- You can ask for a weather entity and then ESPHome has to get the weather state (icon), the current temp, the min and max temp for forecast[0]. That may be a lot of work? (creating a weather component with a `homeassistant` platform?)
Something like:
```
nspanel:
  ...
  weather_entity: weather.home
```
- Or you can ask for different sensors like:
```
nspanel:
  ...
  weather:
    temperature:
    temp_min:
    temp_max:
    icon:
```
which will be sensors form HA:
```
sensor:
  - platform: homeassistant
    entity_id: sensor.weatherhome_temperature
    id: outdoor_temp
    internal: true
  - platform: homeassistant
    entity_id: sensor.weatherhome_min_temp_0
    id: outdoor_temp_min
    internal: true
  - platform: homeassistant
    entity_id: sensor.weatherhome_max_temp_0
    id: outdoor_temp_max
    internal: true
text_sensor:
  - platform: homeassistant
    entity_id: weather.home
    id: outdoor_icon
    internal: true
    filters:          # mapping done directly by the component?
      - map:
        - clear-night -> CLOUD_BLUE
        - cloudy -> SUN_CLOUD
        - exceptional -> RED_THERMOSTAT
        - fog -> CLOUD_FOG
        - hail -> CLOUD_ICE_CRYSTALS
        - lightning -> CLOUD_RAIN_LIGHTNING
        - lightning-rainy -> CLOUD_RAIN_LIGHTNING
        - partlycloudy -> SUN_CLOUD
        - pouring -> CLOUD_RAIN
        - rainy -> CLOUD_RAIN
        - snowy -> CLOUD_SNOWFLAKES
        - snowy-rainy -> CLOUD_RAIN_SNOW
        - sunny -> SUNNY
        - windy -> WIND
        - windy-variant -> WIND
```
Regarding devices, do you plan to add `homeassistant` platform to light, climate and cover?

And finally, regarding groups, a binary sensor and a switch sensor per HA light or switch could do the job? like:
```
binary_sensor:
  - platform: homeassistant
    id: light1_state
    entity_id: light.entree_bulb
    internal: true
switch:
  - platform: template
    id: light1_switch
    internal: true
    lambda: |-
      if (id(light1_state).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - homeassistant.service:
          service: light.turn_on
          data:
            entity_id: light.entree_bulb
    turn_off_action:
      - homeassistant.service:
          service: light.turn_off
          data:
            entity_id: light.entree_bulb
```
then NSPanel group's state and action could come from the switch template. The configuration could look like:
```
- type: group
  name: Light 1
  uiid: SWITCH_VERTICAL
  items:
    - name: "Light 1"
      switch: light1_switch   
```

jesserockz, 2021-12-22:
@mbo18 Ah something I added in my local config, but not in the PR for weather:

```yaml
api:
  services:
    - service: weather_data
      variables:
        icon: int
        temperature: int
        min: int
        max: int
      then:
        - lambda: id(nspanel_id).send_weather_data((nspanel::WeatherIcon)icon, temperature, min, max);
```

The icons list is here: https://github.com/esphome/esphome/pull/2702/files#diff-90cb7313b616f9d6e69e3358d14b0325f04255d8e84891a20431dd5eb57a7543R17-R32

> * Or you can ask for different sensors like:
> 
> ```
> nspanel:
>   ...
>   weather:
>     temperature:
>     temp_min:
>     temp_max:
>     icon:
> ```

I do like this approach and I think that might work nicely.

> Regarding devices, do you plan to add `homeassistant` platform to light, climate and cover?

Not anytime soon, that will be a big task in itself.

> And finally, regarding groups, a binary sensor and a switch sensor per HA light or switch could do the job? like:

I had done something similar although skipped the template switch and just set the group item state directly from the `homeassistant` `binary_sensor` and then used a trigger on the group item to call the ha service directly.

jesserockz, 2021-12-22:
For the light `devices` though I was thinking ESPHome might need a `homeassistant` platform for light, which is why I disabled it completely for now.

mbo18, 2021-12-22:
@jesserockz 

> Ah something I added in my local config, but not in the PR for weather:

Nice but that need an automation in HA that runs every X minutes to stay updated

> I do like this approach and I think that might work nicely

I think that it should be easier :) with the mapping between HA weather states and NSPanel icons done in the component

> I had done something similar although skipped the template switch and just set the group item state directly from the `homeassistant` `binary_sensor` and then used a trigger on the group item to call the ha service directly.

Ah yes, homeassistant.turn_on or homeassistant.turn_off works with lights and switches :)

> For the light `devices` though I was thinking ESPHome might need a `homeassistant` platform for light, which is why I disabled it completely for now.

Yes that would be great but a lot of work I think

mbo18, 2021-12-22:
I have weather working fine now, thanks for the info about the service 👍

jesserockz, 2021-12-22:
I have pushed a commit to the PR that should fix the compile issues with the format_hex_pretty lines.

TyzzyT, 2021-12-22:
I added lines 98 and 99 of nspanel.cpp file again and can confirm that compiling with ESPHome stable version is working now.

nightweyr, 2021-12-25:
@jesserockz Thank you for the fantastic work on getting nsswitch panel into esphome! I am impressed that I can directly flash esphome on a brand new device without having to use Sonoff's firmware. I have the basic functionality setup, and I am trying to tweak it for my use case. I have a couple of questions. This is the first time I am trying to use esphome for a more complex application, and my unfamiliarity is probably showing.

1. Do you have an example of what the home assistant weather_data service should look like? I am playing with a few settings to see how I can get it to work, and I am struggling with it. This is my section from the nspanel yaml, I have tried a few iterations of a home assistant script, and I can't get it to show on the nspanel:
```yaml
# Enable Home Assistant API
api:
  password: !secret api_pw
  services:
    - service: weather_data
      variables:
        icon: int
        temperature: int
        min: int
        max: int
      then:
        - lambda: id(nspanel_id).send_weather_data((nspanel::WeatherIcon)icon, temperature, min, max);
```
2. I have a US version of NSPanel, and it is replacing a one gang ceiling light switch. To make everything cleaner, I am trying to map the relay to a binary light. I am trying to follow the [official docs](https://esphome.io/components/light/binary.html) which require the relay to be mapped to an output. The NSPanel code requires both relays to be [presented as a switch](https://github.com/esphome/esphome/pull/2702/files#diff-90cb7313b616f9d6e69e3358d14b0325f04255d8e84891a20431dd5eb57a7543R90) and I can't get the yaml config to be in a compilable form. One option I see is to create two instances of DemoSwitch and use then like this 
```yaml
nspanel:
  relays:
    - demo_switch_1
    - demo_switch_2
``` 
 Later in the config, I can use the config from binary_light to control the actual light. This seems hacky, so I was wondering if I am missing something obvious.

mbo18, 2021-12-25:
This is my HA automation for the weather data:
```
- alias: Send Weather to NSPanel
  id: Send Weather to NSPanel
  initial_state: true
  trigger:
    - platform: state
      entity_id: weather.home
      attribute: temperature
  condition:
    - condition: state
      entity_id: binary_sensor.nspanel_status
      state: "on"
  action:
    - service: esphome.nspanel_weather_data
      data:
        icon: >-
          {% set weather = states('weather.home') %}
          {% if weather == 'clear-night' %} {{ 1|int }}
          {% elif weather == 'cloudy' %} {{ 2|int }}
          {% elif weather == 'exceptionnal' %} {{ 30|int }}
          {% elif weather == 'fog' %} {{ 11|int }}
          {% elif weather == 'hail' %} {{ 24|int }}
          {% elif weather == 'lightning' %} {{ 15|int }}
          {% elif weather == 'lightning-rainy' %} {{ 15|int }}
          {% elif weather == 'partlycloudy' %} {{ 2|int }}
          {% elif weather == 'pouring' %} {{ 12|int }}
          {% elif weather == 'rainy' %} {{ 12|int }}
          {% elif weather == 'snowy' %} {{ 20|int }}
          {% elif weather == 'snowy-rainy' %} {{ 29|int }}
          {% elif weather == 'sunny' %} {{ 1|int }}
          {% elif weather == 'windy' %} {{ 32|int }}
          {% elif weather == 'windy-variant' %} {{ 32|int }}
          {% endif %}
        temperature: "{{ state_attr('weather.home', 'temperature') }}"
        min: "{{ state_attr('weather.home', 'forecast')[0].templow }}"
        max: "{{ state_attr('weather.home', 'forecast')[0].temperature }}"
```
with weather.home your weather entity.

Regarding the light, you can change the switch to a light in HA instead of ESPHome with this integration: https://www.home-assistant.io/integrations/light.switch/

nightweyr, 2021-12-25:
@mbo18 Thank you so much for the pointers for both items. I think I have everything needed to continue tweaking my yaml. Merry Christmas!

hermanndererdmann, 2021-12-26:
hi i'm having this issue when using the code from @jesserockz code.

[image]

any ideas?

samuelmcconnell, 2021-12-26:
Delete all the spaces on line 29 (\t is a tab, which is invalid per [YAML spec](http://yaml.org/spec/1.2-old/spec.html#id2777534)) and then add the spaces back with your space bar. That should clean it up.

jesserockz, 2021-12-28:
@nightweyr Good point there, I will look into making it work with `light`s too as I guess it will most likely be replacing dumb light switches.

queerdo, 2021-12-28:
Hey folks! Thanks so much for all of your work on this. I just got one of my NSPanels flashed this morning!

For context, I have 2x NSPanels, both are US vertical versions.

I tried setting the temp to F using `temperature_unit_celcius: false`, and the C changed to an F, but the temperature remained the same (e.g. without `temperature_unit_celcius: false` it shows 22 C and with `temperature_unit_celcius: false` it shows 22 F). In the HA entity, it correctly shows ~73 F.

Any ideas on how to convert the temp correctly?

mkaatman, 2021-12-28:
Is there a reference yaml configuration you guys are starting with? I finally flashed today and I'm in a boot loop. Trying to troubleshoot now.

mkaatman, 2021-12-28:
> Hey folks! Thanks so much for all of your work on this. I just got one of my NSPanels flashed this morning!
> 
> For context, I have 2x NSPanels, both are US vertical versions.

I too have two US NSpanels. Mind sharing the the yaml you used?

queerdo, 2021-12-28:
> I too have two US NSpanels. Mind sharing the the yaml you used?

Here's what I have so far. Note that I set the IP address manually, so you'd need to change that if you want to set a static IP. 

I had first flashed Tasmota onto mine, then decided to flash ESPHome. To do that, I had to download the config from ESPHome and flash using ESP-Flasher. After that, I've been able to flash wirelessly.

`esphome:
  name: bathroom-switch

esp32:
  board: esp32dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 33.0.40.229
    # Set this to the IP address of the router. Often ends with .1
    gateway: 33.0.40.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

time:
  - platform: homeassistant

api:
 services:
  - service: weather_data
    variables:
      icon: int
      temperature: int
      min: int
      max: int
    then:
      - lambda: id(nspanel_id).send_weather_data((nspanel::WeatherIcon)icon, temperature, min, max);

logger:

ota:

uart:
  tx_pin: 16
  rx_pin: 17
  baud_rate: 115200

external_components:
  - source: github://pr#2702
    components: ["nspanel"]

nspanel:
  id: nspanel_id
  time_id: 
  temperature: temperature
  eco_mode_switch: eco_mode
  screen_power_switch: screen_power
  temperature_unit_celcius: false
  relays:
    - relay_1
    - relay_2
  widgets:
    - type: scene
      name: MC
      on_click:
        - homeassistant.service:
            service: light.toggle
            data:
              entity_id: light.mission_control_lights
    - type: empty
    - type: empty
    - type: empty
    - type: empty
    - type: empty
    - type: empty
    - type: empty

binary_sensor:
  - platform: gpio
    name: Left Button
    pin:
      number: 14
      inverted: true
    on_click:
      - switch.toggle: relay_1

  - platform: gpio
    name: Right Button
    pin:
      number: 27
      inverted: true
    on_click:
      - switch.toggle: relay_2

output:
  - platform: ledc
    id: buzzer_out
    pin:
      number: 21

switch:
  - platform: gpio
    name: Relay 1
    id: relay_1
    pin:
      number: 22

  - platform: gpio
    name: Relay 2
    id: relay_2
    pin:
      number: 19

  - platform: gpio
    name: Screen Power
    id: screen_power
    entity_category: config
    pin:
      number: 4
      inverted: true
    restore_mode: ALWAYS_ON

  - platform: template
    name: NSPanel Energy Saving Mode
    id: eco_mode
    entity_category: config
    restore_state: true
    optimistic: true

rtttl:
  id: buzzer
  output: buzzer_out

sensor:
  - platform: adc
    id: ntc_source
    pin: 38
    update_interval: 10s
    attenuation: 11db

  - platform: resistance
    id: resistance_sensor
    sensor: ntc_source
    configuration: DOWNSTREAM
    resistor: 11.2kOhm

  - platform: ntc
    id: temperature
    sensor: resistance_sensor
    calibration:
      b_constant: 3950
      reference_temperature: 25°C
      reference_resistance: 10kOhm
    name: "Bathroom Switch Temperature"`

queerdo, 2021-12-28:
> This is my HA automation for the weather data:

@mbo18 Thanks for this automation! I tried setting it up in my HA, but I can't figure out what service to use. There doesn't appear to be any `esphome.***` services available. Am I missing a step to create that service?

mkaatman, 2021-12-28:
@queerdo your yaml is working for me. I'm having the degrees C issue as well.

jesserockz, 2021-12-28:
@queerdo use a filter on the temp sensor to convert it to F

queerdo, 2021-12-28:
@jesserockz excellent, thanks! That did the trick:

```
- platform: ntc
    id: temperature
    sensor: resistance_sensor
    calibration:
      b_constant: 3950
      reference_temperature: 25°C
      reference_resistance: 10kOhm
    name: "Entry Switch Temperature"
    filters:
      - lambda: return x * (9.0/5.0) + 32.0;
```

queerdo, 2021-12-28:
> Later in the config, I can use the config from binary_light to control the actual light. This seems hacky, so I was wondering if I am missing something obvious.

@nightweyr I may be misunderstanding what you're trying to do here - if it's getting the relay to show up as a light in HA then that's already been covered. However, if you're trying to get the physical buttons to control another entity besides the relays, then this may help.

I removed the `on_click` lines from both binary_sensor entries so that they aren't tied to the relays. This way, the buttons are standalone entities that show up in HA that can then be used in automations to control whatever you'd like.

```
binary_sensor:
  - platform: gpio
    name: Left Button
    pin:
      number: 14
      inverted: true

  - platform: gpio
    name: Right Button
    pin:
      number: 27
      inverted: true
```

mbo18, 2021-12-29:
@queerdo @mkaatman with latest code, GPIO restore mode for screen power must be off:
```
  - platform: gpio
    name: Screen Power
    id: screen_power
    entity_category: config
    pin:
      number: 4
      inverted: true
    restore_mode: ALWAYS_OFF

```

tubalainen, 2021-12-29:
Just ordered one!

Cannot wait for it to arrive.

mkaatman, 2021-12-29:
@mbo18 

Still learning how everything works. Does the latest code get pulled every time we build?

mbo18, 2021-12-29:
@mkaatman Clean build files for your NSPanel in the ESPHome addon then rebuild the firmware and upload it

mkaatman, 2021-12-29:
A couple of oddities.

1. Sudden reboot.
```
[E][api:123]: No client connected to API. Rebooting...
[I][app:127]: Forcing a reboot...
```

2. For the right button, it's not triggering my ha service. It does work from HA dev tools. Does this look generally correct? I see the button press in the log.
```
  - platform: gpio
    name: Right Button
    pin:
      number: 27
      inverted: true
    on_click:
      - homeassistant.service:
          service: light.toggle
          data:
            entity_id: light.night_lights
```

3. Item 2 may be related to when I click the icon I setup to toggle my lights, it complains network not connected. Network seems clearly connected.

4. I caught a couple of warnings during that compile if they're relevant.

```
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
```

5. My outdoor weather isn't working. If I understand this (https://github.com/esphome/feature-requests/issues/1469#issuecomment-1001065765) post correctly, I setup that automation in HA and then the weather_home Temperature attribute would have to change before the trigger would fire. Is there any trick to getting it to push the data initially when the NSPanel comes on?

queerdo, 2021-12-29:
> 2\. For the right button, it's not triggering my ha service. It does work from HA dev tools. Does this look generally correct? I see the button press in the log.

I'd recommend removing the `on_click` portion all together, and then setting up an automation triggered off of the button click in HA to toggle your light. More flexible/easier to change in the future that way.

queerdo, 2021-12-29:
> 5\. My outdoor weather isn't working. If I understand this ([Add support for ITead's SONOFF NSPanel touchscreen display (ESP32 with custom Nextion HMI screen)? #1469 (comment)](https://github.com/esphome/feature-requests/issues/1469#issuecomment-1001065765)) post correctly, I setup that automation in HA and then the weather_home Temperature attribute would have to change before the trigger would fire. Is there any trick to getting it to push the data initially when the NSPanel comes on?

I removed the condition from my automation and it started working. Also you could add an additional trigger to start the automation when the screen turns on or when a button is pushed

queerdo, 2021-12-29:
> * Item 2 may be related to when I click the icon I setup to toggle my lights, it complains network not connected. Network seems clearly connected.
> *

Getting this on mine too when I toggle the screen power off and back on via HA. To fix, just repush the firmware from ESPHome

TyzzyT, 2021-12-29:
> 5. My outdoor weather isn't working. If I understand this ([Add support for ITead's SONOFF NSPanel touchscreen display (ESP32 with custom Nextion HMI screen)? #1469 [...]

I did it like this:
```
esphome:
  on_boot:
    priority: -100
    then:
      - wait_until:
          api.connected
      - homeassistant.service:
          service: automation.trigger
          data:
            entity_id: automation.nspanel_send_weather_to_nspanel
```

mbo18, 2021-12-29:
Set restore_mode to ALWAYS_OFF to solve those issues (network not connected)

mkaatman, 2021-12-29:
Here's what I've got.

```
  - platform: gpio
    name: Screen Power
    id: screen_power
    entity_category: config
    pin:
      number: 4
      inverted: true
    restore_mode: ALWAYS_OFF

masto, 2021-12-29:
FWIW, an alternative approach is to replace Sonoff's interface with a custom one made in the Nextion editor, and then just treat it as a generic Nextion display. For that you'll need a small fix to be able to successfully upload the new TFT file: https://github.com/esphome/esphome/pull/2956

mkaatman, 2021-12-29:
> FWIW, an alternative approach is to replace Sonoff's interface with a custom one made in the Nextion editor, and then just treat it as a generic Nextion display. For that you'll need a small fix to be able to successfully upload the new TFT file: [esphome/esphome#2956](https://github.com/esphome/esphome/pull/2956)

With that change you can use the standard editor? It looks pretty cool! https://nextion.tech/editor_guide/

Someone should do a writeup or video on this. :)

mbo18, 2021-12-30:
@TyzzyT 
> I did it like this:

Does this really work for you? It does not work for me, service is never called. It's like everything I add after `on_boot` is not executed

TyzzyT, 2021-12-30:
@mbo18 It's working for me. If I manually set the weather data to something not correct, reboot the NSPanel, then it comes back online with the correct data and I see the automation last triggered timestamp being updated.

But I have a 10 seconds delay before the wait_until api.connected condition. Not sure if that is why it works for me, but it's worth trying at your side.
```
      - delay: 10s
      - wait_until:
          api.connected
```

Probably a stupid question, but did you change the automation.entity_idto your automation entity_id? I do believe it's different then yours.

mbo18, 2021-12-30:
@TyzzyT 
> But I have a 10 seconds delay before the wait_until api.connected condition.

Strange but adding a 10s delay solves the issue :) Thanks!

mkaatman, 2021-12-30:
Has anyone noticed any oddities with the hardware itself? My left button is really intermittent to toggle the relay. ESPHome sees every button press. I'm almost considering bypassing the relay and have HA toggle the switch on button press.

jesserockz, 2021-12-30:
@TyzzyT @mbo18 Watching the logs via ESPHome counts as an api connection and `api.connected` will be true.

Another way to do it is from HA automation directly using this binary sensor going from off to on as the trigger.

```yaml
binary_sensor:
  - platform: status
    ...
```

@mkaatman The temperature sensor still needs more precise calibration. My panel came with a shield over the circuitry so I could not actually measure it properly and took a guess based on info others had found and from what my LYWSD03MMC showed too.

[85 more replies]

closed 2024-04-02 by nagyrobi