esphome/feature-requests #1486: DDP Support for Lights

bkaufx, 2021-10-29:
**Describe the problem you have/What new integration you would like**

DDP support in ESPHome would be nice so that WLED could control an ESPHome device as though the ESPHome device were part of a longer strip. 

**Please describe your use case for this integration and alternatives you've tried:**

As I understand WLED support in ESPHome now, WLED and ESPHome can sync to the same color/effect but they aren't concatenated into one long string like most people would probably want.  WLED's DDP support is more like e1.31.

**Additional context**

Thread from WLED about joining multiple devices into one long strip:
https://github.com/Aircoookie/WLED/issues/338#issuecomment-948572762

More on DDP from WLED:
https://github.com/Aircoookie/WLED/issues/478

jondavidschober, 2021-12-28:
I 2nd this. I use ESPHome in E1.31 mode to run an indoor light show from xLights. DDP would be a great improvement

bkaufx, 2022-12-12:
If anyone sees this, I started working on a DDP component and you can try out an early version here:

https://github.com/KaufHA/common/blob/main/components/README.md

TheAndi, 2023-03-12:
@bkaufx I use your component for a while now and it is awesome! I have 432pixel per light, so using e1.31 is not an option.
I use it for [Hyperion ](https://github.com/hyperion-project/hyperion.ng) and [LedFx](https://github.com/LedFx/LedFx) and it works great. I hope it can be integrated in ESPhome natively soon! Keep up the great work :)

Marcin685, 2025-08-02:
@bkaufx I'm trying to run a white channel on your add-on, but unfortunately it doesn't work. RGB works fine. Maybe you have a solution?

tanishqmanuja, 2025-08-02:
@Marcin685 I had similar issue so I modified kauf's ddp component

https://github.com/tanishqmanuja/esphome-components

Bambi66669, 2025-08-02:
[Standard contract template _ Microsoft Learn.PDF](https://github.com/user-attachments/files/21559141/Standard.contract.template._.Microsoft.Learn.PDF)
[ProductTerms(EAEASSCE)(English)(08012025).docx](https://github.com/user-attachments/files/21559140/ProductTerms.EAEASSCE.English.08012025.docx)
[OnlineSvcsConsolidatedSLA(WW)(English)(July_2025)(CR).docx](https://github.com/user-attachments/files/21559142/OnlineSvcsConsolidatedSLA.WW.English.July_2025.CR.docx)

tanishqmanuja, 2025-08-02:
@Bambi66669 Relevance ?

tanishqmanuja, 2025-08-02:
Are you adding the external component from my repo ?

Marcin685, 2025-08-02:
I control an RGBW bulb based on the bk7231n chipset using the WLED app, and I have configured DDP RGBW (network) for each bulb. Control only works for RGB, but not for W (white). I'd like to be able to control all the LEDs in the bulb from the WLED app. I've recorded a video below showing this control working perfectly, but this bulb has different software installed (OpenBK7231T_App). However, I'd still like to use ESPHOME.

https://github.com/user-attachments/assets/4ccaadde-1aab-4403-9fe2-d257a92d87e2

tanishqmanuja, 2025-08-02:
@Marcin685 thanks for the video, now i fully understand the problem. Try to code like this once (has rgbww) https://github.com/tony-fav/esphome-sync/blob/018d1f0a4f062c695fefbf88ff45930f061a522f/ddp/five-channel-bulb.yaml

Kauf's ddp supports rgb (3 channel)
My modification supports monochromatic (1 channel)
But you need 4 channel support, should be easy to add in kauf's ddp but try the above direct code first.