esphome/feature-requests #1487: DLMS component like the DSMR component to read data from SmartMeter

deejaybeam, 2021-10-30:
**Describe the problem you have/What new integration you would like**
Tried the DSMR integration to read my smartmeter, but it looks like there is an other but similar protocol in use for my Siemens IM350

**Please describe your use case for this integration and alternatives you've tried:**
@mmakaay does already some investigation, because I tried to use the DSMR integration:
 https://github.com/esphome/issues/issues/2600
An other project works, but I would prefer the ESPHome solution, additionaly this would also work with this project:
https://www.zuidwijk.com/product/slimmelezer/
Maybe it would be possible to use nearly the same yaml just exchange dsmr with dlms.
A working alternative is this:
https://github.com/saghonfly/shrdzm/wiki/Siemens-IM350,-Iskra-AM550
https://github.com/saghonfly/shrdzm

**Additional context**
It looks like both protocols are very similar and can run on the same (existing) hardware. So there is an additional component with minimal changes (in esp-yaml file) to use with a larger range of Smartmeter vendors to use the new features of home assistant energy management:
https://www.home-assistant.io/blog/2021/08/04/home-energy-management/

mmakaay, 2021-10-30:
The code in the linked project that implements the protocol can be found in:
https://github.com/saghonfly/shrdzm/blob/master/code/SHRDZMDevice/Device_IM350.cpp

AdmiralStipe, 2021-11-17:
Maybe [this](https://icube.ch/Security/ciphered-context-GMAC-example.html) can also be helpful (cyphering), as some meters have encryption and authentication enabled on P1 port (f.e. Iskraemeco AM550 in Slovenia does).

AdmiralStipe, 2021-11-19:
Additional link with ESP8266 firmware, which partly worked for me (getting the data, but unable to decrypt it).
[https://github.com/NielsOerbaek/Wemos-D1-Mini-Omnipower-Reader](https://github.com/NielsOerbaek/Wemos-D1-Mini-Omnipower-Reader)

zigapovhe, 2022-05-04:
> The code in the linked project that implements the protocol can be found in: https://github.com/saghonfly/shrdzm/blob/master/code/SHRDZMDevice/Device_IM350.cpp

Link is not working anymore because the owner deleted the repo. Luckily, we have a identical fork here: https://github.com/Noschvie/shrdzm/blob/master/code/SHRDZMDevice/Device_IM350.cpp

clutwo, 2022-09-02:
hi everyone! is this issue being worked on, would be really nice to have the DLMS integration to use with the slimmelezer!

cheers

BeniKing99, 2023-08-08:
Are there any updates on this? It would be really great!

SimonFischer04, 2024-04-28:
@BeniKing99 @clutwo and others. I'm currently working on it see: https://github.com/SimonFischer04/esphome/issues/1

latonita, 2024-09-25:
I'm working on DLMS COSEM in binary HDLC mode.  (IEC 62056 Mode E)
Connecting via RS485/RS232.
Works with direct binary conneciton, no support for Ascii IEC handshake yet (don't have meter to test)
https://github.com/latonita/esphome-dlms-cosem

evlo, 2025-09-05:
@latonita  I probably do not understand how to use your component - i tried it, but it always try to send data to the meter, while my meter just pushes DLMS data onto me every minute. Can it be made to work with your component?

closed 2026-01-31 by swoboda1337