esphome/feature-requests #1488: ESP device receives UART text and realizes speech recognition automation

phpphph, 2021-10-31:
I also found an interesting offline speech recognition module. The communication protocol of this module is UART;
1: The voice recognition module model is "SU-03T", the serial port of the module supports input and output of "AA 12 CD" data (hexadecimal) (example)
example:
Say "turn on the light" and the serial port outputs "AA 11 11"
Say "turn off the light" the serial port outputs "AA 11 00"
2: Use this SU-03T module to control the equipment in each room or the entire home.
Imagine how to connect it to the ESP device?
I want to automate Home Assistant.
Similar to Home Assistant listening to the message "AA 11 **" issued by the ESP device, it performs certain actions after receiving a defined message.
[image]

nagyrobi, 2021-10-31:
Understands Chinese and English only?

phpphph, 2021-10-31:
> Understands Chinese and English only?

I only understand Chinese.

kgov777, 2022-10-01:
Hi does the SU-03T module do english?

Nerd-01, 2022-11-03:
Hello ... Is it possible to customize (English) the vocabulary and where can datasheets and software be found ? Can I use 5 V? Thanks for any answers ...

phpphph, 2022-11-04:
> Hello ... Is it possible to customize (in English) the vocabulary (or where is the list of the standard expressions ? ) and where can software be found if necessary ? Thanks for any answers ... I found a datasheet for the hardware (in Chinese, I performed an automatic translation). I wanna use the board with a DIY ATMega board (NOT Arduino) programmed with GC Basic, to control a model railroad, so I must know explicitely the communication protocol.
嗨，su-03tWith 5v power supply,Serial communication level is 3.3v ..Pay attention to the current of his power supply.Total power depends on speaker output power.

Nerd-01, 2022-11-04:
No problem for the serial levels, I have level converters. But what about the vocabulary (fixed, then : what's the list, or customizable, how ? ). And I won't probably not use the loudspeaker, only the recognition function : Speech synthesis can quickly become boring !

phpphph, 2022-11-05:
> 
The offline wake-up of su03t is very useful. I contacted the manufacturer and they expect to launch the WIFI+BLE version of voice recognition by the end of the year. It is also possible to customize the firmware and configuration content on the official website, and development is not difficult.

phpphph, 2022-11-05:
I use serial communication more, because it can achieve more functions and richer sentence control, and you can also let it speak, you can send a variable value to its serial port, such as "the temperature is 25 degrees Celsius now" . Just using io to control some functional deficiencies.

Nerd-01, 2022-11-05:
"The official website" : WHERE, PLEASE, IS IT ? I found the above list, but am not sure it's the official one ...

phpphph, 2022-11-05:
> "The official website" : WHERE, PLEASE, IS IT ? I found the above list, but am not sure it's the official one ...

Google→→“机芯智能 SU-03T”Or→→"http://docs.aimachip.com/zh_CN/latest/"

Nerd-01, 2022-11-05:
Sorry, too many parts in Chinese, impossible to translate them all  ... I will look for another module !‌
 

‌‌‌‌Meilleures Salutations / Freundliche Grüße / Kind Regards,
B. Baroth

De : "phpphph"
A : "esphome/feature-requests" ,"Nerd-01" ,"Comment"
Envoyé: samedi 5 Novembre 2022 13:48
Objet : Re: [esphome/feature-requests] ESP device receives UART text and realizes speech recognition automation (Issue #1488)
 

 

"The official website" : WHERE, PLEASE, IS IT ? I found the above list, but am not sure it's the official one ...

Google→→“机芯智能 SU-03T”Or→→"http://docs.aimachip.com/zh_CN/latest/"

Gio-dot, 2023-05-25:
> > 
> 
> The offline wake-up of su03t is very useful. I contacted the manufacturer and they expect to launch the WIFI+BLE version of voice recognition by the end of the year. It is also possible to customize the firmware and configuration content on the official website, and development is not difficult.

Hi, this module is already provided of a standard firmware out of the box? If yes, provided firmware let to customise commands? Sorry but official sit is in Chinese and not all is translate well automatically.

Akovognon2003, 2023-05-25:
Hi. How are you ? I too am in the same situation as you. It was not me who
wrote the message in Github but someone who has a nickname phphph I
believe. I have the SU-03T module but I don't know how to use it because
all its software and documentations are in Chinese and as you pointed out
earlier, not everything is translated automatically.

ABC756431, 2023-06-02:
非常棒的想法

ROBOLAB5, 2024-04-12:
I just got SU-03T how can I change it from Chinese to English

phpphph, 2024-04-13:
> 你好。你好吗 ？我也和你的情况一样。我相信在 Github 上写这条消息的不是我，而是一个昵称是 phpph 的人。我有 SU-03T 模块，但我不知道如何使用它，因为它的所有软件和文档都是中文的，正如您之前指出的，并非所有内容都会自动翻译。游戏。 2023 年 5 月 25 日 à 22:32，Giovanni ***@***.***> 一条评论：
> [……](#)

Hi, I'm concerned about unread news now. Have you solved your problem?

phpphph, 2024-04-13:
> 我刚拿到SU-03T，如何将其从中文更改为英文

phpphph, 2024-04-13:
> > 我刚拿到SU-03T，如何将其从中文更改为英文

Hi, you can set the language through the official website of 03t module.

ROBOLAB5, 2024-04-19:
Thank you for your response, there's power outage here at my place.

Most responses are in Chinese and I only understand English

Can I have step by step on how to set it from Chinese to English?
Thank you

ROBOLAB5, 2024-04-19:
Thank you for your email
Please can I have step by step on how to set it on English?

phpphph, 2024-04-19:
> Thank you for your email Please can I have step by step on how to set it on English?
> […](#)

Open website [https://smartpi.cn/](https://smartpi.cn/) . Then, according to the screenshot I provided you, set the module language. "英文" is in English. After entering the firmware settings, I suggest that you use Google Translate webpage for operation,    You also need to register an account
The first step

[image]

Step 2
[image]

Step 3
[image]

Gdeeeee, 2024-04-19:
Can i connect that to Arduino Uno?

phpphph, 2024-04-19:
> Can i connect that to Arduino Uno?

Of course, su-03t can output through serial port and set more voice commands

Gdeeeee, 2024-04-19:
> > Can i connect that to Arduino Uno?
> 
> Of course, su-03t can output through serial port and set more voice commands

can you give me the proper connection and how to set commands using arduino? I spent my time searching for tutorials but I cant find one. It well help me a lot for my school project. Thankyou in advance.

phpphph, 2024-04-19:
> can you give me the proper connection and how to set commands using arduino? I spent my time searching for tutorials but I cant find one. It well help me a lot for my school project. Thankyou in [...]

You can click on this link, which is the English firmware setting method I just shared with another guy today. After entering the firmware settings, it is best to use Google Translate in Google Chrome to translate web pages.
https://github.com/esphome/feature-requests/issues/1488#issuecomment-2065899403

ROBOLAB5, 2024-04-19:
> 。您收到此消息是因为您发表了评论。消息 ***@***.***> Hi, I'm concerned about unread news now. Have

Thank you for your email it's really helpful

gadingnst, 2025-01-28:
@phpphph I've already downloaded the firmware, now how can I upload it to module? what software I'm must use?

[image]

phpphph, 2025-02-08:
> [@phpphph](https://github.com/phpphph) I've already downloaded the firmware, now how can I upload it to module? what software I'm must use?
> 
> [image]

The su03t official website has it, including module manual, chip data, burning tutorial, use CH340 burner, burning software is also available on the official website