esphome/feature-requests #1490: Add Support for the Olimex ESP32-S2

RAYs3T, 2021-10-31:
the Olimex board is currently not supported. Choosing the `esp32-s2-saola-1` instead does produce a boot-loop as the chip seems to differ
```
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32-S2
Connecting.....

Chip. (96 %)Writing at 0x000b98e1... (100 %)Wrote 699504 bytes (462248 compressed) at 0x00010000 in 11.2 seconds (effective 498.2 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[20:36:05]ESP-ROM:esp32s2-rc4-20191025
[20:36:05]Build:Oct 25 2019
[20:36:05]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[20:36:05]Saved PC:0x4000f79a
[20:36:05]Invalid chip id. Expected 2 read 0. Bootloader for wrong chip?
[20:36:05]ets_main.c 386 
[20:36:06]ESP-ROM:esp32s2-rc4-20191025
[20:36:06]Build:Oct 25 2019
[20:36:06]rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[20:36:06]Invalid chip id. Expected 2 read 0. Bootloader for wrong chip?
[20:36:06]ets_main.c 386
```

Also tried it with `nodemcu-32s` but this produces the same boot error:
```
[20:47:30]ESP-ROM:esp32s2-rc4-20191025
[20:47:30]Build:Oct 25 2019
[20:47:30]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[20:47:30]Saved PC:0x4000f79a
[20:47:30]Invalid chip id. Expected 2 read 0. Bootloader for wrong chip?
[20:47:30]ets_main.c 386
```

**Describe the problem you have/What new integration you would like**

Support esphome on the `Olimex ESP32-S2`

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

Manufacturer Page: https://www.olimex.com/Products/IoT/ESP32-S2/ESP32-S2-DevKit-Lipo/open-source-hardware

gatti-d, 2022-01-08:
Unfortunately I have the same issue with an ESP32-s2 Ai-Thinker board....

nagyrobi, 2022-07-01:
https://github.com/esphome/feature-requests/issues/996
https://github.com/esphome/esphome/pull/3264

closed 2022-07-01 by nagyrobi