esphome/feature-requests #1491: Please add DSMR support for Sanxing SX6x1 meters (E.on - Émász Hungary)

nagyrobi, 2021-10-31:
**Describe the problem you have/What new integration you would like**
Please add DSMR support for Sanxing SX6x1 smart meters (E.on - Émász Hungary).
Sends data at 115200 baud every 10 seconds.

**Please describe your use case for this integration and alternatives you've tried:**
Works with [Tasmota](https://tasmota.github.io/docs/Smart-Meter-Interface/#sanxing-sx6x1-sxxu1x-ascii-obis) - see for OBISes and example data.

**Additional context**
OBIS codes description (same): [E.on](https://www.eon.hu/content/dam/eon/eon-hungary/documents/Muszaki-ugyek/Fogyasztasmerok-leirasa/sanxing-sx6x1/EON-leiras-az-ugyfelek-szamara-SX6X1-S12U16-S34U18-V010.pdf) and [Émász](https://mvmemaszhalozat.hu/elmu/file/downloadfile?id=e7d5d332-ce9a-483d-9dc8-7baf0fe42ab1).

nagyrobi, 2022-05-31:
@zuidwijk

nagyrobi, 2022-05-31:
I originally submitted this request to your [dsmr repo](https://github.com/zuidwijk/dsmr) just before it was merged into ESPHome and got a promise back then that support for this energy provider will be added, but lets wait until it gets merged into ESPHome first. I was asked to re-submit the request here, and the original issue was closed (Issues list disabled in your repo so can't point at it at the moment).

Got merged in, but support was not added as promised :-(

nagyrobi, 2022-05-31:
I'd love to support your work by purchasing a couple of SlimmeLezer+ :D

nagyrobi, 2022-07-01:
Hi @mmakaay can you please have a look at this when you have some time?

kizmanj, 2022-07-04:
Thank you in advance sir! Looking forward to it!

kissgy85, 2023-02-21:
Please add MVM Démász Áramhálózati Kft.	
Official OBIS codes:

1F mérő SX601 	
Idő	8, 0-0:1.0.0.255
COSEM logikai eszköznév	1, 0-0:42.0.0.255
Eszköz ID 1: Mérő sorozatszám	1, 0-0:96.1.0.255
Jelenleg aktív tarifa	1, 0-0:96.14.0.255
Megszakító állapota	1, 0-0:96.50.68.255
Megszakító limiter küszöbértéke	71, 0-0:17.0.0.255
Hatásos import energia (+A)	3, 1-0:1.8.0.255
Hatásos import energia (+A) – tarifa 1	3, 1-0:1.8.1.255
Hatásos import energia (+A) – tarifa 2	3, 1-0:1.8.2.255
Hatásos export energia (-A)	3, 1-0:2.8.0.255
Meddő import energia (+R)	3, 1-0:3.8.0.255
Meddő export energia (-R)	3, 1-0:4.8.0.255
Pillanatnyi feszültség L1	3, 1-0:32.7.0.255
Pillanatnyi áram L1	3, 1-0:31.7.0.255
Pillanatnyi teljesítménytényező L1	3, 1-0:33.7.0.255
Pillanatnyi hatásos import teljesítmény (+A)	3, 1-0:1.7.0.255
Pillanatnyi hatásos export teljesítmény (-A)	3, 1-0:2.7.0.255
Számlázási adatok 1. periódus (utolsó bejegyzés)	7, 0-0:98.1.0.255
Szöveges üzenet max 1024 karakter	1, 0-0:96.13.0.255
	
3F mérő SX631	
Idő	8, 0-0:1.0.0.255
COSEM logikai eszköznév	1, 0-0:42.0.0.255
Eszköz ID 1: Mérő sorozatszám	1, 0-0:96.1.0.255
Jelenleg aktív tarifa	1, 0-0:96.14.0.255
Megszakító állapota	1, 0-0:96.50.68.255
Megszakító limiter küszöbértéke	71, 0-0:17.0.0.255
Hatásos import energia (+A)	3, 1-0:1.8.0.255
Hatásos import energia (+A) – tarifa 1	3, 1-0:1.8.1.255
Hatásos import energia (+A) – tarifa 2	3, 1-0:1.8.2.255
Hatásos export energia (-A)	3, 1-0:2.8.0.255
Meddő import energia (+R)	3, 1-0:3.8.0.255
Meddő export energia (-R)	3, 1-0:4.8.0.255
Pillanatnyi feszültség L1	3, 1-0:32.7.0.255
Pillanatnyi feszültség L2	3, 1-0:52.7.0.255
Pillanatnyi feszültség L3	3, 1-0:72.7.0.255
Pillanatnyi áram L1	3, 1-0:31.7.0.255
Pillanatnyi áram L2	3, 1-0:51.7.0.255
Pillanatnyi áram L3	3, 1-0:71.7.0.255
Pillanatnyi teljesítménytényező L1	3, 1-0:33.7.0.255
Pillanatnyi teljesítménytényező L2	3, 1-0:53.7.0.255
Pillanatnyi teljesítménytényező L3	3, 1-0:73.7.0.255
Pillanatnyi hatásos import teljesítmény (+A)	3, 1-0:1.7.0.255
Pillanatnyi hatásos export teljesítmény (-A)	3, 1-0:2.7.0.255
Számlázási adatok 1. periódus (utolsó bejegyzés)	7, 0-0:98.1.0.255
Szöveges üzenet max 1024 karakter	1, 0-0:96.13.0.255

dbius, 2023-03-04:
Would be someone so nice and helpful to help me making a working config file? I live also in Hungary, my meter is a Sanxing SX631 (E.ON and Opusz Titász).

Basically the slimmelizer+ works, but the meter provides more data than slimmelezer logging with the default config file and I don't know how to modfiy the values.

Sanxing SX631 [Hungarian user manual](https://www.eon.hu/content/dam/eon/eon-hungary/documents/Muszaki-ugyek/Fogyasztasmerok-leirasa/sanxing-sx6x1/EON-leiras-az-ugyfelek-szamara-SX6X1-S12U16-S34U18-V010.pdf) describes (on page 19 and 20):
- 115200 Baud-os rate communication,
- 8N1 data format
- 1 start bit
- 8 adat bit
- parity none
- 1 stop bit

The OBIS codes are:

Idő 0-0:1.0.0.255
COSEM logikai készüléknév 0-0:42.0.0.255
Mérő gyáriszám 0-0:96.1.0.255
Aktuális tarifa 0-0:96.14.0.255
Megszakító státusz 0-0:96.50.68.255
Limiter határérték 0-0:17.0.0.255
Hatásos import energia (+A) 1-0:1.8.0.255
Hatásos import energia (+A) – tarifa 1 1-0:1.8.1.255
Hatásos import energia (+A) – tarifa 2 1-0:1.8.2.255
Hatásos import energia (+A) – tarifa 3 1-0:1.8.3.255
Hatásos import energia (+A) – tarifa 4 1-0:1.8.4.255
Hatásos export energia (-A) 1-0:2.8.0.255
Hatásos export energia (-A) – tarifa 1 1-0:2.8.1.255
Hatásos export energia (-A) – tarifa 2 1-0:2.8.2.255
Hatásos export energia (-A) – tarifa 3 1-0:2.8.3.255
Hatásos export energia (-A) – tarifa 4 1-0:2.8.4.255
Import meddő energia (+R) 1-0:3.8.0.255
Export meddő energia (-R) 1-0:4.8.0.255
Meddő energia (QI) 1-0:5.8.0.255
Meddő energia (QII) 1-0:6.8.0.255
Meddő energia (QIII) 1-0:7.8.0.255
Meddő energia (QIV) 1-0:8.8.0.255
Hatásos energia kombinált (|+A|+|-A|) 1-0:15.8.0.255
Pillanatnyi fázis feszültség L1 1-0:32.7.0.255
Pillanatnyi fázis feszültség L2 (csak 3fázisú mérő esetén) 1-0:52.7.0.255
Pillanatnyi fázis feszültség L3 (csak 3fázisú mérő esetén) 1-0:72.7.0.255
Pillanatnyi áram L1 1-0:31.7.0.255
Pillanatnyi áram L2 (csak 3 fázisú mérő esetén) 1-0:51.7.0.255
Pillanatnyi áram L3 (csak 3 fázisú mérő esetén) 1-0:71.7.0.255
Pillanatnyi teljesítmény tényező 1-0:13.7.0.255
Pillanatnyi teljesítmény tényező L1 1-0:33.7.0.255
Pillanatnyi teljesítmény tényező L2 (csak 3fázisú mérő esetén)1-0:53.7.0.255
Pillanatnyi teljesítmény tényező L3 (csak 3fázisú mérő esetén)1-0:73.7.0.255
Frekvencia 1-0:14.7.0.255
Pillanatnyi import teljesítmény (+A) 1-0:1.7.0.255
Pillanatnyi export teljesítmény (-A) 1-0:2.7.0.255
Pillanatnyi meddő teljesítmény (QI) 1-0:5.7.0.255
Pillanatnyi meddő teljesítmény (QII) 1-0:6.7.0.255
Pillanatnyi meddő teljesítmény (QIII) 1-0:7.7.0.255
Pillanatnyi meddő teljesítmény (QIV) 1-0:8.7.0.255
Áram korlátozás határérték 1 1-0:31.4.0.255
Áram korlátozás határérték 2 (csak 3 fázisú mérő esetén)1-0:51.4.0.255
Áram korlátozás határérték 3 (csak 3 fázisúmérő esetén) 1-0:71.4.0.255
Hónap végi tárolt adatok (utolsó havi) 0-0:98.1.0.255
Áramszolgáltatói szöveges üzenet 0-0:96.13.0.255

Thank you guys!

Alex07790, 2025-07-16:
Any update on this?