esphome/feature-requests #1493: Control external WLED from ESPHome

muxa, 2021-11-01:
**Describe the problem you have/What new integration you would like**
I want to be able to control addressable LEDs connected to an external WLED (similar to how it's possible to have ESPHome node with [WLED Effect](https://esphome.io/components/light/index.html#wled-effect) controlled by an external WLED, only I want to reverse the control. 

**Please describe your use case for this integration and alternatives you've tried:**
I'm working on a MIDI integration for ESPHome when MIDI notes light up an addressable LED strip. I want to extend it so that the the lights don't have to be directly attached to the node to be controlled by MIDI. 

**Additional context**
I know there are 2 types of network control for LEDs that ESPHome supports: [WLED UDP](https://esphome.io/components/light/index.html#wled-effect) and [E1.31](https://esphome.io/components/light/index.html#e1-31-component). Any tips on how to best approach the implementation of the reverse control will be appreciated.

oxan, 2021-11-10:
I think the way to go about this is to implement a WLEDLightOutput that talks to a WLED node.

muxa, 2021-11-11:
I might give it a go implementing `WLEDLightOutput` according to https://github.com/Aircoookie/WLED/wiki/UDP-Realtime-Control

muxa, 2022-05-29:
I got it working: https://github.com/muxa/esphome-wled-output. It was surprisingly easy to implement. 
The project that this functionality is needed in is still in progress, so wont adds this to ESPHome yet. I want to make sure things are working well first.

nagyrobi, 2022-07-05:
Fell free to submit as PR.

closed 2022-07-05 by nagyrobi