esphome/feature-requests #1494: UART: support sending of break signal

andyboeh, 2021-11-01:
**Describe the problem you have/What new integration you would like**

I am working on a DMX-512 implementation based around the MAX485 IC. The UART can be used for sending data, but a break signal of a specific length needs to be sent before starting the transmission. Unfortunately, this is not supported by the UART component.

**Please describe your use case for this integration and alternatives you've tried:**

I suggest a new `sendBreak(usecs)` method that allows sending of the break signal.

**Additional context**

There are a few workaround options that I could find for different controllers that involve either changing the baud rate on-the-fly and sending a 0 byte or detaching the Tx pin and driving it manually for the break generation. However, that is hardware-specific and can't be done in the DMX component.

martgras, 2021-11-03:
Probably the idf function `uart_write_bytes_with_break` is what you need. But I don't know if it can be used in esphome.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#_CPPv427uart_write_bytes_with_break11uart_port_tPKv6size_ti

andyboeh, 2021-11-04:
Thanks for the answer and adding this possibility. Unfortunately, according to the docs, it doesn't work without sending data. For DMX, the break has to be sent first. 

I think that I'll still be able to come up with a portable solution, though:

  * ESP32 Arduino: `pinMatrixOutDetach`, set the pin, wait, clear the pin, `pinMatrixOutAttach`
  * ESP32 IDF: `uart_set_line_inverse`, wait, `uart_set_line_inverse`
  * ESP8266: `SET_PERI_REG_MASK`, delay, `CLEAR_PERI_REG_MASK`

Tuckie, 2021-12-04:
Thank you for starting to dig into this. Adding this functionality seems a lot cleaner for the dmx-512 implementation.

pruwait, 2023-09-22:
> * ESP32 Arduino: `pinMatrixOutDetach`, set the pin, wait, clear the pin, `pinMatrixOutAttach`

can you write an example of use in the esphpme component?

andyboeh, 2023-09-22:
Check https://github.com/andyboeh/esphome-dmx512, there are separate files for ESP32, ESP8266 and ESP32IDF