esphome/feature-requests #1495: Shared addressable effects

muxa, 2021-11-04:
**Describe the problem you have/What new integration you would like**
I want to be able to create custom addressable effects so that they can be:

- shared between different lights
- packaged up and published as external components to share them between different projects

**Please describe your use case for this integration and alternatives you've tried:**
I'm working on a MIDI integration, which I want to control LED lights. I want to have the logic of mapping MIDI events to LEDs in an effect module, so that a single addressable light can have multiples effects, including multiple MIDI effects.
  
**Additional context**
An idea to made this work:

```yaml
addressable_effects:
  - platform: midi # I want to create this this component
     id: random_midi_fx
     start_note: "0x15"
     note_on_fade: 500ms
     note_off_fade: 5s
     palette: random # rainbow, white, ...
  - platform: midi # I want to create this this component
     id: rainbow_midi_fx
     start_note: "0x15"
     note_on_fade: 500ms
     note_off_fade: 5s
     palette: rainbow.

light:
  - platform: fastled_clockless
  - ...
  - effects:
    - addressable_component: # this does not exist
        name: "MIDI Random"
        id: random_midi_fx
    - addressable_component: # this does not exist
        name: "MIDI Rainbow"
        id: rainbow_midi_fx
```

oxan, 2021-11-10:
I'm working on custom effects, but life has been busy lately... If you're feeling adventurous, the [light/transitions](https://github.com/oxan/esphome/tree/light/transitions) branch in my repository should be somewhat usable.

tungmeister, 2021-11-14:
@oxan will this offer the ability to create, say, a twinking random color effect on a single light?

oxan, 2021-11-14:
Wait, I misread, this is about effects and not transitions. For effects it's already possible to do this.

tungmeister, 2021-11-14:
@oxan, are you saying it's already possible to create a twinkle effect that changes color at random on a single rgb light? If so I couldn't see any way to achieve this via the effects within the light component.

oxan, 2021-11-14:
Yes. You can do it either with an [addressable lambda](https://esphome.io/components/light/index.html#addressable-lambda-effect) from YAML, or from a component by using one of the `register_*_effect` decorators (from `esphome.components.light.effects`, see e.g. the `adalight` component for an example).

tungmeister, 2021-11-14:
Thanks, I'll take a look 👍

muxa, 2021-11-15:
Cool, ESPHome already supports what I want to do, looking at example here: https://github.com/esphome/esphome/tree/dev/esphome/components/adalight 

So my custom effects are going to look something like this:

```yaml
midi_in: # MIDI In support is being added here: https://github.com/esphome/esphome/pull/2698
   - id: midi1

light:
  - platform: fastled_clockless
    ...
    effects:
      - addressable_rainbow: # this exists
      - midi_effect: # this is what I want to build
        name: "MIDI Random"
        midi_in_id: midi1
        start_note: "0x15"
        keys: 88
        note_on_fade: 500ms
        note_off_fade: 5s
        palette: random
```

closed 2021-11-15 by muxa