esphome/feature-requests #1497: Add support of SSD1306 64x32 0.42' OLED display to SSD1306 base Component and Add mirror/flip function on horizontal / vertical directions to Display Component

ychieux, 2021-11-04:
**Describe the problem you have/What new integration you would like**

The smallest size of LCD display supported to date by the SSD1306 component is 64x48. I am currently using an offset of 16 with the model "SSD1306 64x48"
Upon reviewing the source ssd1306_base.h and ssd1306_base.cpp, a new model can be added to SSD1306Model as SSD1306_MODEL_64_32

Since the display is projected thru a lens, the projected image is flipped horizontally. The parameters flip_x and flip_y are ineffective for this application as the text displayed is to be flipped not the x0,y0 location.

The Display Component does not include any horizontal flip / mirror function at this point of time. I would like to avoid having to use a mirror to flip the image so that it gets projected appropriately

**Please describe your use case for this integration and alternatives you've tried:**
I am working on building a small LCD projector to project time and key environment parameters (e.g. temperature, humidity) on a bedroom ceiling at night.

To do so I am using a 0.42' 64x32 SSD1306 LCD display (https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20211104110336&SearchText=oled+ssd1306+0.42) and a lens with adjustable focus recovered from a broken projector clock.

As for the case, I use a custom one that I designed and 3d printed.

**Additional context**
 Building an ESP8266-based smart projection clock.

Mikei386, 2021-11-05:
I'm trying to accomplish the exact same thing right now. SSD1306 as a ceiling projector through a lens. But I can't flip the Image...

(What kind of lens do you use and how is your brightness?)

ychieux, 2021-11-06:
Brightness is not amazing on my end either... I'm re-using a lens assembly from a broken projector clock like this [one](https://ae01.alicdn.com/kf/HTB1sys6MXXXXXX.aXXXq6xXFXXXX/Desktop-Clock-Digital-Alarm-Clock-with-Projector-Color-Screen-Time-Projection-Clock-Multi-function-Weather-Calendar.jpg_Q90.jpg_.webp)

When our bedroom clock projector broke few weeks ago, my wife was looking to purchase a replacement unit... So I searched online for DIY builds and found the following [article on hackaday](https://hackaday.com/2018/05/05/simple-home-built-projection-clock-projects-time/). The small LCD used on that build is not supported by ESPhome but more importantly near impossible to source. I found one website in Poland listing it and few on AliBaba but you need to go for bulk orders of 1000 pieces...

As the OLED display aren't that bright, the best would be to use the following assembly: strong LED > condensor > negative transmissive LCD (with no backlight) > lens

Mikei386, 2021-11-08:
I’m still trying deferent methods.

I printed this one: [https://www.thingiverse.com/thing:4834890](url) and used it with some old google cardboard lenses. It worked okey. 

Now I’m testing an old lens from an Optoma DLP Beamer. It’s way better. Pretty sharp projection. The brightness ist okey in a very dark room but could be a little bit brighter… just a tiny bit…

I use it to project some sensor states from my HomeAssistant.

But everything won’t matter if there is no way to mirror the image in ESPHome.
I also thought about a different color OLED screen to be a bit brighter…

ychieux, 2021-11-08:
I managed to flip the display as required for projection. There was a the typo in `__init__.py` that prevented the flip functions to work as intended. I created a PR for it (you can refer to the linked PR above)

The following configuration with flip_x to true and flip_y to false work for me:
```
display:
  - platform: ssd1306_i2c
    model: "SSD1306 64x32"
    reset_pin: D0
    flip_x: true
    flip_y: false
    address: 0x3C
    id: mydisplay
    lambda: |-
      // Print time in HH:MM format
      it.strftime(0, 0, id(font_Arial), "%H:%M", id(esptime).now());
```

closed 2021-11-09 by ychieux