esphome/feature-requests #1499: upgrade to a minimal firmware

Bram1nat0r, 2021-11-07:
**Describe the problem you have/What new integration you would like**

After the introduction of the Noise encryption protocol my binaries have grown slightly bigger. I now have to first create a minimal configuration of the device and upload it to the device. 
After this I am able to upload the full build. 
It would be nice to have a button : Upgrade to minimal, where it only takes the necessary info of a full configuration to upload a minimal build. 
After successfull upload of the minimal build, you can use the normal upload button to upload the final build. 

**Please describe your use case for this integration and alternatives you've tried:**

The alternative way of going about this, is to copy your full configuration to a notepad, manually make a minimal configuration and upload this. Then paste your full configuration and upload the final build. 
It’s a bit of a hassle. 

**Additional context**

dbuezas, 2021-11-19:
There's a slightly better workaround:
* Create an extra node with the minimal configuration 
* use the `use_address` [wifi component](https://esphome.io/components/wifi.html) parameter to direct it to whatever node you need.

Not perfect, but at least for me, it is a bit more convenient :)

dbuezas, 2021-11-20:
this is what I do:
```yaml
substitutions:
  target: name-of-the-node-to-flash

esphome:
  name: $target
  platform: ESP8266
  board: esp8285
  build_path: ./build/minimal

wifi:
  ssid: !secret wifi_name
  password: !secret wifi_pass
  use_address: $target
  fast_connect: true

ota:

```

nagyrobi, 2022-07-08:
Duplicate of https://github.com/esphome/feature-requests/issues/1264
Issue fixed via https://github.com/esphome/esphome/pull/2601
Also see https://github.com/esphome/feature-requests/issues/819