esphome/feature-requests #1503: Implement Homeassistant Entity Categories

meridani, 2021-11-08:
**Describe the problem you have/What new integration you would like**

Homeassistant has the new feature called entity categorisation and I would like to have an option for the output and sensor components to have an option for setting the category manually.

**Please describe your use case for this integration and alternatives you've tried:**

Use the new entity category to its full potential to organize a device's entities .
For example: status, uptime etc. are sensor entities right now but they would be better to be diagnostic entities.

**Additional context**

Bram1nat0r, 2021-11-08:
https://www.home-assistant.io/blog/2021/11/03/release-202111/#entity-categorization

"Oh, and soon, we can add ESPHome to that list!"

I guess you should be a little bit more patient ;-)

oxan, 2021-11-10:
Will be in 2021.11 ;)

closed 2021-11-19 by oxan