esphome/feature-requests #1504: Assistance creating a new component for the TM1638 LED & KEY board for contribution to ESPHome project

skykingjwc, 2021-11-08:
**Describe the problem you have/What new integration you would like**

I am creating a component for the TM1638 based LED&KEY board.  When completed I would like to contribute via a pull request.  

**Please describe your use case for this integration and alternatives you've tried:**

I have most of the code written and I am able to display text on the 7segment display as well as turn the top row of LEDs on and off via the SPI interface.  The issue arises when I attempt to read the condition of the buttons.  I can get it to work on an Arduino, but it does not work when building ESPHome and running the code on an ESP32.   I have verified using an oscilloscope that the correct clock and strobe signals are being sent and that the correct register is being read.  However the chip does not appear to be sending data back.  The Arduino code and the ESPHome code are nearly identical, so I am not sure what is wrong.
I am hoping I could work with another developer to complete the project and contribute to the ESPHome project.  I just need assistance from someone more familiar with the chip and who perhaps knows more about SPI than I do.

**Additional context**

I will be able to provide code and screenshots from an oscilloscope as needed to assist in troubleshooting.

taintedkernel, 2021-11-19:
Funny on the timing, I just started looking into trying out ESPHome with one of these devices as well.  I have a little experience with SPI and new to the chip as well, but more then happy to try and help out.

atrent, 2021-12-23:
Can this merge request be of help?
https://github.com/danja/TM1638lite/pull/6/files

skykingjwc, 2022-01-11:
> Funny on the timing, I just started looking into trying out ESPHome with one of these devices as well. I have a little experience with SPI and new to the chip as well, but more then happy to try and help out.

I have a working component, with the help of Ssieb from the Discord chat.   I've had it running for a few weeks with no issues.  I'm working on documentation for it before creating the PR.  If you want to help with that it would be great, because I'm having trouble finding time to do it.    Feel free to PM me on Discord and I can show you everything.

randybb, 2022-01-11:
Probably it would make sense to merge the existing tm1651 and tm1637 to an universal tm16xx component as there are many other versions https://awesomeopensource.com/project/maxint-rd/TM16xx

taintedkernel, 2022-01-23:
I ended up writing something from scratch and leveraging some existing Arduino libraries for the device over the holidays.  I'd like to help, but lately I'm in the same boat - my free time is no longer available so unfortunately I can't be of much assistance.

nagyrobi, 2022-07-01:
Please join our Discord server to ask for development support.

closed 2022-07-01 by nagyrobi