esphome/feature-requests #1509: Add support for sdp8xx sensors

alzyee, 2021-11-14:
**Describe the problem you have/What new integration you would like**

There is currently a sdp3x sensor ( https://esphome.io/components/sensor/sdp3x.html ). As it is written this sensor component doesn't check or recognize sdp8xx sensor. If additional checks were implemented in sdp3x.cpp (https://github.com/esphome/esphome/blob/108b8e6705a6f5080fa2cdc408437d2b4d3c525a/esphome/components/sdp3x) to check for the sdp8xx variants and set the correct pressure_scale_factor_ This 8 byte product identifier could also be used to set the i2c address if not set in user yaml config.
It looks like the commands and interfaces are the same for the other implemented i2c commands.

 

**Please describe your use case for this integration and alternatives you've tried:**

I copied all the code for sdp3x into a custom sensor and added an else that just sets the scaling factor for the sensor I am using.

**Additional context**

The code change here seams easy enough either add a branch to the spd3x or add a new spd8xx sensor which uses the spd3x as a starting point. If wanted to contribute this change should it be created as a separate sensor or handled under spd3x?

martgras, 2021-11-16:
I attempted to implement support for the SDP8xx familiy.
Since I don't have the device I can't test - so it may fail. 

To test add sdp3x as an external component 
```
external_components:
  - source: github://martgras/esphome@sdp8xx
    components: ["sdp3x"]
```

I also added a new option `measurement_mode` - it defaults to `differential_pressure` but can be set to `mass_flow' as well. 
Not that I understand the difference between both modes but the datasheet listed it. 
Another change is using the scale_factor as provided by the sensor itself in the measurement result instead of hardcoding it. 
Please enable verbose logging when testing

alzyee, 2021-11-19:
Is there an easy way to modify the sdp3x.cpp file in home assistant esphome addon? Or if I just put it the folder as a custom sensor does it just override the build in sensor of the same name?

If so I can validate that it works for my sensor.

martgras, 2021-11-20:
You don't have to modify the file. 
Just add this to your yaml file 
The updated files from my github repository will then be used for sdp3x 
```
external_components:
  - source: github://martgras/esphome@sdp8xx
    components: ["sdp3x"]

# Example configuration entry
- platform: sdp3x
  name: "HVAC Filter Pressure drop"
  id: filter_pressure
```

alzyee, 2021-11-21:
I was able to use the sensor as you had suggested but the result appears to be off by a factor of around 100. I get ~75hPa from the 125Pa sensor. Where as with my code resulted in ~0.75hPa. At first glance I'm not sure what is incorrect.

alzyee, 2021-11-21:
I believe the issue is the scaling factor in the reading returns Pa. I left some comments on the commit in your fork.

alzyee, 2021-11-21:
Your updates fixed the scaling. So it works as expected for me. Thanks for getting it working.

I'd think the verbose logging scaling factor + temp value printed out should also be called raw plus a comma but it works as is.

current
> Got raw pressure=18237, scale factor =240 temperature=4132
suggested change
> Got raw pressure=18237, **raw** scale factor =240 **, raw** temperature=4132
suggested change option 2
> Got raw pressure=18237, scale factor =24000 **, raw** temperature=4132

martgras, 2021-11-21:
Thanks for the suggestion - created https://github.com/esphome/esphome/pull/2779

closed 2021-11-24 by jesserockz