esphome/feature-requests #1511: Adafruit ESP32S2 feather with BME280

ScottMacWatters, 2021-11-17:
**Describe the problem you have/What new integration you would like**

Adafruit recently released an ESP32S2 board with a built in BME280. It would be awesome to get this working, as it would provide an out of the box way to get a temperature sensor deployed without jumpers or soldering. 

**Please describe your use case for this integration and alternatives you've tried:**

From looking at other issues and documentation, esp32s2 support hasn’t been added yet. It looks to be in progress, so I’m hoping to add this other board to the list. The use case here is to just support more boards. 

**Additional context**

Here’s the link to the board:
https://www.adafruit.com/product/5303

arunderwood, 2022-01-01:
I was able to get my [Adafruit ESP32-S2 Feather - 2 MB PSRAM and STEMMA QT / Qwiic](https://www.adafruit.com/product/5000) (no BME280) working out of the box with the following config:

```
esphome:
  name: test

esp32:
  board: featheresp32-s2
  framework:
    type: esp-idf
    version: latest
```

I also put up https://github.com/esphome/esphome/pull/2970 which makes using i2c on the ESP32-S2 marginally easier.

Maybe this same config would work on the version with the BME280?

Here's a blind attempt at writing a config that may work for the Adafruit ESP32-S2 Feather with BME280

```yaml
esphome:
  name: test

esp32:
  board: featheresp32-s2
  framework:
    type: esp-idf
    version: latest

i2c:
  # Setting pins for sda and scl will be required until https://github.com/esphome/esphome/pull/2970 is released
  sda: GPIO3
  scl: GPIO4

# This switch pulls pin 7 low to enable power to the BME.  I'm not fully sure this is required, I'm thinking I2C_POWER may be on by default in my Feather https://learn.adafruit.com/adafruit-esp32-s2-feather/pinouts
switch:
  - platform: gpio
    id: I2C_POWER
    restore_mode: RESTORE_DEFAULT_ON 
    pin:
      number: 7
      inverted: true

sensor:
  - platform: bme280
    temperature:
      name: "Temperature"
      id: bme280_temperature
      oversampling: 16x
    pressure:
      name: "Pressure"
      id: bme280_pressure
      oversampling: 16x
    humidity:
      name: "Humidity"
      id: bme280_humidity
      oversampling: 16x
    address: 0x76 # Address is not default!
```

ScottMacWatters, 2022-01-02:
Hey @arunderwood, thanks for taking a look at this.

I tried tonight and was unable to successfully get it working. 

Here's the yaml I tried, it's basically the one you provided, but let me know if there's an issue.

```
esphome:
  name: feather-esp32-s2-bme280

esp32:
  board: featheresp32-s2
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:

i2c:
  # Setting pins for sda and scl will be required until https://github.com/esphome/esphome/pull/2970 is released
  sda: GPIO3
  scl: GPIO4
  

# This switch pulls pin 7 low to enable power to the BME.  I'm not fully sure this is required, I'm thinking I2C_POWER may be on by default in my Feather https://learn.adafruit.com/adafruit-esp32-s2-feather/pinouts
switch:
  - platform: gpio
    id: I2C_POWER
    restore_mode: RESTORE_DEFAULT_ON 
    pin:
      number: 7
      inverted: true

sensor:
  - platform: bme280
    temperature:
      name: "Temperature"
      id: bme280_temperature
      oversampling: 16x
    pressure:
      name: "Pressure"
      id: bme280_pressure
      oversampling: 16x
    humidity:
      name: "Humidity"
      id: bme280_humidity
      oversampling: 16x
    address: 0x76 # Address is not default!

ota:
  password: # redacted

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Feather-Esp32-S2-Bme280"
    password: # redacted
```

I run ESPHome in a docker container on a pi, and haven't mapped out the USB. Typically, I just do the first flash using EspHome Flasher then when the device connects to my network, I map the IP address and do future updates wirelessly. 

I was able to get this YAML to build, and I downloaded the .bin file. From there, I tried a few different ways to install it. All ended with the same result of nothing ever showing up on my network. I'm also not seeing the captive portal, so I think there must be some issue with the .bin file that's been generated. 

* https://adafruit.github.io/Adafruit_WebSerial_ESPTool/ -- Set the baud rate correctly and flashed it to offset 0, after erasing the device
* esptool.py `esptool.py --port /dev/tty.usbmodem01 write_flash 0x0 ~/Downloads/feather-esp32-s2-bme280.bin`
* espHome Flasher Selecting the firmware and serial port. 

My version of ESPHome is 2021.12.3. 

I'm not sure if this is an issue with my setup, my board, the way I've been doing things or what. If there's any recommendations, that would be great. Or if anyone else can give it a try, they're welcome to do so. 

I'll keep trying in the coming days, but that's where I'm leaving off for the night.

arunderwood, 2022-01-02:
Well at least no config or compilation errors is a nice start.  I would try using the logs feature of the ESPHome Flasher to stream the device logs over the USB interface.  This may reveal a crash on startup or some other issue that's preventing it from connecting to wifi.  I've also had issues in the past where the device was drawing more power than could be supplied, there's usually a brownout log in this case.  Either way, getting logs from the device is a good next step.

ashald, 2022-03-13:
@arunderwood how did you flash the board?
I've got the same one as you, and upon arrival it would blink LEDs when powered on. I tried flashing it with the same config as you showed above, but it was not showing up under `/dev`. I used https://adafruit.github.io/Adafruit_WebSerial_ESPTool/ to erase it, and it still didn't show up. Until I hold Boot when I plug it in. And when I do so, I can flash it, but after I unplug it and plug it back, it blinks orange light near connector few times and nothing happens. It never shows up under `/dev` and I cannot see logs or whatever. I probably did something wrong, but I wonder what exactly...

nagyrobi, 2022-07-01:
https://github.com/esphome/feature-requests/issues/996
https://github.com/esphome/esphome/pull/3264

closed 2022-07-01 by nagyrobi

btoconnor, 2022-10-13:
@ashald did you ever resolve your issue? I'm seeing something very similar on my ESP32S2 with BME280.

cTurtle98, 2022-12-27:
> @ashald did you ever resolve your issue? I'm seeing something very similar on my ESP32S2 with BME280.

I was having the same issue but after changing from esp-idf to arduino for the framework I dont have anymore issues

btoconnor, 2022-12-29:
I ended up figuring this out, I think. It's been a few weeks since I looked at this, but if I recall correctly, the issue was related to the boot priority and powering the BME280.  I'm reaching into memory, but IIRC it was something that ESPHome was attempting to bring up the I2C bus before turning the GPIO pin that turns the I2C bus on the board.  Changing the boot order on the i2c bus and the GPIO pin seemed to work.  This I believe is the working configuration I had going before switching projects:

```yaml
esphome:
  name: adafruit

  platformio_options:
    board_build.variant: um_feathers2

  on_boot:
    priority: 801
    then:
      - switch.turn_on: I2C_POWER

esp32:
  board: featheresp32-s2
  variant: esp32s2
  framework:
    type: esp-idf
    version: recommended

logger:
  level: DEBUG

ota:

api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain
  power_save_mode: HIGH

i2c:
  # # Setting pins for sda and scl will be required until https://github.com/esphome/esphome/pull/2970 is released
  sda: GPIO3
  scl: GPIO4
  # scan: True
  id: bus_a
  setup_priority: 700

sensor:
  - platform: uptime
    update_interval: 10s
    id: up
  - platform: bme280
    i2c_id: bus_a
    temperature:
      name: "Temperature"
      id: bme280_temperature
      oversampling: 16x
    pressure:
      name: "Pressure"
      id: bme280_pressure
      oversampling: 16x
    humidity:
      name: "Humidity"
      id: bme280_humidity
      oversampling: 16x
    address: 0x77 # Address is not default!
```

There's probably a better way to do this but I was able to at least run ESPHome on the feather board with this.