esphome/feature-requests #1513: Add Climate on web interface

nagyrobi, 2021-11-17:
**Describe the problem you have/What new integration you would like**
Currently the embedded web server doesn't display the climate configured on the node. Would be nince to at least have:
- current sensed temperature (if available)
- current preset
- current mode (operating or idle)
- current target temperature
- +/- buttons to modify the target temperature
- a preset selector dropdown with available modes (some of `off`, `heat_cool`, `cool`, `heat`, `dry`, `fan_only`, `auto` depeding on what's configured)

**Please describe your use case for this integration and alternatives you've tried:**
Several climates controllers around the house, none has controls on the web interface.

**Additional context**

N/A

wilberforce, 2021-11-17:
Not yet - I plan to add this as I want it too!

nagyrobi, 2022-07-07:
Webserver v2 implements it: https://esphome.io/components/web_server.html