esphome/feature-requests #1514: support for the wt32-sc01 dev board

whytewolf, 2021-11-17:
**Describe the problem you have/What new integration you would like**

I would love to be able to use a WT32-SC01_v3.2 board+screen from ESPHome

**Please describe your use case for this integration and alternatives you've tried:**

currently the display is not supported, making this device almost worthless as is. 
specs for this device can be found at http://www.wireless-tag.com/wp-content/uploads/2020/12/WT32-SC01DataSheetV3.3-2-with-nuts.pdf

**Additional context**

need access to the tp and display unit this device comes with. esp32 already fully supported.

geiseri, 2021-11-20:
You might be better off with https://github.com/HASwitchPlate/openHASP.  esphome is better for sensor data collection and controlling physical elements. Doing anything nontrivial with a display is quite painful and any sort of touch screen control asymptotically approaches impossible.  So if your focus is on "headless" sensors esphome is by far superior to anything else out there.  If you are looking for a terminal without any sort of sensor input openHASP is a much better fit.

cpyarger, 2022-01-22:
this is a rather common request.... as you can see from 

#1084 

Currently I am using my wt32_sc01 for handling my Xiaomi-Mi-Scale readings and sending them to home assistant.  My Ideal is to install the display at eye level in front of my scale and have it show the readings from my scale for ~60 seconds after a new reading comes in, or after a tap or two on the screen. 

personal opinion here, but @geiseri  just because something may be painful, doesn't mean that it shouldn't be done.

geiseri, 2022-01-22:
@cpyarger I am just explaining that depending on what you want to do with the device you might want to use one solution or the other.  Since the wt32 has a nice large touchscreen and tons of psram you can do a very slick responsive data input/display screen.  Animations, feedback, swipe, etc...  In that solution, you just edit a JSON file.   Now if you want to hook up anything beyond a GPIO or an analog pin openHASP is a weekend with platformio and vscode =).  So if you primarily want a sensor that has a simple mostly data display ESPHome is probably 50 lines of properly formatted YAML, 10 lines of c++,  and you can get on with your life.  If you want a few debounced touchscreen buttons with feedback, or god help you a slider...  well, I told you so ;)

As an example this UI:
[image]
was ~80 lines of JSON, and I have a controller for my LED light strip.  Now the PIR sensor I was using to "wake up" the display would have been 3-4 lines of YAML in ESPHome.  With openHASP, it was a *bit* more effort.  So much effort, in fact, it was easier to hook the PIR  sensor to the M5 Atom (running ESPHome) I was using to control the actual lights and have it notify the display to wake up!

cpyarger, 2022-07-01:
Specifically, When I try to do anything with the display, I find myself unable to compile display code without it crashing due to out of memory errors, the defaults for the esp32 do not account for having the  ESP32-WROVER-B Chip and 4/16MB Flash & 8MB PSRAM.
the requisite yaml could be
```yaml
esp32:
  board: wt32sc01
  ```

nagyrobi, 2022-07-01:
https://github.com/esphome/feature-requests/issues/1721

ianmxyz, 2022-07-04:
> You might be better off with https://github.com/HASwitchPlate/openHASP. esphome is better for sensor data collection and controlling physical elements. Doing anything nontrivial with a display is [...]

Haspmota being a Tasmota/openHASP hybrid will cover both, needs a little more effort but works nicely enough. Starting point: https://tasmota.github.io/docs/LVGL/ upload the correct display.ini then follow the links.

DellanX, 2022-09-03:
It looks like the comments above are focusing on the Display portion of the WT32-SC01, and suggesting alternatives to ESPHome. I'd like to refresh that some would still need support for the display/TFT/Wrover-B in ESPHome (I especially, for my more complex setup)

For my use case, I am wanting to combine a bunch of sensors and a display onto a single ESP32. To which, the WT32-SC01 was the perfect candidate. 
The Display needs I have are few, but the IO/I2C needs I have are more complex, to which I've only seen ESPHome be able to meet.
- DHT sensor
- Buzzer/Speaker
- RFID Reader
- Battery Monitor Sensor
- Capacitive Touch Sensor
- Some IO Switches
- Custom I2C Button board (Which is why I don't need a strong UI)

My display needs are just top level status info, and maybe a button or two.
This system currently runs on a battery, so I'd prefer to have my entire system on a single ESP32 if possible.
(As mentioned earlier in the thread #1721 is a request to add OpenHASP into ESPHome, and could solve the Display portion in ESPHome, if implemented)

geiseri, 2022-09-08:
Yeah, OpenHASP is not ideal for any sensors beyond a GPIO switch.  I do agree with you though.  It would be better to be able to ingest the OpenHASP configuration JSON and use that to display the ESPHome sensors.  I have to admit as far as code bases go ESPHome has the most sane design, and it really makes other projects in this ecosystem feel quite kludgy.  My biggest issue is that the ESPHome API does everything via a single drawpixel in 24bit color.  Every time I have tried to integrate with LVGL it will render static okay, but any sort of swipe or animation falls apart.  That being said @fvanroie has done some pretty impressive stuff, and seems to have an in with the LVGL guys.  So maybe he can make that magic happen.

rarroyo6, 2022-12-23:
I vote for this also. I know there's other software that can be used, but the idea is to incorporate it into ESP-Home.

poundy, 2023-03-25:
This is now in 2023.3 folks.....

jd1, 2024-04-04:
Has anyone a working config for the display and can share it here?

nagyrobi, 2024-04-04:
Work in progress, LVGL will be natively supported by ESPHome: https://github.com/esphome/esphome/pull/6363

[image]

nagyrobi, 2024-08-30:
Done: https://esphome.io/components/lvgl/

closed 2024-08-30 by nagyrobi