esphome/feature-requests #1515: ESPHome remote configs repository

mehuman, 2021-11-18:
**Describe the problem you have/What new integration you would like**

A repository of device configs for off the shelf devices that can be imported via the packages remote function in the config
https://esphome.io/guides/configuration-types.html?highlight=remote#remote-git-packages

The configs would be just the base requirements for the device, pin outs, chip definitions, etc.

**Please describe your use case for this integration and alternatives you've tried:**

Alternatives like https://www.esphome-devices.com exist but ask the user to copy and paste the config instead of just being able to import it.

**Additional context**

Had a discussion with @mmakaay about some of the intricacies about releasing and breaking changes. If configs were tagged in the same fashion as releases it would just require a user to update the config or use a variable for the ESPHome version.

Another potential issue would be breaking changes but since breaking changes are already tracked for the release notes they could be used to scan the configs for potential issues.

This would also help to alleviate new user onboarding and device discovery since a user wouldn't need to search for already existing configs and research the pinouts of a device.

mehuman, 2021-11-18:
Decision is to submit changes to esphome-devices.com with links to remote configs.

sourbrambles, 2021-11-23:
Not directly related but the official ESPhome [Cookbook](https://esphome.io/index.html#cookbook ) feels neglected vs 
https://www.esphome-devices.com it'd be great if these were a little more interlinked or merged.

New users would be immediately opened up to a lot more options for converting existing products they might have. 

There was a smart plug I wanted to convert and as a lazy person I never bothered to create my own config but when I found https://www.esphome-devices.com there was the smart plug. 

Again not really related to your suggestion but it's the first time I've personally seen https://www.esphome-devices.com/ mentioned in the feature requests etc.