esphome/feature-requests #1516: Green status bar and text online when device is online in ESPhome dashboard

dracula385, 2021-11-18:
**Describe the problem you have/What new integration you would like**
The current ESPHome dashboard uses colors to indicate if a device is offline/update available. But there is no color or text online when device is online.

**Please describe your use case for this integration and alternatives you've tried:**
I would love to see green status bar and text online when device is online... This was concept is the same for offline/update available

**Additional context**

tomlut, 2022-03-02:
As I don't update my multitude of ESPHome devices unless there is an issue fix or a new feature I want, I would rather see display priority given to the Online/Offline status than the availability of an update.

CFenner, 2023-01-02:
In my eyes it would already be sufficient to have the `online` text colored green.

ssieb, 2023-01-02:
That was specifically removed, so it's not likely to be coming back.

ezpl, 2023-01-02:
When having long list of devices, it is difficult to spot the ones which are online. They should definitely have a colour!

randybb, 2023-01-02:
Why? Normal status is online, there is no need to let user know that devices is online. If something is wrong, like there is a possible update or device is offline, then a color bar will appear with UPDATE/OFFLINE status. The same apply for HA - normally entities are without any status, only disabled/unavailable/restored entities have their status.

Ankita12vanshette, 2024-01-22:
i want green line when device is online in esphome