esphome/feature-requests #1519: Enable Tuya air conditioner mode change using TuyaMCU and climate components

rickydg, 2021-11-19:
**Describe the problem you have/What new integration you would like**

Change air conditioner mode
**Please describe your use case for this integration and alternatives you've tried:**

My Tuya air conditioner has 4 modes; Auto, Cool, Fan and Dry. I can see no way of controlling the modes through the current component abilities. I would like to be able to set a mode datapoint to control these modes. For my air conditioner modes it is datapoint 4 (enum).
**Additional context**

Currently, I am only able to control the fan using the Tuya fan component and the Cool mode using the Tuya Climate component. I seem to be able to use all other functions through ESPHome.

user897943, 2022-06-09:
If this has been open since November 2021 with no action I wonder if there's hope of ever happening. I have the same issue now. Has different modes, can't use them... I would tackle this if I had any experience with programming but my knowledge is zero. It's not like I'm lazy either, I opened a feature request and ended up getting the AC mostly working pretty much on my own after being given some clues by another helpful guy: https://github.com/esphome/feature-requests/issues/1734

blacktirion, 2023-07-31:
@rickydg There may be this functionality coming soon, based on the PR I created.