esphome/feature-requests #1520: Add localisation support for display component

jsuanet, 2021-11-20:
**Describe the problem you have/What new integration you would like**
When using the display component, the print functions always uses a dot as decimal seperator. It would be nice if it is possible to use a localisation variable to set a different separator. 
It would also be nice if support is added for a thousand seperator, so you have the same representation in HA and ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**
I would like to display values in the same format in HA and on an ESPHome display.

**Additional context**

mmakaay, 2021-11-20:
Some random first thoughts on this:

Next to locales, i18n would be a good complementary addition too. At minimum providing 1-to-1 translation of strings to the selected languages.
This would open the road to GitHub configuration packages that can be used to easily support firmwares for multiple languages.

A generalized i18n solution with locales would be beneficial for other components too, like text sensors and telegram bot.

Multi-language support would probably have to live in the code generation world, to not have to compile in all unused language strings.

Locale support would live in c++ firmware land, sinvlce that needs runtime support.

I'd say that all this would be captured best in code as two separate ESPHome components: locale and i18n.

jackdrake, 2023-06-30:
Would be highly appreciated as I eg use a weather forecast presented in esphome display. Would be nice to have german weather (ok, rain keeps ugly, even if it is "Regen"....)

olicooper, 2023-09-12:
I too would be interested in this. I need to display the date (including day-of-week), time and weather forecast.

There may be some functionality built in to C++ std library already which might help?
See [set_locale](https://en.cppreference.com/w/cpp/locale/setlocale)
e.g. `std::setlocale(LC_ALL, "de_DE.UTF-8")`

carlos22, 2024-12-18:
Any updates on this? In my case I want to have another format of floats.