esphome/feature-requests #1522: fingerprint_grow multiple instances

Atomlifetce, 2021-11-23:
**Describe the problem you have/What new integration you would like**

The component "fingerprint_grow" does not support more instances? 

**Please describe your use case for this integration and alternatives you've tried:**

I want to make two of fingerprint_grow sensors on ESP32.
Home assistant version 2021.10.5, ESP Home version 2021.11.1
The editor says by the rows with the triggers "on_finger_scan_matched"... this: Too many candidates found for 'id' type 'fingerprint_grow::FingerprintGrowComponent' Some are 'fingerprint_in' , 'fingerprint_out'.
My YAML code:
```
uart:
  - id: uart_out
    tx_pin: GPIO17
    rx_pin: GPIO16
    baud_rate: 57600
  - id: uart_in
    tx_pin: GPIO23
    rx_pin: GPIO22
    baud_rate: 57600

fingerprint_grow:
  - id: fingerprint_in
    uart_id: uart_in
    sensing_pin: GPIO32
    on_finger_scan_matched:
      - switch.turn_on: relay_lock
      - fingerprint_grow.aura_led_control:
          state: BREATHING
          speed: 200
          color: BLUE
          count: 1
      - text_sensor.template.publish:
          id: fingerprint_out_state
          state: "Authorized finger"
    on_finger_scan_unmatched:
      - fingerprint_grow.aura_led_control:
          state: FLASHING
          speed: 10
          color: RED
          count: 10
      - text_sensor.template.publish:
          id: fingerprint_out_state
          state: "Unauthorized finger"
    on_enrollment_scan:
      - fingerprint_grow.aura_led_control:
          state: FLASHING
          speed: 25
          color: BLUE
          count: 2
      - fingerprint_grow.aura_led_control:
          state: ALWAYS_ON
          speed: 0
          color: PURPLE
          count: 0
      - text_sensor.template.publish:
          id: fingerprint_out_state
          state: "Finger scanned"
    on_enrollment_done:
      - fingerprint_grow.aura_led_control:
          state: BREATHING
          speed: 20
          color: BLUE
          count: 3
      - text_sensor.template.publish:
          id: fingerprint_out_state
          state: "Enrolled fingerprint"
    on_enrollment_failed:
      - fingerprint_grow.aura_led_control:
          state: FLASHING
          speed: 25
          color: RED
          count: 4
      - text_sensor.template.publish:
          id: fingerprint_out_state
          state: "Failed to enroll fingerprint"
    
  - id: fingerprint_out
    uart_id: uart_out
    sensing_pin: GPIO27
    on_finger_scan_matched:
      - switch.turn_on: relay_lock
      - fingerprint_grow.aura_led_control:
          state: BREATHING
          speed: 200
          color: BLUE
          count: 1
      - text_sensor.template.publish:
          id: fingerprint_out_state
          state: "Authorized finger"
    on_finger_scan_unmatched:
      - fingerprint_grow.aura_led_control:
          state: FLASHING
          speed: 10
          color: RED
          count: 10
      - text_sensor.template.publish:
          id: fingerprint_out_state
          state: "Unauthorized finger"
    on_enrollment_scan:
      - fingerprint_grow.aura_led_control:
          state: FLASHING
          speed: 25
          color: BLUE
          count: 2
      - fingerprint_grow.aura_led_control:
          state: ALWAYS_ON
          speed: 0
          color: PURPLE
          count: 0
      - text_sensor.template.publish:
          id: fingerprint_out_state
          state: "Finger scanned"
    on_enrollment_done:
      - fingerprint_grow.aura_led_control:
          state: BREATHING
          speed: 20
          color: BLUE
          count: 3
      - text_sensor.template.publish:
          id: fingerprint_out_state
          state: "Enrolled fingerprint"
    on_enrollment_failed:
      - fingerprint_grow.aura_led_control:
          state: FLASHING
          speed: 25
          color: RED
          count: 4
      - text_sensor.template.publish:
          id: fingerprint_out_state
          state: "Failed to enroll fingerprint"

switch:
  - platform: gpio
    id: relay_lock
    pin: GPIO23
    name: "Lock"
  
sensor:
  - platform: fingerprint_grow
    fingerprint_grow_id: fingerprint_out
    fingerprint_count:
      name: "Fingerprint Count"
    last_finger_id:
      name: "Fingerprint Last Finger ID"
    last_confidence:
      name: "Fingerprint Last Confidence"
    status:
      name: "Fingerprint Status"
    capacity:
      name: "Fingerprint Capacity"
    security_level:
      name: "Fingerprint Security Level"

text_sensor:
  - platform: template
    id: fingerprint_out_state
    name: "Fingerprint OUT State"
  - platform: template
    id: fingerprint_in_state
    name: "Fingerprint IN State"
    
      
binary_sensor:
  - platform: fingerprint_grow
    fingerprint_grow_id: fingerprint_out
    id: fingerprint_enrolling
    name: "Fingerprint Enrolling"

api:
  services:
  - service: enroll_out
    variables:
      finger_id: int
      num_scans: int
    then:
      - fingerprint_grow.enroll:
          finger_id: !lambda 'return finger_id;'
          num_scans: !lambda 'return num_scans;'
  - service: cancel_enroll_out
    then:
      - fingerprint_grow.cancel_enroll:
  - service: delete_out
    variables:
      finger_id: int
    then:
      - fingerprint_grow.delete:
          finger_id: !lambda 'return finger_id;'
  - service: delete_all_out
    then:
      - fingerprint_grow.delete_all
```

... I'm new in YAML. The guys on the community.home-assistant.io says I should open this request.
I thank you for the update or help.

**Additional context**

Atomlifetce, 2021-12-20:
After some tries and help from "https://community.home-assistant.io/"
Here is the working code.
`
fingerprint_grow:
  - id: fingerprint_in
    uart_id: uart_in
    sensing_pin: GPIO32
    on_finger_scan_matched:
      - switch.turn_on: relay_unlock
      - fingerprint_grow.aura_led_control:
          id: fingerprint_in
          state: BREATHING
          speed: 200
          color: BLUE
          count: 1
      - text_sensor.template.publish:
          id: fingerprint_out_state
          state: "Authorized finger"
      - switch.turn_on: relay_unlock
      - delay: 500ms
      - switch.turn_off: relay_unlock
    
    
  - id: fingerprint_out
    uart_id: uart_out
    sensing_pin: GPIO12
    on_finger_scan_matched:
      - switch.turn_on: relay_unlock
      - fingerprint_grow.aura_led_control:
          id: fingerprint_out
          state: BREATHING
          speed: 200
          color: BLUE
          count: 1
      - text_sensor.template.publish:
          id: fingerprint_out_state
          state: "Authorized finger"
      - delay: 500ms
      - switch.turn_off: relay_unlock`

Atomlifetce, 2022-01-15:
So, the new info.
The code can be compiled and loaded into ESP32, but after connecting two fingerprint sensors, the ESP want not to boot to run state. It looks like on the background there is really not a support to connect two fingerprint sensors. If the developers can update it, it will be great. Thank you ;)

Buddeli, 2025-01-23:
@Atomlifetce 
Did you manage to make the two sensor work? If yes, can you please share the whole code (text sensor, binary sensor, sensor and services part)? Many thanks in advance!

Atomlifetce, 2025-01-23:
Hey,
Unfortunately, I couldn't make it.
I'm currently using two ESP32 side by side for it.

Buddeli, 2025-01-23:
Thanks for the quick replay. I wished to avoid that but maybe I'll do the same.