esphome/feature-requests #1529: implement source code text sensor

stanislav-brabec, 2021-11-29:
**Describe the problem you have/What new integration you would like**

Users often want to use their devices across a different systems and accounts (e. g. Arduino IDE on desktop vs. Home Assistant ESPHome plugin). It opens a risk of inconsistent source codes or event a complete source code loss.

It would be nice to create a new sensor: source code sensor.

The source code integration would have a text sensor, that would provide a complete source code.

I can imagine another component that could provide set of text sensors that will contain versions of all modules used for the compilation.

**Please describe your use case for this integration and alternatives you've tried:**

Get a device, download source code from the device itself, upload an update, without need to obtain a source code together with the device.

It will also make possible updates of third party ESPHome devices without search for the source code.

Providing versions could make possible to create byte-by-byte identical clones of the firmware of the device.

**Additional context**

oxan, 2021-12-08:
The problem with this is that a text sensor state has a max length of 255 characters.

stanislav-brabec, 2021-12-09:
> The problem with this is that a text sensor state has a max length of 255 characters.

Then provide more sensors with 255 characters, if needed.

Or at least make possible to provide a source current URL and source latest URL.

oxan, 2021-12-13:
> Or at least make possible to provide a source current URL and source latest URL.

That's already possible with the [dashboard_import](https://esphome.io/guides/creators.html) function.

closed 2022-07-01 by nagyrobi