esphome/feature-requests #1530: Add support for internal pullup and debouncing for Rotary Encoders

feeling001, 2021-12-02:
**Describe the problem you have/What new integration you would like**

Rotary encoders can't be connected without pull up resistors without "unpredictable behavior". Pullup + debouncing is needed.
https://esphome.io/components/sensor/rotary_encoder.html

As most esp support internal pull up resistors, it would be nice:

- to be able to configure these internal pull up into the rotary encoder yaml declaration.
- to be able to set a debounce timer value into the rotary encoder yaml declaration.
 
This could result in something like :

`
pin_a: 
  pin : D1
  pullup : true
  debounce: 5ms
pin_b: 
  pin : D2
  pullup : true
  debounce: 5ms
`
This behavior allow to connect directly the encoder to the esp without dedicated hardware components (resistor, condensator).

**Please describe your use case for this integration and alternatives you've tried:**

My actual workaround is to use GPIO binary_sensor (which allow debounce/pullup), and implement by myself the rotary compute trough lambda's.

**Additional context**

Tried on ESP32 (nodemcu 32s)

martgras, 2021-12-02:
did you already try using the explicit pin definition . It doesn't allow debouncing but enables the pullup mode. 

```yaml
sensor:
  - platform: rotary_encoder
    name: "Rotary Encoder"
    pin_a:
      number: 18
      mode:
        input: true
        pullup: true
    pin_b: 
      number: 25
      mode:
        input: true
        pullup: true

    
binary_sensor:
  - platform: gpio
    name: "Rotary Button"
    pin: 
      number: 19
      mode:
        input: true
        pullup: true
    filters:
      - delayed_off: 50ms
      - delayed_on: 50ms
```

balk77, 2021-12-11:
I am successfully using debounce on a rotary encoder in combination with the internal pullups. Values are "collected" for 250ms before being sent out.
```
sensor:
  - platform: rotary_encoder
    name: "dimmer knob"
    id: my_rotary_encoder
    resolution: 4
    min_value: 0
    max_value: 255
    pin_a:
      number: GPIO26
      mode:
        input: true
        pullup: true
    pin_b:
      number: GPIO27
      mode:
        input: true
        pullup: true
    filters:
      debounce: 0.25s
```

DdP13, 2021-12-29:
my esphome says string value cannot be dictionary or list next to input of my rotary encoder how to fix this , trying to make google home control volume.

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO27
      mode: INPUT_PULLUP
      inverted: true
    name: decapo_1

  - platform: gpio
    pin:
      number: GPIO26
      mode: INPUT_PULLUP
      inverted: true
    name: decapo_2

  - platform: gpio
    pin:
      number: GPIO25
      mode: INPUT_PULLUP
      inverted: true
    name: decapo_3

  - platform: gpio
    pin:
      number: GPIO33
      mode: INPUT_PULLUP
      inverted: true
    name: decapo_4
    
  - platform: gpio
    pin:
      number: GPIO32
      mode: INPUT_PULLUP
      inverted: true
    name: decapo_5
    
sensor:
  - platform: rotary_encoder
    id: knob_Decapo
    pin_a:
      number: GPIO34
      mode:
        input: true
        pullup: true
    pin_b:
      number: GPIO35
      mode:
        input: true
        pullup: true
    min_value: -3
    max_value: 3
    on_value_range:
      - above: 1
        then:
          - script.execute: volume_up
      - below: -1
        then:
          - script.execute: volume_down

script:
  - id: volume_up
    then:
      - homeassistant.service:
          service: media_player.volume_up
          data:
            entity_id: media_player.googlehome8649
      - logger.log: "vol up trigger"
      - sensor.rotary_encoder.set_value:
          id: knob_Decapo
          value: 0
  - id: volume_down
    then:
      - homeassistant.service:
          service: media_player.volume_down
          data:
            entity_id: media_player.googlehome8649
      - logger.log: "vol down trigger"
      - sensor.rotary_encoder.set_value:
          id: knob_Decapo
          value: 0

closed 2022-07-01 by nagyrobi