esphome/feature-requests #1534: add support for DHT21/AM2301

johnnychicago, 2021-12-04:
**Describe the problem you have/What new integration you would like**

The DHT21 sensor is not currently supported in esphome. It is a slightly chunkier 'outdoor' variant of the well supported DHT22, the hardware being packaged in a more useful way for mounting a device. It quotes AM2301 on the case. Not adding a link since I don't want to spam for vendors.
Tasmota and similar support this.

**Please describe your use case for this integration and alternatives you've tried:**

I have tested a sensor, defined it in YAML as AUTO, as DHT22 and as all other options for model. Logfile clearly recognises there is a sensor, but can't speak to it.

**Additional context**

Here's hoping this thing is supported in underlying libraries and it would not be a major issue to activate support.

freezir12, 2021-12-21:
> **Describe the problem you have/What new integration you would like**
>
> The DHT21 sensor is not currently supported in esphome. It is a slightly chunkier 'outdoor' variant of the well supported [...]

In ESPhome i did not set the model. As i can read in the logs it is recognized as DHT11. Sometimes my sensor shows the correct values, sometimes just crap like -8 °C in my room^^ I am sure the sensor is kinda broken too. I will test that with a new one.
Did you try to set the model to DHT11?

freezir12, 2021-12-30:
Yes support for DHT21 would be awesome!

vilnis-jaunais, 2022-01-14:
Hoping on this as well!

theonteg, 2022-01-16:
DHT21 is a very good solution for outdoor usage. Implementing it in ESPhome would be more than welcome

freezir12, 2022-01-17:
Dont know why but my ESP8266 with connected DHT21 started working with 
sensor:
  - platform: dht
    model: DHT22

ed-davies, 2022-04-10:
Can confirm: an AM2301 seems to work fine with an ESP8266-based Sonoff TH16 and ESPHome 2022.2.5. Complete YAML stanza with faster-than-sensible updates just for my testing:

`sensor:
  - platform: dht
    pin: GPIO14
    model: DHT22
    temperature:
      name: temperature
      accuracy_decimals: 1
    humidity:
      name: humidity
      accuracy_decimals: 1
    update_interval: 10s
`

sumirati, 2022-04-25:
Created a PR to reflect the DHT21/AMS2301 as variants in the DHT documentation - https://github.com/esphome/esphome-docs/pull/2044

en-ver, 2022-05-20:
Can't get it working either.
Use it with nodeMCU
`sensor:
  - platform: dht
    pin: D7
    model: DHT22
    temperature:
      name: Roof Temp
      accuracy_decimals: 1
    humidity:
      name: Roof Humidity
      accuracy_decimals: 1
    update_interval: 10s`

```
[10:34:36][W][dht:163]: Waiting for DHT communication to clear failed!
[10:34:36][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[10:34:36][D][sensor:124]: 'Roof Temp': Sending state nan °C with 1 decimals of accuracy
[10:34:36][D][sensor:124]: 'Roof Humidity': Sending state nan % with 1 decimals of accuracy
```