esphome/feature-requests #1538: External / public / WAN IP sensor

eisengrau, 2021-12-07:
**Describe the problem you have/What new integration you would like**

External / public / WAN IP sensor. Either a request to a public server that returns the public IP  as a text sensor value of the gateway behind which the Esphome device is connected

**Please describe your use case for this integration and alternatives you've tried:**

HTTP get request to various JSON REST APIs is cumbersome, requires specific lambda configs to use. It should be an easier way.

**Additional context**

Might be useful for deployments in remote networks where no Home Assistant infrastructure is available, or on a remote network where such Esphome device us used (also) as a standalone network monitoring device.

111lisu, 2023-12-27:
I use ESPHome devices via IPSEC. I use double NAT for upgrade, but it is a very inconvenient solution. An IP WAN option would be useful to send the firmware to the OTA

J3bs, 2024-10-26:
I managed to get this working for myself earlier today so I'm making my way through my browser tabs I left open while searching for a solution. It's a bit late for OP I imagine, but perhaps this will help the next guy.  
PS: LOL @ this feature request 

`
http_request:

wifi:
<REMOVED>
  on_connect:
    - http_requeat.send:
        method: get
        url: "http://ipecho.net/plain
        capture_response: true
        on_response:
          then:
            - lambda: |-
                 id(public_ip).publish_state(body);

text_sensor:
  - platform: template
     name: "Public IP"
     id:  public_ip
     entity_category: "diagnostic"
`

ssieb, 2024-10-26:
How does knowing the external IP of the network help in any way?  Who is even going to see the resulting value?

J3bs, 2024-10-26:
@ssieb meaning no disrespect, but what is it to you? OP had a need but couldn't figure it out, I had a need and managed to figure it out, so providing the solution here so others seeking a similar solution won't need to work as hard.

If you don't see a benefit, that's perfectly fine, just move along.