esphome/feature-requests #1539: Friendly name in Home Assistant

the-cyberspace, 2021-12-07:
**Describe the problem you have/What new integration you would like**

Using a friendly name in Home Assistant

**Please describe your use case for this integration and alternatives you've tried:**

In every esphome config i`m using this to give different sensors a name.

```
# substitutions #
substitutions:
  device_name: mi_flora
  friendly_name: Mi flora
```

And i have made a sensor in ESPHome:
```
# binary sensor #
binary_sensor:
  # status #
  - platform: status
    name: "${device_name}_device_status"
```
 
And getting this in Home Assistant:

[image]

Is it possible to use the friendly_name in "Naam" in the screenshot?

aartokke, 2021-12-08:
I have it without underscores but with spaces and friendy name substitution, like:

`name: ${friendly_name} Status`

Works, but can’t remember how this shows up in HA. Haven’t installed/added a new device in a while.

the-cyberspace, 2021-12-08:
Thanks @aartokke, i know this is possible but i only want to have the name Status in the sensor name and not in "Naam". Right now this isn`t possible without filling the $friendly_name by myself in "Naam" (see sreenshot) and i want to do this in the source (esphome) so i have to do this one time. I`m having 10+ esphome devices, and when i have to give each sensor in HA a $friendly_name this is a ton of work.

aartokke, 2021-12-08:
aha sorry, i understand now what you mean. Esphome sensor's name is used as HA entityid (with underscores) and as a HA name, you want this different. i don't know if that's possible.

the-cyberspace, 2021-12-08:
@aartokke You don`t have to say sorry.. And exactly that! Maybe i wasn`t clear enough. And thats why i posted this as a Feature Request and hopefully my wish becomes reality :)

nagyrobi, 2021-12-15:
Yes this could be useful for us when we rename the friendly name, but want to keep the entity_id.

jumping2000, 2022-01-19:
I agree, it could be useful!

iticory, 2022-02-06:
+1.  I too would love it if I could specify the friendly_name and id separately for each entity.  That way when I update the friendly name to a new value, my entity_id won't change and I won't need to update all of my automations/dashboards/etc that use it. ;)

nagyrobi, 2022-02-06:
That's what I also hate.

Bascht74, 2022-05-30:
What I do right now is to set the friendly names in the sensors and change the name of the esphome device in HA (just one change per device) so it matches the "prefix" of the friendly name. 
This way I get only "status" for the entity name if you display the entities via the device. 

As @nagyrobi found I created a feature request that so can have a friendly name for the devices as well or support e.g. spaces and uppercase chars as supported in the entity names in ESPHOME.
Maybe you guys could comment the feature request as well, if you like that it gets more attention... :-)
Right now ESPHOME makes it almost impossible to have those nice truncated names in HA without fiddling afterwards.

turgu1, 2022-06-02:
I concur with this request. I see two main reasons for this feature: 1) Using a different *language* for internal ids and captions on-screen and 2) When you want the same caption on-screen to appear for more than one sensor coming from the same device. 

Sure, you can change the caption in Home Assistant by hand, but it would be a lot easier to do it in the ESPHome YAML files.

ruimarinho, 2022-06-21:
@the-cyberspace I was trying to find the commit implementing this change but can't find on on esphome/esphome. Is this being closed as it is not going to be implemented?

Bascht74, 2022-06-21:
@the-cyberspace Hi, can you point me to the solution?

nagyrobi, 2022-06-22:
Reopened

nagyrobi, 2022-07-01:
https://github.com/esphome/feature-requests/issues/1539

Bascht74, 2022-07-01:
+1 here.

Sholofly, 2022-07-02:
Wanted here too!

the-cyberspace, 2022-07-07:
Please do not reopen this issue without my permission! Open your own one! Do not tag me or this issue and do not reply here with unnecessary messages. Thank you.

closed 2022-07-07 by the-cyberspace

Sholofly, 2022-07-07:
Because?