esphome/feature-requests #1540: Support select component for modbus_controller and mapping for select component

Bascht74, 2021-12-07:
**Describe the problem you have/What new integration you would like**
The existing modbus_controller component should support the select component
if used together with new new mapping feature, this could be very sweet...

**Please describe your use case for this integration and alternatives you've tried:**
You can select a value (e.g. "low" and the right modbus command (=1) is transmitted.

**Additional context**
Could look like this: 

select:
  - platform: modbus_controller
    name: "User ventilation speed set"
    id: nilan_user_ventilation_speed_set
    address: 603
    options:
      - map:
        - 3 -> High
        - 2 -> Medium
        - 1 -> Low

martgras, 2021-12-08:
thanks for bringing this up - so far I didn't see a real use case for implementing select 
Fan is also on the todo list when I find the time and a way to test it

However you don't have to wait for the implementation and use a template select and send the modbus command yourself.
In the example below I assume setting register 603 to 1,2 or 3 sets the speed. 

```yaml
substitutions:
  modbus_ctrl_id: epever

modbus_controller:
  - id: ${modbus_ctrl_id}
    address: 0x1

select:
  - platform: template
    name: "Template select"
    optimistic: true
    options:
      - slow
      - medium
      - high
    set_action:  
      - lambda: |-
          modbus_controller::ModbusController *controller = id(${modbus_ctrl_id});
          uint16_t speed = 0; 
          uint16_t reg = 603 ;
          if (0 == x.compare(std::string{"slow"})) {
            speed = 1;
          }
          if (0 == x.compare(std::string{"medium"})) {
            speed = 2;
          }
          if (0 == x.compare(std::string{"high"})) {
            speed = 3;
          }
          modbus_controller::ModbusCommandItem set_speed_command = modbus_controller::ModbusCommandItem::create_write_single_command(controller, reg, speed);
          controller->queue_command(set_speed_command);
```

Bascht74, 2021-12-08:
@martgras 
Thx for your quick response. 
Yeah, fan (and maybe climate) would be a nice addition as well. 

Right now I am using a custom component for nilan climate devices here:
https://github.com/Jopand/esphome_components/tree/use_modbus_controller/components/nilan/climate
He wrote select for that, using your component (modbus_controller). 

I can control and get statistic data for me complete ventilation and heating system...
Lots of registers:
http://www.nilan.at/Files//Filer/Download/German/Dokumentation/Montageanleitung/Modbus%20CTS%20602/MODBUS_CTS-602_2.30_Installation-and-user-guide.pdf

He (and me) could move everything from his custom component to your modbus_controller, except
- select
- fan (I am using that)
- climate 

It would be really nice not to depend on a custom component too much and use standard components like your great addition to esphome. 

btw. my wallbox is using modbus as well and I could use a `select` for modbus_controller here as well: to select e.g. the charging mode (only when PV is available = 1, right away = 2, Standby = 3, Stop = 4) of the device. 

So I think that adding select could be a nice addition like the other stuff you added in the last release (really great!)

I switched away from the HA integration for modbus (https://www.home-assistant.io/integrations/modbus/ ) some months ago and it feels great! The maintainer of the HA integration has support for fan and climate as well so there must be a use case for it...

I'll be patient :-)

Sebastian

Bascht74, 2021-12-08:
> Fan is also on the todo list when I find the time and a way to test it

Oh, I could test it if you like...
My "nilan" fan supports modes like 0 = off, 1 = light, 2 = medium ... 4 = full as well als 0 - 100%. 
So I can choose a modbus register to set predefined values (0-4 that the devices translates to stored % values) or another register to set 0-100% directly...

Docs: 
Output.ExhaustSpeed holding register: 200 --> % Exhaust fan speed
Output.InletSpeed holding register: 201 100 --> % Inlet fan speed
Control.VentSet holding register: 1003 --> Step User ventilation step, select 0 : Off, 1..4 : Step number

nagyrobi, 2021-12-15:
+1 for this! Soon I will start working on implementing this component with Alpha-Innotec vetilation

Bascht74, 2021-12-20:
> However you don't have to wait for the implementation and use a template select and send the modbus command yourself.
> In the example below I assume setting register 603 to 1,2 or 3 sets the speed.

Hi @martgras, 

I tried your code, but it seems that my device forces me to use `create_write_multiple_command`  (but with a single address).  Using the number entity, I need to use `use_write_multiple: true` to get it to work...

Could you give me a hint how to adapt the line
`modbus_controller::ModbusCommandItem set_speed_command = modbus_controller::ModbusCommandItem::create_write_single_command(controller, reg, speed);`
that it uses `create_write_multiple_command`
... or does it make more sense to wait for the mobus_controller `select`?

Thx!

nagyrobi, 2021-12-20:
> Using the number entity

How do you use `number` entity with modbus, btw?

nagyrobi, 2021-12-20:
> Could you give me a hint

Maybe have a look at this https://github.com/martgras/esphome/blob/ad880d48227645a498fade014aa0945ce1c5f9cb/esphome/components/modbus_controller/examples/epever/epever-all-in-one.yaml#L41

martgras, 2021-12-20:
```
          modbus_controller::ModbusController *controller = id(${modbus_ctrl_id});
	  std::vector<uint16_t> speed; 
          uint16_t reg = 603 ;
          if (0 == x.compare(std::string{"slow"})) {
             speed.push_back(1);
          }
          if (0 == x.compare(std::string{"medium"})) {
            speed.push_back(2);
          }
          if (0 == x.compare(std::string{"high"})) {
            speed.push_back(3);
          }

          modbus_controller::ModbusCommandItem set_speed_command = modbus_controller::ModbusCommandItem::create_write_multiple_command(controller, reg, 1,speed);
          controller->queue_command(set_speed_command);                                                                    
```

Bascht74, 2021-12-20:
> How do you use `number` entity with modbus, btw?

https://esphome.io/components/number/modbus_controller.html

martgras, 2021-12-20:
I'm using it only for testing to change the value for Battery Capacity of my solar charger
```
number:
  - platform: modbus_controller
    modbus_controller_id: epever
    id: battery_capacity_number
    name: "Battery Cap Number"
    address: 0x9001
    use_write_multiple: true
    response_size: 1
    #offset: 0 
    byte_offset: 0
    value_type: U_WORD
    lambda: "return  x * 1.0; "
    multiply: 1.0
```

nagyrobi, 2021-12-20:
Got it meanwhile, works perfect:
```
- platform: modbus_controller
  modbus_controller_id: ${modbus_ctrl_id}
  name: "Alap elszívás"
  icon: "mdi:fan-speed-1"
  id: lg_300_speed_out_1
  address: 10
  unit_of_measurement: "%"
  value_type: U_WORD
  min_value: 0
  max_value: 100
  step: 1
  mode: slider
```

Bascht74, 2022-05-15:
Implemented! Great! Closing this issue

closed 2022-05-15 by Bascht74