esphome/feature-requests #1541: Support for the Beca/Moes/Qiumi thermostat with ESP8266 WiFi module

bensuffolk, 2021-12-08:
**Describe the problem you have/What new integration you would like**

Support for the Beca/Moes/Qiumi thermostat with ESP8266 WiFi module. E.G BHT-002, BHT-6000, BHT-3000, BAC-1000

**Please describe your use case for this integration and alternatives you've tried:**

I have been using the custom firmware option provided by https://github.com/fashberg/WThermostatBeca to access my thermostats via Home Assistant. They work with this, but rely on MQTT, and I would really like them to be supported via the native API like all my other ESPHome devices.

The version of the thermostats I have is the underfloor heating one with a second sensor in the floor as well.

**Additional context**

Choochmeque, 2022-01-03:
@bensuffolk , have you looked at [tuya](https://esphome.io/components/climate/tuya.html) ?
Moes BHT-002 perfectly works with this setup.

Fran6u, 2022-01-08:
> @bensuffolk , have you looked at [ESPHome Tuya Climate](https://esphome.io/components/climate/tuya.html) ? Moes BHT-002 perfectly works with this setup.

Hi @Choochmeque. I cant quite figure the process out (especially the part about registering a TUYA MCU connection), so could you possibly share your esphome code as a guide and maybe some pointers too? Many thanks in advance.

Choochmeque, 2022-01-09:
Hi @Fran6u.

When you register your tuya connection by using **tuya:** and upload fw to the thermostat, you will smth like this in the log:

`[13:56:42][C][tuya:028]: Tuya:
[13:56:42][C][tuya:039]:   Datapoint 1: switch (value: OFF)
[13:56:42][C][tuya:041]:   Datapoint 2: int value (value: 60)
[13:56:42][C][tuya:041]:   Datapoint 3: int value (value: 46)
[13:56:42][C][tuya:045]:   Datapoint 4: enum (value: 1)
[13:56:42][C][tuya:039]:   Datapoint 5: switch (value: OFF)
[13:56:42][C][tuya:039]:   Datapoint 6: switch (value: OFF)
[13:56:42][C][tuya:037]:   Datapoint 101: raw (value: 00.06.28.00.08.1E.1E.0B.1E.1E.0D.1E.00.11.2C.00.16.1E.00.06.28.00.08.28.1E.0B.28.1E.0D.28.00.11.28.00.16.1E.00.06.2A.00.08.28.1E.0B.28.1E.0D.28.00.11.28.00.16.1E (54))
[13:56:42][C][tuya:041]:   Datapoint 102: int value (value: 169)
[13:56:42][C][tuya:039]:   Datapoint 104: switch (value: OFF)
[13:56:42][C][tuya:055]:   Product: 'IAYz2WK1th0cMLmL1.0.0'
[13:56:49][D][tuya:272]: Datapoint 3 update to 46
[13:56:49][D][climate:381]: 'Thermostat' - Sending state:
[13:56:49][D][climate:384]:   Mode: OFF
[13:56:49][D][climate:386]:   Action: OFF
[13:56:49][D][climate:395]:   Preset: NONE
[13:56:49][D][climate:404]:   Current Temperature: 23.00°C
[13:56:49][D][climate:410]:   Target Temperature: 30.00°C
[13:56:49][D][sensor:113]: 'Temperature': Sending state 23.00000 °C with 1 decimals of accuracy
[13:56:49][D][tuya:272]: Datapoint 102 update to 169`

and then you can play whit datapoints (ie create sensors, switches etc).

Here below my working yaml for the Moes BHT-002 thermostat with HA integration support:

`esphome:
  name: in-therm-pwt-002-1
  platform: ESP8266
  board: esp01_1m
  on_boot:
    priority: 600
    then:
      - switch.turn_off: power_switch

# Make sure logging is not using the serial port
logger:
  baud_rate: 0
  
# Enable Home Assistant API
api:

ota:
  password: "YOUR_OTA_PASSWORD"

wifi:
  ssid: "YOUR_WIFI_SSID"
  password: "YOUR_WIFI_PASSWORD"
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "In-Therm-Pwt-002-1"
    password: "YOUR_GENERATED_PASSWORD"

captive_portal:

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

# Getting time from HA
time:
  - platform: homeassistant
    id: ha_time

# Register the Tuya MCU connection
tuya:
  time_id: ha_time

climate:
  - platform: tuya
    name: "Thermostat"
    switch_datapoint: 1
    target_temperature_datapoint: 2
    current_temperature_datapoint: 3
    eco_datapoint: 5
    eco_temperature: 20 °C
    temperature_multiplier: 0.5
    visual:
      min_temperature: 20 °C
      max_temperature: 35 °C
      temperature_step: 0.5 °C

# Create a sensor
sensor:
  - platform: "tuya"
    name: "Temperature"
    sensor_datapoint: 3
    unit_of_measurement: "°C"
    device_class: "temperature"
    accuracy_decimals: 1
    filters:
      - multiply: 0.5

switch:
  - platform: "tuya"
    name: "Power"
    id: power_switch
    switch_datapoint: 1
    internal: true
  - platform: "tuya"
    name: "Eco Mode"
    icon: "mdi:earth"
    switch_datapoint: 5
    internal: true
  - platform: "tuya"
    name: "Lock"
    icon: "mdi:lock"
    switch_datapoint: 6
    internal: true`

Fran6u, 2022-01-09:
> Hi @Fran6u.
> 
> When you register your tuya connection by using "**tuya:**" keyword and upload fw to the thermostat, you will see smth like this in the log:
> 
> ```
> [13:56:42][C][tuya:028]: Tuya:
> [13:56:42][C][tuya:039]:   Datapoint 1: switch (value: OFF)
> [13:56:42][C][tuya:041]:   Datapoint 2: int value (value: 60)
> [13:56:42][C][tuya:041]:   Datapoint 3: int value (value: 46)
> [13:56:42][C][tuya:045]:   Datapoint 4: enum (value: 1)
> [13:56:42][C][tuya:039]:   Datapoint 5: switch (value: OFF)
> [13:56:42][C][tuya:039]:   Datapoint 6: switch (value: OFF)
> [13:56:42][C][tuya:037]:   Datapoint 101: raw (value: 00.06.28.00.08.1E.1E.0B.1E.1E.0D.1E.00.11.2C.00.16.1E.00.06.28.00.08.28.1E.0B.28.1E.0D.28.00.11.28.00.16.1E.00.06.2A.00.08.28.1E.0B.28.1E.0D.28.00.11.28.00.16.1E (54))
> [13:56:42][C][tuya:041]:   Datapoint 102: int value (value: 169)
> [13:56:42][C][tuya:039]:   Datapoint 104: switch (value: OFF)
> [13:56:42][C][tuya:055]:   Product: 'IAYz2WK1th0cMLmL1.0.0'
> [13:56:49][D][tuya:272]: Datapoint 3 update to 46
> [13:56:49][D][climate:381]: 'Thermostat' - Sending state:
> [13:56:49][D][climate:384]:   Mode: OFF
> [13:56:49][D][climate:386]:   Action: OFF
> [13:56:49][D][climate:395]:   Preset: NONE
> [13:56:49][D][climate:404]:   Current Temperature: 23.00°C
> [13:56:49][D][climate:410]:   Target Temperature: 30.00°C
> [13:56:49][D][sensor:113]: 'Temperature': Sending state 23.00000 °C with 1 decimals of accuracy
> [13:56:49][D][tuya:272]: Datapoint 102 update to 169
> ```
> 
> and then you can play whit datapoints (ie create sensors, switches etc).
> 
> Here below my working yaml for the Moes BHT-002 thermostat with HA integration support:
> 
> ```
> esphome:
>   name: in-therm-pwt-002-1
>   platform: ESP8266
>   board: esp01_1m
>   on_boot:
>     priority: 600
>     then:
>       - switch.turn_off: power_switch
> 
> # Make sure logging is not using the serial port
> logger:
>   baud_rate: 0
>   
> # Enable Home Assistant API
> api:
> 
> ota:
>   password: "YOUR_OTA_PASSWORD"
> 
> wifi:
>   ssid: "YOUR_WIFI_SSID"
>   password: "YOUR_WIFI_PASSWORD"
>   fast_connect: true
> 
>   # Enable fallback hotspot (captive portal) in case wifi connection fails
>   ap:
>     ssid: "In-Therm-Pwt-002-1"
>     password: "YOUR_GENERATED_PASSWORD"
> 
> captive_portal:
> 
> uart:
>   rx_pin: GPIO3
>   tx_pin: GPIO1
>   baud_rate: 9600
> 
> # Getting time from HA
> time:
>   - platform: homeassistant
>     id: ha_time
> 
> # Register the Tuya MCU connection
> tuya:
>   time_id: ha_time
> 
> climate:
>   - platform: tuya
>     name: "Thermostat"
>     switch_datapoint: 1
>     target_temperature_datapoint: 2
>     current_temperature_datapoint: 3
>     eco_datapoint: 5
>     eco_temperature: 20 °C
>     temperature_multiplier: 0.5
>     visual:
>       min_temperature: 20 °C
>       max_temperature: 35 °C
>       temperature_step: 0.5 °C
> 
> # Create a sensor
> sensor:
>   - platform: "tuya"
>     name: "Temperature"
>     sensor_datapoint: 3
>     unit_of_measurement: "°C"
>     device_class: "temperature"
>     accuracy_decimals: 1
>     filters:
>       - multiply: 0.5
> 
> switch:
>   - platform: "tuya"
>     name: "Power"
>     id: power_switch
>     switch_datapoint: 1
>     internal: true
>   - platform: "tuya"
>     name: "Eco Mode"
>     icon: "mdi:earth"
>     switch_datapoint: 5
>     internal: true
>   - platform: "tuya"
>     name: "Lock"
>     icon: "mdi:lock"
>     switch_datapoint: 6
>     internal: true
> ```

@Choochmeque. Thanks for your response, most of which is clear, apart from where to register the 'tuya' device. Thanks again

Choochmeque, 2022-01-09:
@Fran6u, 

>  apart from where to register the 'tuya' connection.

This line in the my yaml:
```
# Register the Tuya MCU connection
tuya:
```

Fran6u, 2022-01-09:
@Choochmeque. Okay. Be back in 10minutes, let me give a go.

Fran6u, 2022-01-09:
> @Fran6u,
> 
> > apart from where to register the 'tuya' connection.
> 
> This line in the my yaml:
> 
> ```
> # Register the Tuya MCU connection
> tuya:
> ```
> 
> It means that ESPHome should use Tuya mcu support.

@Choochmeque. It worked! I am so excited now, thanks a million. Now to pull everything together and see how it works in the thermostat. I so much appreciate your help, thanks again.

Fran6u, 2022-01-09:
@Choochmeque,
Hi, it's me again. I was wondering, could you help me make sense of this;
[12:38:00][D][tuya:284]: Datapoint 23 update to 0
[12:38:00][D][tuya:264]: Datapoint 1 update to ON
[12:38:00][D][tuya:272]: Datapoint 24 update to 18
[12:38:00][D][tuya:272]: Datapoint 101 update to 0
[12:38:00][D][tuya:272]: Datapoint 16 update to 17
[12:38:00][D][tuya:284]: Datapoint 2 update to 1
[12:38:00][D][tuya:272]: Datapoint 19 update to 35
[12:38:00][D][tuya:272]: Datapoint 26 update to 5
[12:38:00][D][tuya:272]: Datapoint 106 update to 1
[12:38:00][D][tuya:272]: Datapoint 27 update to 0
[12:38:00][D][tuya:284]: Datapoint 43 update to 0
[12:38:00][D][tuya:284]: Datapoint 104 update to 1
[12:38:00][D][tuya:301]: Datapoint 45 update to 00000000
[12:38:00][D][tuya:264]: Datapoint 103 update to OFF
[12:38:00][D][tuya:256]: Datapoint 108 update to 06.00.14.08.00.0F.0B.1E.0F.0C.1E.0F.11.00.16.16.00.0F.08.00.16.17.00.0F (24)

Datapoint 1 and 103 are the only ones that appear to make any sense to me. Thanks.

Choochmeque, 2022-01-09:
@Fran6u ,
I did the following:
Assigned switches to "switch" datapoints and exposed them to HA, then used this switches and checked changes on the thermostat's display.
```
switch:
  - platform: "tuya"
    name: "Power"
    id: power_switch
    switch_datapoint: 1
    internal: true
  - platform: "tuya"
    name: "Eco Mode"
    icon: "mdi:earth"
    switch_datapoint: 5
    internal: true
  - platform: "tuya"
    name: "Lock"
    icon: "mdi:lock"
    switch_datapoint: 6
    internal: true
```
After that I hide this switches by using internal:true

As you can see in my log:
```
[13:56:42][C][tuya:041]:   Datapoint 2: int value (value: 60)
[13:56:42][C][tuya:041]:   Datapoint 3: int value (value: 46)
```
datapoint 2 is a target temperature multiplied by two (60/2 = 30C)
datapoint 3 is a current temperature multiplied by two (46/2 = 23C)  

that's why i put
```
temperature_multiplier: 0.5
```
means 60 * 0.5 = 30C etc

Fran6u, 2022-01-09:
@Choochmeque,
Thanks a million for your time, you are a star. I have it all set up now, and will be installing it on my wall in a few minutes.
thanks again for your time.

beppexx80, 2022-09-20:
@Choochmeque hi, in my thermostat i have the firmware Wthermostat, for programming firmware ESPhome, do i have to connect it with the wires and program it from usb?

Choochmeque, 2022-09-20:
@beppexx80 , hi. 
Can you have access to web interface? If yes, then you can update fw from there. (I did the same ;) )

beppexx80, 2022-09-20:
> @beppexx80 , hi. 
> Can you have access to web interface? If yes, then you can update fw from there. (I did the same ;) )

yes, i have access to web... 
test... thanks

beppexx80, 2022-09-20:
@Choochmeque hi, ok, is all ok, Is there a list with the explanation of the datapoints?

Choochmeque, 2022-09-21:
@beppexx80 hi.
I have not found this list yet. So I used method I described above.
You can start from https://tasmota.github.io/docs/TuyaMCU/, maybe it could help.

vapustovoyt, 2022-10-19:
Hello. Thank you very much for supporting the thermostat with ESP8266. My thermostat BHT-002 has a schedule mode on Datapoint 4. If I turn it on via the device:
```
[20:22:17][D][tuya:300]: Datapoint 4 update to 0
[20:22:17][D][switch:037]: 'schedules': Sending state ON
[20:22:19][D][tuya:300]: Datapoint 4 update to 1
[20:22:19][D][switch:037]: 'schedules': Sending state OFF
```
If I click in the web interface:
```
[20:22:20][D][switch:013]: 'schedules' Turning ON.
[20:22:20][D][tuya:547]: Setting datapoint 4 to 0
[20:22:20][E][tuya:552]: Attempt to set datapoint 4 with incorrect type```

I added this to the switch section:

```
  - platform: "tuya"
    name: "schedules"
    icon: "mdi:calendar"
    switch_datapoint: 4```

How can i fix it? Thx!

piotr-kubiak, 2022-11-13:
> I added this to the switch section:
> 
> ```
> - platform: "tuya"
>     name: "schedules"
>     icon: "mdi:calendar"
>     switch_datapoint: 4
> ```
> 
> How can i fix it? Thx!

Hey @vapustovoyt, 
Just use `select` component:
```
select:
  - platform: "tuya"
    name: "schedules"
    icon: "mdi:calendar"
    enum_datapoint: 4
    options:
      0: Schedule
      1: Manual
```

vapustovoyt, 2022-11-18:
> Hey @vapustovoyt, Just use `select` component:
>
> ```
> select:
>   - platform: "tuya"
>     name: "schedules"
>     icon: "mdi:calendar"
>     enum_datapoint: 4
>     options:
>       0: Schedule
>       1: Manual
> ```

Thx!!!!!!!