esphome/feature-requests #1542: help

cxc288, 2021-12-09:
LibraryManager: Installing id=12620 @ 1.2.7
UndefinedPackageVersion: Could not find a version that satisfies the requirement '1.2.7' for your system 'linux_aarch64':
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/root/.platformio/platforms/espressif8266/builder/main.py", line 238:
    target_elf = env.BuildProgram()
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/platformio.py", line 62:
    env.ProcessProjectDeps()
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/piolib.py", line 1036:
    project.install_dependencies()
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/piolib.py", line 868:
    lm.install(uri)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/lib.py", line 281:
    force=force,
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/package.py", line 704:
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/lib.py", line 119:
    requirements or "latest", util.get_systype()
========================= [FAILED] Took 10.04 seconds =========================

cxc288, 2021-12-09:
WHY?

martgras, 2021-12-09:
This repository is meant to suggest new features for esphome and not for troubleshooting.
I suggest going to https://discord.gg/KhAMKrd and ask your question (and add a bit more context please)

closed 2022-07-06 by nagyrobi