esphome/feature-requests #1544: Support disabling the physical switch

andreas-bulling, 2021-12-11:
Hi,

It would be super nice if there was a way to remotely disable the physical switch - e.g. to prevent the kids from pressing certain switches. Tasmota supports this it seems - would be great to see it in esphome as well.

nagyrobi, 2021-12-12:
This should be possible using `homeassistant` [platform](https://esphome.io/components/binary_sensor/homeassistant.html). You can route entities back from Home Assistant into ESPHome firmware, and based on that, you can run checks within ESPHome to take the physical input in consideration or not, when they are toggled.

andreas-bulling, 2021-12-12:
Not sure I understand what you mean - I mean the **physical** switch, i.e. limiting switching of the light to HA

randybb, 2021-12-12:
One option is just sent the command to HA and then HA will decide based on programed logic if it can switch your light on or not. Or another option is to modify your (light/switch) components and add while condition based on remote variable/state in HA. Don't think that there is a need for a special function.

nagyrobi, 2021-12-12:
> Not sure I understand what you mean - I mean the **physical** switch, i.e. limiting switching of the light to HA

Yes, the **physical** switch enters as a GPIO to the MCU, you can control what to do or not to do. You can of course disable it if you want to.

andreas-bulling, 2021-12-13:
> One option is just sent the command to HA and then HA will decide based on programed logic if it can switch your light on or not. Or another option is to modify your (light/switch) components and [...]

Thanks for your suggestion. But this means that this solution requires an active connection to HA - if that is lost/not available this solution wouldn't work, would it?

That'd be the advantage of what I propose - you remotely set a "flag" in your node that indicates that the physical switch should be ignored. Even if HA is temporarily down this flag would be respected. Kids is one thing but there could also be safety-critical use case where it's even more important that it can be ensured that the switch is disabled.

For those who stumble over this feature request in the future (and me, if this feature will not be picked up/implemented): Do you happen to have example code for how to do this? I have not seen this being done anywhere and I have no idea where to start...

andreas-bulling, 2021-12-13:
> > Not sure I understand what you mean - I mean the **physical** switch, i.e. limiting switching of the light to HA
> 
> Yes, the **physical** switch enters as a GPIO to the MCU, you can control what to do or not to do. You can of course disable it if you want to.

I see - same question to you: Do you happen to have/know how to implement this? Ideally without the need for an active connection to HA (local)?

nagyrobi, 2021-12-13:
You can achieve what you want using [global variables](https://esphome.io/guides/automations.html#global-variables).
You might reconsider your approach though, disabling a device without having a way to re-enable it can be a dangerous scenario in emergency situations, I wouldn't ever do that.

andreas-bulling, 2021-12-13:
My idea was to implement functionality that would allow to re-enable it using a double press for example. As long as our kids don't know, this would work

nagyrobi, 2021-12-13:
Can be done, [double-click](https://esphome.io/components/binary_sensor/index.html) is supported locally-

andreas-bulling, 2021-12-13:
And these global variables can be set remotely (through HA)?

nagyrobi, 2021-12-13:
Yes, you need to learn how to use lambdas.

andreas-bulling, 2021-12-13:
I have used lambdas on a Sonoff UV for our garage doors - thanks for the pointer!

remailednet, 2021-12-14:
@andreas-bulling - You can check out Kaufman's configuration for their PLF10 on GitHub for an idea of how they made the power button on a plug configurable through the Home Assistant UI. [https://github.com/KaufHA/PLF10](https://github.com/KaufHA/PLF10)

andreas-bulling, 2021-12-14:
@remailednet Thanks a lot - this looks super helpful!

andreas-bulling, 2021-12-22:
worked, closing

closed 2021-12-22 by andreas-bulling